nutdotnet/WinNUT-Client #1: TypeInitializationException when loading WinNUT

gbakeman, 2022-05-05:
_Imported from https://github.com/gawindx/WinNUT-Client/issues/75_

Seen when starting WinNUT without any user data beforehand. Happens on Windows 10, Server 2019 and Server 2012.

Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_client.WinNUT_Params' threw an exception.
Exception stack trace:
at WinNUT_client.WinNUT_Params.Init_Params()
at WinNUT_client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
at System.EventHandler.Invoke(Object sender, EventArgs e)
at System.Windows.Forms.Form.OnLoad(EventArgs e)
at System.Windows.Forms.Form.OnCreateControl()
at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
at System.Windows.Forms.Control.CreateControl()
at System.Windows.Forms.Control.WmShowWindow(Message& m)
at System.Windows.Forms.Control.WndProc(Message& m)
at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
at System.Windows.Forms.Form.WmShowWindow(Message& m)
at System.Windows.Forms.Form.WndProc(Message& m)
at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

gbakeman, 2022-10-19:
I believe this will be corrected by #35 , someone please comment if this is not the case.

mightyoj, 2022-10-23:
I do not have the registry key seen in #35 and am still seeing this particular issue with the latest release, v2.2.8328.

`Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_Client_Common.WinNUT_Params' threw an exception.
Exception stack trace: 
   at WinNUT_Client.My.MyApplication.Generate_Button_Click(Object sender, EventArgs e) in C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\ApplicationEvents.vb:line 109
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

gbakeman, 2022-10-23:
Thank you for commenting @mightyoj . What were you doing/clicking on when this occurred? Could you reproduce it and provide a log as well?

mightyoj, 2022-10-23:
The above occurs when you click "Generate Report and Close WinNUT"

This is the error I see on launch:
`Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_Client_Common.WinNUT_Params' threw an exception.
Exception stack trace: 
   at WinNUT_Client_Common.WinNUT_Params.Init_Params()
   at WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e) in C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:line 140
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ContainerControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

mightyoj, 2022-10-23:
I saw that line 140 references the registry keys being created, so I copied them from another system which WinNUT has no trouble. When I started, I now see this original error on launch:

`Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_client.WinNUT_Params' threw an exception.
Exception stack trace: 
   at WinNUT_client.WinNUT_Params.Init_Params()
   at WinNUT_client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ContainerControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

gbakeman, 2022-10-23:
Ok, that's really interesting. You may have actually found two different bugs. @gawindx was trying to get to the bottom of this in the old issue, but I don't think a fix was found.

I'd like to know more about your situation:

* OS Edition and update version
* This is the first time you installed/ran the program?
* Is there anything unusual about the user account you're using to run WinNUT?

I'll try to take a deeper dive tomorrow.

mightyoj, 2022-10-23:
OS info is at my comment in [gawindx/WinNUT-Client #75](https://github.com/gawindx/WinNUT-Client/issues/75#issuecomment-997285911).

I've installed WinNUT on three additional machines but this one has been the problematic one. I've been using a 1.x release since my first comment in your old repo. 

The accounts are the same in all the machines.

gbakeman, 2022-10-23:
@mightyoj I've created a debug build [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3308825545). Please download & execute it. I've attempted to upgrade the exception dialog so when an exception is encountered, it will try to get more details about what happened. I'd like for you to copy the contents of the message box, and also try to generate a bug report and upload that here as well. Thank you!

mightyoj, 2022-10-24:
First launch:
`Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_Client_Common.WinNUT_Params' threw an exception.
Exception stack trace:
   at WinNUT_Client_Common.WinNUT_Params.Init_Params()
   at WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e) in D:\a\WinNUT-Client\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:line 140
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

InnerException present:
Exception type: System.NullReferenceException
Exception message: Object reference not set to an instance of an object.
Exception stack trace:
   at WinNUT_Client_Common.CryptData.GetProcessorId() in D:\a\WinNUT-Client\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\CryptData.vb:line 116
   at WinNUT_Client_Common.CryptData.Get_UniqueKey_Hash() in D:\a\WinNUT-Client\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\CryptData.vb:line 97
   at WinNUT_Client_Common.CryptData..ctor() in D:\a\WinNUT-Client\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\CryptData.vb:line 20
   at WinNUT_Client_Common.WinNUT_Params..cctor() in D:\a\WinNUT-Client\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\WinNUT_Params.vb:line 15
`

Generate bug report:
`Exception type: System.UnauthorizedAccessException
Exception message: Access to the path 'C:\Program Files (x86)\WinNUT-client\debug\Data\CrashLog' is denied.
Exception stack trace:
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.Directory.InternalCreateDirectory(String fullPath, String path, Object dirSecurityObj, Boolean checkHost)
   at System.IO.Directory.InternalCreateDirectoryHelper(String path, Boolean checkHost)
   at System.IO.Directory.CreateDirectory(String path)
   at Microsoft.VisualBasic.FileIO.FileSystem.CreateDirectory(String directory)
   at Microsoft.VisualBasic.MyServices.FileSystemProxy.CreateDirectory(String directory)
   at WinNUT_Client.My.MyApplication.Generate_Button_Click(Object sender, EventArgs e) in D:\a\WinNUT-Client\WinNUT-Client\WinNUT_V2\WinNUT-Client\ApplicationEvents.vb:line 174
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
`

gbakeman, 2022-10-26:
@mightyoj When you downloaded the WinNUT debug build, did you extract it to the WinNUT installation directory in the program files folder? I don't think we've ever accounted for the possibility that WinNUT could be running from a directory that it doesn't have access to write in. I've upgraded some code to account for this so WinNUT will write in the temp directory as a failsafe, but I haven't gotten to the core of this crypto issue yet. If you wouldn't mind running the [latest debug build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3333364963), I just want to make sure that error reporting is working correctly before I move on. You're uncovering several bugs here :)

gbakeman, 2022-10-28:
I've been able to look into this a little. It seems like some WMI objects aren't present on certain systems, which WinNUT depends on for cryptography purposes. I'm going to need to upgrade how our crypto system works, which will likely mean passwords and such need to be re-entered. Will update.

gbakeman, 2022-10-30:
Heads up everyone, I believe I have a fix ready for you. If you are experiencing this error, please first [try my debug build here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3333364963) and ensure that the crash log functionality is working correctly (I don't actually need the crash log, just want to know that it's working.)

After you've done that, please [try the latest debug build 8338](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3357034334) here and see if you can start WinNUT successfully. I've redone the cryptography in this program so it uses standard .Net functionality as much as possible. This means that if you have your username and password saved already, it will need to be reset in the program.

gbakeman, 2022-11-12:
Just a heads up, but I think this bug is fixed so I'm going to merge the changes into the dev branch and make a new pre-release soon. Please comment back here to confirm if it's fixed or if there's still a problem.

closed 2022-11-12 by gbakeman

mightyoj, 2022-11-13:
> Heads up everyone, I believe I have a fix ready for you. If you are experiencing this error, please first [try my debug build [...]

Crash report is working!

> After you've done that, please [try the latest debug build 8338](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3357034334) here and see if you can start WinNUT successfully. I've redone [...]

Build 8338 is good!

You fixed it! Thanks!

gbakeman, 2022-11-13:
Thank you for verifying! 😊

FatsWaler, 2023-05-26:
This error has just stated happening to me out of nowhere. I have tried the latest stable edition. Is the any step by step instructions to fix it?

Thanks

gbakeman, 2023-05-26:
@FatsWaler could you please confirm what version of WinNUT you're running, and provide the generated error report? This was fixed in a pre-release version so you may need to update to that.

PR #49 (2022-11-12): Upgrade unhandled exception handling, cryptography
Troubleshooting the TypeInitializationException being thrown at startup.

* Create a subroutine to generate a nice message from exception data.
* Call the sub to generate the initial message, then run it again if an InnerException is present.
* Wrap getting the Reg_Key var in a Try block to catch the TypeInit error and hopefully catch the inner exception.
- WinNUT used WMI objects to generate a hash for cryptography. Sometimes these objects were not present on the end user's system, causing an exception. WinNUT now uses the Data Protection API [... cut]