nutdotnet/WinNUT-Client #4: WinNUT v2.0.7722.30975 Always stays disconnected after suspend/resume cycle.

UltraHKR, 2022-05-28:
Hi, I´m running WinNUT on my Surface Pro3 this is connected to a CDP UPS (monitored by NUT on OpenWRT), everything works wonderfully.

Because this is my main machine I do suspend/resume cycle various times a day, WinNUT sometimes comes back with a Max retries reached and doesn't retry connecting.
The same things happens if one of my Wi-Fi mesh AP's goes down for long enough (time to hop to next AP), it disconnects uses all retries and ends up disconnected.

Even if my network and NUT server are still up.

Could you please add code to check network status change and retry reconnection if network get disconnected / reconnected, or power status (coming out of hibernation) changes, by tracking said network status.

gbakeman, 2022-07-27:
Thank you for reporting this. Can you confirm what version of WinNUT this is occurring on? We believe it may be a bug introduced in the pre-release version 2.1. For tracking purposes, I think this is related to:

- https://github.com/gawindx/WinNUT-Client/issues/88
- https://github.com/gawindx/WinNUT-Client/issues/124
- https://github.com/gawindx/WinNUT-Client/issues/132

UltraHKR, 2022-08-02:
So I updated to 2.1.7740.35837.... Last available version that I can find an installer...

And now the runtime displays as "" (empty), on last v2.0.7722.30975 it was working what gives....?

gbakeman, 2022-08-02:
I'm working on a fairly big update in #11 right now. Feel free to follow my progress there, and I'll upload a debug build if you wouldn't mind testing later. If it doesn't fix this issue, then it should at least narrow down what's causing it.

gbakeman, 2022-08-06:
@UltraHKR please test the pre-release [here](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.2.8253) and let me know if you still have the same problem as above. If so, then please make sure you upload the full debug log.

UltraHKR, 2022-08-10:
Hey, sorry for answering late...

I think it works better, at least you worked out the startup hangs

UltraHKR, 2022-08-11:
Hi,

on lastest version "remaining time:" is totally broken....

UPS config:
TL (Tripp Lite/SU2200RTXL2UA/208)
battery.charge (Battery charge (percent of full)) : 100
battery.charge.low (Remaining battery level when UPS switches to LB (percent)) : 25
battery.charge.warning (Battery level when UPS switches to Warning state (percent)) : 50
battery.runtime (Battery runtime (seconds)) : 653
battery.temperature (Battery temperature (degrees C)) : 34.9
battery.type (Battery chemistry) : PbAc
battery.voltage (Battery voltage (V)) : 54.3
battery.voltage.nominal (Nominal battery voltage (V)) : 48.0
device.mfr (Description unavailable) : Tripp Lite
device.model (Description unavailable) : SU2200RTXL2UA
device.serial (Description unavailable) : 1234567
device.type (Description unavailable) : ups
driver.name (Driver name) : usbhid-ups
driver.parameter.pollfreq (Description unavailable) : 30
driver.parameter.pollinterval (Description unavailable) : 2
driver.parameter.port (Description unavailable) : auto
driver.parameter.synchronous (Description unavailable) : no
driver.version (Driver version - NUT release) : 2.7.4
driver.version.data (Description unavailable) : TrippLite HID 0.82
driver.version.internal (Internal driver version) : 0.41
input.frequency (Input line frequency (Hz)) : 59.9
input.transfer.high (High voltage transfer point (V)) : 138.0
input.transfer.high.max (greatest settable high voltage transfer point (V)) : 138
input.transfer.high.min (smallest settable high voltage transfer point (V)) : 133
input.transfer.low (Low voltage transfer point (V)) : 65.0
input.transfer.low.max (greatest settable low voltage transfer point (V)) : 70
input.transfer.low.min (smallest settable low voltage transfer point (V)) : 65
input.voltage (Input voltage (V)) : 118.9
input.voltage.nominal (Nominal input voltage (V)) : 120
output.current (Output current (A)) : 7.8
output.frequency (Output frequency (Hz)) : 59.9
output.frequency.nominal (Nominal output frequency (Hz)) : 60
output.voltage (Output voltage (V)) : 119.8
output.voltage.nominal (Nominal output voltage (V)) : 120
runtimecal (Description unavailable) : 270,100,840,50,8460,6
ups.beeper.status (UPS beeper status) : enabled
ups.delay.shutdown (Interval to wait after shutdown with delay command (seconds)) : 20
ups.delay.start (Interval to wait before (re)starting the load (seconds)) : 30
ups.firmware (UPS firmware) : 208
ups.load (Load on UPS (percent of full)) : 56
ups.mfr (UPS manufacturer) : Tripp Lite
ups.model (UPS model) : SU2200RTXL2UA
ups.power (Current value of apparent power (VA)) : 916.0
ups.power.nominal (UPS power rating (VA)) : 2200
ups.productid (Product ID for USB devices) : 4003
ups.realpower.nominal (UPS real power rating (W)) : 1600
ups.serial (UPS serial number) : 1234567
ups.status (UPS status) : OL
ups.test.result (Results of last self test) : Done and passed
ups.timer.reboot (Time before the load will be rebooted (seconds)) : -1
ups.timer.shutdown (Time before the load will be shutdown (seconds)) : -1
ups.timer.start (Time before the load will be started (seconds)) : 0
ups.vendorid (Vendor ID for USB devices) : 09ae
ups.watchdog.status (Description unavailable) : 0

[image]

gbakeman, 2022-08-11:
No worries, thank you for getting back to me. Would you please test a suspend/resume cycle and let me know of the results? If successful, I think we should close this issue as resolved. If you wouldn't mind, I'd appreciate it if you opened a new ticket for the remaining time being blank. I know it's been reported on gawindx's repository before and I'd like to investigate why the remaining time calculation was disabled.

cinergi2, 2022-08-12:
Reconnect after suspend/resume is still not working for me on version 2.2.8255.34401.  I have to manually reconnect using Connection > Reconnect.

gbakeman, 2022-08-12:
Could you please provide a debug log including when the computer resumes, and about 30 seconds after?

UltraHKR, 2022-08-12:
Hi, gbakeman I had to create a bug report because WinNUT is not behaving properly
(on my server it works fine, at least is no hanging...)

gbakeman, 2022-08-12:
Was there an issue creating the report? When you go to make it, it will open a folder where the report is located. You can then drag the report file into the comment response below to upload it. Thank you for reporting!

btehan, 2022-10-19:
FWIW, I'm seeing this on 2.2.8303.20598.

gbakeman, 2022-10-21:
Everyone, I believe this issue has been fully fixed as of pre-release v2.2.8328. Please update or [download the latest version](https://github.com/nutdotnet/WinNUT-Client/releases) from the releases and report back if this issue persists.

closed 2022-10-21 by gbakeman

btehan, 2022-10-21:
I updated to 2.2.8328 last night and it reconnected on its own this morning.  Thanks gbakeman :)

gbakeman, 2022-10-21:
You're welcome, thank you for letting me know!