nutdotnet/WinNUT-Client #5: WinNUT immediately disconnects upon UPS going on battery power

MartinKurtz, 2022-06-24:
See here

https://github.com/gawindx/WinNUT-Client/issues/140

MartinKurtz, 2022-08-04:
Copy from the previously mentioned issue

I have not tested this, but you can possibly skip 1 to 4 and use the "latest" version and do the same thing

    Uninstall WinNUT
    Resart PC
    Install the newest beta release
    It should now crash when starting it up saying something about string length
    Close/Shutdown WinNUT
    Go to whereever you installed WinNUT
    Enter a folder named de-DE for german, if you have another language, pick that folder
    Delete the .dll file in there and start WinNUT
    WinNUT should now fall back to the default language, which is english, but if you can read this or are computerlingual in any way this shouldnt be a problem, and in that language the string that is apparently the issue, is shorter and does apparently not cause this issue usually
    Magic

After following these steps it worked fine for me, however just installing the newest pre release version (https://github.com/gawindx/WinNUT-Client/releases/tag/2.1.7740.35837) at the time of writing did not fix the issue even though it said so in the changelog

gbakeman, 2022-08-05:
Thank you for making this issue report. I also appreciate that you included steps to reproduce the issue, but I'm not sure they are the same? I think the https://github.com/gawindx/WinNUT-Client/issues/140 issue you linked above is also related to https://github.com/gawindx/WinNUT-Client/issues/133, but the steps you mentioned in your second comment sound like https://github.com/gawindx/WinNUT-Client/issues/74, https://github.com/gawindx/WinNUT-Client/issues/135, and #3 here on the repository. Can you confirm if these are indeed separate issues?

MartinKurtz, 2022-08-05:
The steps i posted are not steps to reproduce the issue, they fix the issue

As far as I can tell these are all the same issue, if you look through the log file i attached in the issue on the no 140 issue you find this:
24.06.2022 16:48:41 Pid: 5388 UPS_Network : Error When Retrieve_UPS_Data : Der Text muss kürzer als 64 Zeichen sein.
Parametername: Text
Der tatsächliche Wert war WinNUT - 2.0
In Verbindung gebracht
Batterie (100%)
Batterie OK.

which is the same as in numbers 3 135 74 and i suspect quite a few others with similar problems
which leads me to believe they are all the same issue

MartinKurtz, 2022-08-05:
Another thing:
I upgraded to version [v2.1.7740.35837](https://github.com/gawindx/WinNUT-Client/releases/tag/2.1.7740.35837) before finding the deleting the dll fix in the issues and the issue persists in that version

WinNUT Bug Report
Os Version : 6.2.9200.0
WinNUT Version : 2.1.7740.35837

WinNUT Parameters : 
{
  "Delay": 1000,
  "AutoReconnect": "True",
  "MinInputVoltage": 220,
  "MaxInputVoltage": 245,
  "FrequencySupply": 0,
  "MinInputFrequency": 40,
  "MaxInputFrequency": 60,
  "MinOutputVoltage": 220,
  "MaxOutputVoltage": 245,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 18,
  "MaxBattVoltage": 30,
  "MinimizeToTray": "True",
  "MinimizeOnStart": "True",
  "CloseToTray": "True",
  "StartWithWindows": "True",
  "UseLogFile": "True",
  "Log Level": 3,
  "ShutdownLimitBatteryCharge": 50,
  "ShutdownLimitUPSRemainTime": 600,
  "ImmediateStopAction": "True",
  "Follow_FSD": "False",
  "TypeOfStop": 2,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "True",
  "VerifyUpdateAtStart": "False",
  "DelayBetweenEachVerification": 0,
  "StableOrDevBranch": 0,
  "LastDateVerification": ""
}

Error Message : 
Exception type: System.ArgumentOutOfRangeException
Exception message: Der Text muss kürzer als 64 Zeichen sein.
Parametername: Text
Der tatsächliche Wert war WinNUT-Client - 2.1
In Verbindung gebracht
Online
Batterie OK.
Exception stack trace: 
   bei System.Windows.Forms.NotifyIcon.set_Text(String value)
   bei WinNUT_Client.WinNUT.Event_UpdateNotifyIconStr(String Reason, String Message)
   bei WinNUT_Client.WinNUT.Update_UPS_Data()
   bei WinNUT_Client.WinNUT.Retrieve_UPS_Datas(Object sender, EventArgs e)
   bei System.Windows.Forms.Timer.OnTick(EventArgs e)
   bei System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
05.08.2022 16:20:54 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:54 Pid: 18100 UPS_Device : Process Result With ups.status : VAR nutdev1 ups.status "OL"
05.08.2022 16:20:54 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:54 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
05.08.2022 16:20:54 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:54 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
05.08.2022 16:20:54 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:54 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
05.08.2022 16:20:54 Pid: 18100 UPS_Device : UPS is On Line
05.08.2022 16:20:54 Pid: 18100 WinNUT : Battery Charged
05.08.2022 16:20:54 Pid: 18100 WinNUT : Update Dial
05.08.2022 16:20:54 Pid: 18100 WinNUT : Update Icon
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter Retrieve_UPS_Datas
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter Retrieve_UPS_Data
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With ups.mfr : VAR nutdev1 ups.mfr ""
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With ups.model : VAR nutdev1 ups.model "2000VA"
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving ups.serial
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With ups.firmware : VAR nutdev1 ups.firmware ""
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With battery.charge : VAR nutdev1 battery.charge "100"
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With battery.voltage : VAR nutdev1 battery.voltage "27.60"
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving battery.runtime
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving output.frequency
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With input.frequency : VAR nutdev1 input.frequency "50.0"
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With input.voltage : VAR nutdev1 input.voltage "236.8"
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With output.voltage : VAR nutdev1 output.voltage "236.8"
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With ups.load : VAR nutdev1 ups.load "36"
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Process Result With ups.status : VAR nutdev1 ups.status "OL"
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Enter GetUPSVar
05.08.2022 16:20:55 Pid: 18100 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
05.08.2022 16:20:55 Pid: 18100 UPS_Device : UPS is On Line
05.08.2022 16:20:55 Pid: 18100 WinNUT : Battery Charged
05.08.2022 16:20:55 Pid: 18100 WinNUT : Update Dial
05.08.2022 16:20:55 Pid: 18100 WinNUT : Update Icon

gbakeman, 2022-08-05:
Ok, thank you for posting and clearing that up for me. I'm working on a lot of changes in #11 at the moment for another issue, but maybe between that work and the new German translations, maybe this problem will be fixed. Please stay tuned for a pre-release (make sure you're watching the project) and I'll try to have this issue fixed.

For my own reference, here's what the error reads in English:

Exception type: System.ArgumentOutOfRangeException
Exception message: The text must be shorter than 64 characters.
Parameter name: Text
Actual value was WinNUT client - 2.1
Linked
Online online
Battery OK.
Exception stack trace:
at System.Windows.Forms.NotifyIcon.set-Text(String value)
at WinNUT-Client.WinNUT.Event-UpdateNotifyIconStr(String Reason, String Message)
with WinNUT-Client.WinNUT.Update-UPS-Data()
at WinNUT-Client.WinNUT.Retrieve-UPS-Datas(Object sender, EventArgs e)
at System.Windows.Forms.Timer.OnTick(EventArgs e)
at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
for System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

gbakeman, 2022-08-06:
@MartinKurtz I've just made a pre-release [here](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.2.8253). This issue wasn't the focus of the build, but just in case I'd like you to check and see if the problem still exists. If so, then please make sure you upload the full debug log here.

MartinKurtz, 2022-08-07:
[image]
This basically says that the file it mentions is not found and prompts me to cancel or try again.
This appears after closing the version i have installed and then executing the msi

MartinKurtz, 2022-08-07:
Simply downloading the archive and running it yields this error report:

WinNUT Bug Report
Os Version : 10.0.19044.0
WinNUT Version : 2.2.8253.28732

WinNUT Parameters : 
{
  "Delay": 1000,
  "AutoReconnect": "True",
  "MinInputVoltage": 220,
  "MaxInputVoltage": 245,
  "FrequencySupply": 0,
  "MinInputFrequency": 40,
  "MaxInputFrequency": 60,
  "MinOutputVoltage": 220,
  "MaxOutputVoltage": 245,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 18,
  "MaxBattVoltage": 30,
  "MinimizeToTray": "True",
  "MinimizeOnStart": "True",
  "CloseToTray": "True",
  "StartWithWindows": "True",
  "UseLogFile": "True",
  "Log Level": 3,
  "ShutdownLimitBatteryCharge": 50,
  "ShutdownLimitUPSRemainTime": 600,
  "ImmediateStopAction": "True",
  "Follow_FSD": "False",
  "TypeOfStop": 2,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "True",
  "VerifyUpdateAtStart": "False",
  "DelayBetweenEachVerification": 0,
  "StableOrDevBranch": 0,
  "LastDateVerification": ""
}

Error Message : 
Exception type: System.ArgumentOutOfRangeException
Exception message: Die Länge darf nicht kleiner als 0 (null) sein.
Parametername: length
Exception stack trace: 
   bei System.String.Substring(Int32 startIndex, Int32 length)
   bei WinNUT_Client.WinNUT.Event_UpdateNotifyIconStr(String Reason, String Message) in C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WInNUT_GUI\WinNUT.vb:Zeile 558.
   bei WinNUT_Client.WinNUT.NewRetry_NotifyIcon() in C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WInNUT_GUI\WinNUT.vb:Zeile 501.
   bei WinNUT_Client_Common.UPS_Device.New_RetryEventHandler.Invoke()
   bei WinNUT_Client_Common.UPS_Device.Reconnect_Socket(Object sender, EventArgs e) in C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb:Zeile 239.
   bei System.Windows.Forms.Timer.OnTick(EventArgs e)
   bei System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
07.08.2022 14:53:50 Pid: 9832 Nut_Socket : Authenticated successfully.
07.08.2022 14:53:50 Pid: 9832 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
07.08.2022 14:53:50 Pid: 9832 UPS_Device : TCP Socket Created
07.08.2022 14:53:50 Pid: 9832 UPS_Device : Apply Fallback Value when retrieving ups.serial
07.08.2022 14:53:50 Pid: 9832 UPS_Device : Apply Fallback Value when retrieving battery.capacity
07.08.2022 14:53:50 Pid: 9832 UPS_Device : Apply Fallback Value when retrieving output.frequency.nominal
07.08.2022 14:53:50 Pid: 9832 WinNUT : nutdev1 has indicated it's ready to start sending data.
07.08.2022 14:53:50 Pid: 9832 WinNUT : Connection to Nut Host Established
07.08.2022 14:53:50 Pid: 9832 String : New Log to CB_Current Log : Verbindung zum Nut Host 192.168.2.181:3493 hergestellt
07.08.2022 14:53:50 Pid: 9832 WinNUT : Update Icon
07.08.2022 14:53:50 Pid: 9832 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
In Verbindung gebracht
[... 27 lines cut ...]
07.08.2022 14:53:51 Pid: 9832 UPS_Device : Processing request to disconnect...
07.08.2022 14:53:51 Pid: 9832 Nut_Socket : NutSocket has been Disconnected.
07.08.2022 14:53:51 Pid: 9832 UPS_Device : Socket has reported a Broken event.
07.08.2022 14:53:51 Pid: 9832 WinNUT : Notify user of lost connection
07.08.2022 14:53:51 Pid: 9832 String : New Log to CB_Current Log : Verlorene Verbindung zu 192.168.2.181:3493
07.08.2022 14:53:51 Pid: 9832 WinNUT : Update all informations displayed to empty values
07.08.2022 14:53:51 Pid: 9832 WinNUT : Update Icon
07.08.2022 14:53:51 Pid: 9832 WinNUT : Status Icon Changed
07.08.2022 14:53:51 Pid: 9832 WinNUT : New Icon Value For Systray : 1216
07.08.2022 14:53:51 Pid: 9832 WinNUT : New Icon Value For Gui : 1216
07.08.2022 14:53:51 Pid: 9832 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
Nicht verbunden
07.08.2022 14:53:51 Pid: 9832 WinNUT : Battery Status => Unknown
07.08.2022 14:53:51 Pid: 9832 WinNUT : Disconnected from Nut Host
07.08.2022 14:53:51 Pid: 9832 String : New Log to CB_Current Log : Vom Nut Host getrennt
07.08.2022 14:53:51 Pid: 9832 WinNUT : Status Icon Changed
07.08.2022 14:53:51 Pid: 9832 WinNUT : New Icon Value For Systray : 1344
07.08.2022 14:53:51 Pid: 9832 WinNUT : New Icon Value For Gui : 1344
07.08.2022 14:53:51 Pid: 9832 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
Verlorene Verbindung zu 192.168.2.181:3493
07.08.2022 14:53:51 Pid: 9832 WinNUT : Battery Status => Unknown
07.08.2022 14:53:51 Pid: 9832 WinNUT : Update Icon
07.08.2022 14:53:51 Pid: 9832 UPS_Device : Reconnection Process Started

MartinKurtz, 2022-08-07:
Removing the dll file from the de-DE-folder fixes the issue and allows me to simply run the archive version, which works, at least for now

MartinKurtz, 2022-08-07:
If NotifyStr.Length > 64 Then
            NotifyStr = NotifyStr.Substring(0, (64 - NotifyStr.Length - 4)) & "..."
End If

As far as i can tell the issue is that if you get a number between 59 and 65 the length argument you get becomes 0 or negative, which causes this exeption, so changing this to 

If NotifyStr.Length > 59 Then
            NotifyStr = NotifyStr.Substring(0, (64 - NotifyStr.Length - 4)) & "..."
End If

"should" fix at least this exeption, but i am not sure if this will cause issues somewhere else in the program

MartinKurtz, 2022-08-07:
I would like to compile the project myself, however i do not have any experience with VS, what additions to VS do i need to run and compile this project

gbakeman, 2022-08-07:
> [image] This basically says that the file it mentions is not found and prompts me to cancel or try again. This appears after closing the version i have installed and then executing the msi

Looks like my first release didn't go so well! 😅 I've just made a hotfix release [here](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.2.8254), please download you preferred format again and let me know how it goes.

> If NotifyStr.Length > 64 Then NotifyStr = NotifyStr.Substring(0, (64 - NotifyStr.Length - 4)) & "..." End If
>
> As far as i can tell the issue is that if you get a number between 59 and 65 the length [...]

I looked at the offending line in the code. So from the documentation, the Text field must be 63 characters or less. I don't know what the `.Substring(0, (64 - NotifyStr.Length - 4))` line is trying to accomplish, since we know that the final string must be cut down to 60 characters if we want to attach the three ellipses (...) to the end. I've made a change in the updated release so that it just cuts the NotifyStr to 60 characters before adding the three dots. Please also let me know if that fixes it.

> I would like to compile the project myself, however i do not have any experience with VS, what additions to VS do i need to run and compile this project

Thank you for your interest in contributing! I made a small info section in the README [here](https://github.com/nutdotnet/WinNUT-Client/tree/Dev-2.1#code), I'm curious if you'd like me to extend or add to it to make it more informative. Basically, you must have Visual Studio 2019 (community is fine) to compile the project. It depends on .Net Framework 4.7.2 at the latest, so we can't use Studio 2022 yet. If you want to compile the installer too, there's a link to what you need for that.

Thank you again for your help with fixing these bugs!

MartinKurtz, 2022-08-08:
Okay, the setup.exe ran through perfectly, installed correctly.

The string length issue also does not occur any longer for me at least

MartinKurtz, 2022-08-08:
Since the original Issue has been fixed, do you want to move this conversation about the readme somewhere else?

gbakeman, 2022-08-08:
> Since the original Issue has been fixed, do you want to move this conversation about the readme somewhere else?

Let's do that, thank you again for your feedback and help. I'll close this issue as fixed, then feel free to open another issue regarding the readme.

closed 2022-08-08 by gbakeman