nutdotnet/WinNUT-Client #13: Exception type: System.NullReferenceException (v2.2.8253)

shaarkys, 2022-08-07:
`WinNUT Bug Report
Os Version : 10.0.19043.0
WinNUT Version : 2.2.8253.28732

WinNUT Parameters : 
{
  "Delay": 5000,
  "AutoReconnect": "True",
  "MinInputVoltage": 220,
  "MaxInputVoltage": 250,
  "FrequencySupply": 0,
  "MinInputFrequency": 40,
  "MaxInputFrequency": 60,
  "MinOutputVoltage": 210,
  "MaxOutputVoltage": 250,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 6,
  "MaxBattVoltage": 18,
  "MinimizeToTray": "True",
  "MinimizeOnStart": "True",
  "CloseToTray": "True",
  "StartWithWindows": "True",
  "UseLogFile": "False",
  "Log Level": 0,
  "ShutdownLimitBatteryCharge": 0,
  "ShutdownLimitUPSRemainTime": 0,
  "ImmediateStopAction": "False",
  "Follow_FSD": "False",
  "TypeOfStop": 1,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "True",
  "VerifyUpdateAtStart": "True",
  "DelayBetweenEachVerification": 1,
  "StableOrDevBranch": 0,
  "LastDateVerification": "06.08.2022 21:22:48"
}

Error Message : 
Exception type: System.NullReferenceException
Exception message: Odkaz na objekt není nastaven na instanci objektu.
Exception stack trace: 
   v WinNUT_Client.Update_Gui.VerifyUpdate() v C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WInNUT_GUI\Update_Gui.vb:řádek 64
   v WinNUT_Client.Update_Gui.SetVisibleCore(Boolean value) v C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WInNUT_GUI\Update_Gui.vb:řádek 50
   v System.Windows.Forms.Control.set_Visible(Boolean value)
   v WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e) v C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WInNUT_GUI\WinNUT.vb:řádek 282
   v System.EventHandler.Invoke(Object sender, EventArgs e)
   v System.Windows.Forms.Form.OnLoad(EventArgs e)
   v System.Windows.Forms.Form.OnCreateControl()
   v System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   v System.Windows.Forms.Control.CreateControl()
   v System.Windows.Forms.Control.WmShowWindow(Message& m)
   v System.Windows.Forms.Control.WndProc(Message& m)
   v System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   v System.Windows.Forms.Form.WmShowWindow(Message& m)
   v System.Windows.Forms.Form.WndProc(Message& m)
   v System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   v System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   v System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
07.08.2022 9:51:51 Pid: 16808 WinNUT : Initialisation Globals Variables Complete
07.08.2022 9:51:51 Pid: 16808 WinNUT : Initialisation Params Complete
07.08.2022 9:51:51 Pid: 16808 WinNUT : Loaded Params Complete
07.08.2022 9:51:51 Pid: 16808 WinNUT : Logging is configured.
07.08.2022 9:51:51 Pid: 16808 WinNUT : NotifyIcons Initialised
07.08.2022 9:51:51 Pid: 16808 WinNUT : Windows 10 Toast Notification Available
07.08.2022 9:51:51 Pid: 16808 WinNUT : Windows App Use Light Theme
07.08.2022 9:51:51 Pid: 16808 WinNUT : Windows Use Dark Theme
07.08.2022 9:51:51 Pid: 16808 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
Not Connected
07.08.2022 9:51:51 Pid: 16808 WinNUT : Status Icon Changed
07.08.2022 9:51:51 Pid: 16808 WinNUT : New Icon Value For Systray : 1216
07.08.2022 9:51:51 Pid: 16808 WinNUT : New Icon Value For Gui : 1152
07.08.2022 9:51:51 Pid: 16808 WinNUT : Update Icon at Startup
07.08.2022 9:51:51 Pid: 16808 WinNUT : Run Automatic Update
`

shaarkys, 2022-08-07:
I get this error immediately upon starting. Also can't install via MSI, getting error : System.ObjectModel.dll can't be found (translation).

shaarkys, 2022-08-07:
Btw I was trying to update from 2.0.7722.30975 - also uncompressing archive ZIP didn't help - same error. I'm back on 2.0.7722.30975 , no issues there.

gbakeman, 2022-08-07:
Hi @shaarkys, thank you for your report. It looks like my first release didn't go very well. I've made a hotfix release [here](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.2.8254), please download either the installer or archive and let me know if that fixes the issues.

shaarkys, 2022-08-07:
Hi @gbakeman , this time installed fine, just not error unknown UPS, will open new issue for that ;-(

commit d31c3929 (2022-08-07): Fixed #13, Logger class variable not set
- Fixed an NRE because the Logger object was not set for the class. It now refers to the instance in WinNUT.
closed 2022-08-07 by shaarkys