nutdotnet/WinNUT-Client #14: Uknown UPS (Synology NAS DSM), with latest 2.2.8254 18205

shaarkys, 2022-08-07:
After installed latest WinNUT...

[image]

And I'm getting Unknown UPS reported suddenly, while previous version just worked fine and I did no config. change except of installing latest version. 
Also got this error upon shutdown...but looks like it might be useful also for this kind of error.

`WinNUT Bug Report
Os Version : 6.2.9200.0
WinNUT Version : 2.2.8254.18205

WinNUT Parameters : 
{
  "Delay": 5000,
  "AutoReconnect": "True",
  "MinInputVoltage": 220,
  "MaxInputVoltage": 250,
  "FrequencySupply": 0,
  "MinInputFrequency": 40,
  "MaxInputFrequency": 60,
  "MinOutputVoltage": 210,
  "MaxOutputVoltage": 250,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 6,
  "MaxBattVoltage": 18,
  "MinimizeToTray": "True",
  "MinimizeOnStart": "True",
  "CloseToTray": "True",
  "StartWithWindows": "True",
  "UseLogFile": "False",
  "Log Level": 0,
  "ShutdownLimitBatteryCharge": 0,
  "ShutdownLimitUPSRemainTime": 0,
  "ImmediateStopAction": "False",
  "Follow_FSD": "False",
  "TypeOfStop": 1,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "True",
  "VerifyUpdateAtStart": "True",
  "DelayBetweenEachVerification": 1,
  "StableOrDevBranch": 0,
  "LastDateVerification": "06.08.2022 21:22:48"
}

Error Message : 
Exception type: System.NullReferenceException
Exception message: Odkaz na objekt není nastaven na instanci objektu.
Exception stack trace: 
   v WinNUT_Client.About_Gui.About_Gui_Load(Object sender, EventArgs e)
   v System.Windows.Forms.Form.OnLoad(EventArgs e)
   v System.Windows.Forms.Form.OnCreateControl()
   v System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   v System.Windows.Forms.Control.CreateControl()
   v System.Windows.Forms.Control.WmShowWindow(Message& m)
   v System.Windows.Forms.Control.WndProc(Message& m)
   v System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   v System.Windows.Forms.Form.WmShowWindow(Message& m)
   v System.Windows.Forms.Form.WndProc(Message& m)
   v System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   v System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   v System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
07.08.2022 21:56:27 Pid: 19164 WinNUT : Initialisation Globals Variables Complete
07.08.2022 21:56:27 Pid: 19164 WinNUT : Initialisation Params Complete
07.08.2022 21:56:27 Pid: 19164 WinNUT : Loaded Params Complete
07.08.2022 21:56:27 Pid: 19164 WinNUT : Logging is configured.
07.08.2022 21:56:27 Pid: 19164 WinNUT : NotifyIcons Initialised
07.08.2022 21:56:27 Pid: 19164 WinNUT : Windows 10 Toast Notification Not Available. Too Old Windows Version
07.08.2022 21:56:27 Pid: 19164 WinNUT : Windows App Use Light Theme
07.08.2022 21:56:27 Pid: 19164 WinNUT : Windows Use Dark Theme
07.08.2022 21:56:27 Pid: 19164 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
Not Connected
07.08.2022 21:56:27 Pid: 19164 WinNUT : Status Icon Changed
07.08.2022 21:56:27 Pid: 19164 WinNUT : New Icon Value For Systray : 1216
07.08.2022 21:56:27 Pid: 19164 WinNUT : New Icon Value For Gui : 1152
07.08.2022 21:56:27 Pid: 19164 WinNUT : Update Icon at Startup
07.08.2022 21:56:27 Pid: 19164 WinNUT : Run Automatic Update
07.08.2022 21:56:27 Pid: 19164 Update_Gui : Verify Update
07.08.2022 21:56:27 Pid: 19164 WinNUT : Main Gui Has Focus
07.08.2022 21:56:27 Pid: 19164 WinNUT : Update Icon
07.08.2022 21:56:27 Pid: 19164 WinNUT : Update Icon
07.08.2022 21:56:27 Pid: 19164 WinNUT : Minimize WinNut On Start
07.08.2022 21:56:27 Pid: 19164 WinNUT : Update Icon
07.08.2022 21:56:27 Pid: 19164 WinNUT : Minimize Main Gui To Notify Icon
07.08.2022 21:56:27 Pid: 19164 WinNUT : Main Gui Lose Focus
07.08.2022 21:56:27 Pid: 19164 WinNUT : Update Icon
07.08.2022 21:56:27 Pid: 19164 WinNUT : Auto-connecting to UPS on startup.
07.08.2022 21:56:27 Pid: 19164 UPS_Device : Beginning connection: monuser@10.0.0.11:3493, Name: UPS [AutoReconnect]
07.08.2022 21:56:27 Pid: 19164 Nut_Socket : Attempting TCP socket connection to 10.0.0.11:3493...
07.08.2022 21:56:27 Pid: 19164 Nut_Socket : Connection established and streams ready for 10.0.0.11:3493
07.08.2022 21:56:27 Pid: 19164 Nut_Socket : Attempting authentication...
07.08.2022 21:56:27 Pid: 19164 Nut_Socket : Authenticated successfully.
07.08.2022 21:56:27 Pid: 19164 Nut_Socket : NUT server reports VER: DSM7-1-42464-211216 NETVER: 1.2
07.08.2022 21:56:27 Pid: 19164 UPS_Device : TCP Socket Created
07.08.2022 21:56:27 Pid: 19164 UPS_Device : Given UPS Name is unknown
07.08.2022 21:56:27 Pid: 19164 WinNUT : Update Icon
07.08.2022 21:56:27 Pid: 19164 WinNUT : Status Icon Changed
07.08.2022 21:56:27 Pid: 19164 WinNUT : New Icon Value For Systray : 1216
07.08.2022 21:56:27 Pid: 19164 WinNUT : New Icon Value For Gui : 1216
07.08.2022 21:56:27 Pid: 19164 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
Unknown UPS Name
07.08.2022 21:56:27 Pid: 19164 WinNUT : Cannot Connect : Unknow UPS Name
07.08.2022 21:56:27 Pid: 19164 String : New Log to CB_Current Log : Unknown UPS Name
07.08.2022 21:56:27 Pid: 19164 WinNUT : Update Icon
07.08.2022 21:56:27 Pid: 19164 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
Unknown UPS Name
07.08.2022 21:56:27 Pid: 19164 WinNUT : Cannot Connect : Unknow UPS Name
07.08.2022 21:56:27 Pid: 19164 String : New Log to CB_Current Log : Unknown UPS Name
07.08.2022 21:56:27 Pid: 19164 WinNUT : Completed WinNUT_Shown
07.08.2022 21:56:31 Pid: 19164 WinNUT : Restore Main Gui On Mouse Click Notify Icon
07.08.2022 21:56:31 Pid: 19164 WinNUT : Main Gui Has Focus
07.08.2022 21:56:31 Pid: 19164 WinNUT : Update Icon
07.08.2022 21:56:33 Pid: 19164 WinNUT : Open About Gui From Menu
07.08.2022 21:56:33 Pid: 19164 WinNUT : Main Gui Lose Focus
07.08.2022 21:56:33 Pid: 19164 WinNUT : Update Icon
`

gbakeman, 2022-08-07:
Thank you for checking the new changes. It looks like the About menu crash is because of the same change I made that broke the update menu. I'll need to take a deeper look at the code and make sure I fix this problem everywhere.

For the unknown UPS error, I can't think of any change I would've made to suddenly make this error start happening (I've tested with my own NUT server, distributed on FreeBSD.) Can you confirm what the last version is that works?

shaarkys, 2022-08-07:
I went back to 2.17740.35837 and it's OK....not sure if I can provide some logs ?

[image]

`
07.08.2022 22:28:59 Pid: 9232 Logger : Log file is initialized at c:\WinNUT-Client\Logs\WinNUT-Client-2022-08-07.log
07.08.2022 22:28:59 Pid: 9232 WinNUT : WinNut Preferences Changed
07.08.2022 22:29:00 Pid: 9232 Logger : Log file is initialized at c:\WinNUT-Client\Logs\WinNUT-Client-2022-08-07.log
07.08.2022 22:29:00 Pid: 9232 WinNUT : WinNut Preferences Changed
07.08.2022 22:29:09 Pid: 2644 Logger : Log file is initialized at C:\WinNUT-Client\Logs\WinNUT-Client-2022-08-07.log
07.08.2022 22:29:09 Pid: 2644 WinNUT : Auto-connecting to UPS on startup.
07.08.2022 22:29:09 Pid: 2644 Nut_Socket : Attempting TCP socket connection to 10.0.0.11:3493...
07.08.2022 22:29:09 Pid: 2644 Nut_Socket : Connection established and streams ready for 10.0.0.11:3493
07.08.2022 22:29:09 Pid: 2644 Nut_Socket : Attempting authentication...
07.08.2022 22:29:09 Pid: 2644 Nut_Socket : Authenticated successfully.
07.08.2022 22:29:10 Pid: 2644 Nut_Socket : NUT server reports VER: DSM7-1-42464-211216 NETVER: 1.2
07.08.2022 22:29:10 Pid: 2644 UPS_Device : TCP Socket Created
07.08.2022 22:29:10 Pid: 2644 UPS_Device : Given UPS Name is unknown
07.08.2022 22:29:26 Pid: 15932 Logger : Log file is initialized at C:\WinNUT-Client\Logs\WinNUT-Client-2022-08-07.log
07.08.2022 22:29:26 Pid: 15932 WinNUT : Auto-connecting to UPS on startup.
07.08.2022 22:29:26 Pid: 15932 Nut_Socket : Attempting TCP socket connection to 10.0.0.11:3493...
07.08.2022 22:29:26 Pid: 15932 Nut_Socket : Connection established and streams ready for 10.0.0.11:3493
07.08.2022 22:29:26 Pid: 15932 Nut_Socket : Attempting authentication...
07.08.2022 22:29:26 Pid: 15932 Nut_Socket : Authenticated successfully.
07.08.2022 22:29:26 Pid: 15932 Nut_Socket : NUT server reports VER: DSM7-1-42464-211216 NETVER: 1.2
07.08.2022 22:29:26 Pid: 15932 UPS_Device : TCP Socket Created
07.08.2022 22:29:26 Pid: 15932 UPS_Device : Given UPS Name is unknown

`

gbakeman, 2022-08-07:
That's... odd. You're still getting the UPS Unknown error, but somehow it's still pulling data regardless. It looks like it's real, too...? So either the error is a false negative and the NUT server is sending that response even though it can find the UPS, or the program is displaying old data (??) I guess back then, WinNUT didn't consider an Unknown UPS an error condition and subsequently disconnect. The UPS name on a Synology is actually _ups_, all lower-case so I wonder if that's part of the issue. Try changing your UPS name to lowercase on the latest pre-release version and let me know if that helps.

shaarkys, 2022-08-07:
> UPS name to lowercase on the latest pre-release version and let me know if that helps.

Solved, thank you ;-)

[image]

gbakeman, 2022-08-07:
I think there could be improvements in how an error like this is communicated, maybe a suggestion to double-check the name and try again. For now I'll mark this as an enhancement and leave it open, but I'm not sure if this will make it into WinNUT as I'm investigating rebuilding the application. Do check it out if you want to make any suggestions.

gbakeman, 2022-08-08:
See #16 for the About dialog crash.

gbakeman, 2024-11-11:
Closing this issue as the invalid UPS name problem is now being tracked here:

- https://github.com/nutdotnet/WinNUT-Client/issues/194

closed 2024-11-11 by gbakeman