nutdotnet/WinNUT-Client #17: Crash when opening UPS Variables window

MartinKurtz, 2022-08-08:
This occurs when opening the  UPS Variables window

WinNUT Bug Report
Os Version : 6.2.9200.0
WinNUT Version : 2.2.8254.18205

WinNUT Parameters : 
{
  "Delay": 1000,
  "AutoReconnect": "True",
  "MinInputVoltage": 220,
  "MaxInputVoltage": 245,
  "FrequencySupply": 0,
  "MinInputFrequency": 49,
  "MaxInputFrequency": 51,
  "MinOutputVoltage": 220,
  "MaxOutputVoltage": 245,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 18,
  "MaxBattVoltage": 30,
  "MinimizeToTray": "True",
  "MinimizeOnStart": "True",
  "CloseToTray": "True",
  "StartWithWindows": "True",
  "UseLogFile": "True",
  "Log Level": 3,
  "ShutdownLimitBatteryCharge": 50,
  "ShutdownLimitUPSRemainTime": 600,
  "ImmediateStopAction": "True",
  "Follow_FSD": "False",
  "TypeOfStop": 2,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "True",
  "VerifyUpdateAtStart": "False",
  "DelayBetweenEachVerification": 0,
  "StableOrDevBranch": 0,
  "LastDateVerification": ""
}

Error Message : 
Exception type: System.NullReferenceException
Exception message: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.
Exception stack trace: 
   bei WinNUT_Client.List_Var_Gui.List_Var_Gui_Load(Object sender, EventArgs e)
   bei System.Windows.Forms.Form.OnLoad(EventArgs e)
   bei System.Windows.Forms.Form.OnCreateControl()
   bei System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   bei System.Windows.Forms.Control.CreateControl()
   bei System.Windows.Forms.Control.WmShowWindow(Message& m)
   bei System.Windows.Forms.Control.WndProc(Message& m)
   bei System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   bei System.Windows.Forms.Form.WmShowWindow(Message& m)
   bei System.Windows.Forms.Form.WndProc(Message& m)
   bei System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   bei System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
08.08.2022 12:06:17 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
08.08.2022 12:06:17 Pid: 13128 UPS_Device : UPS is On Line
08.08.2022 12:06:17 Pid: 13128 WinNUT : Updating UPS data for Form.
08.08.2022 12:06:17 Pid: 13128 WinNUT : Battery Charged
08.08.2022 12:06:17 Pid: 13128 WinNUT : Update Dial
08.08.2022 12:06:17 Pid: 13128 WinNUT : Update Icon
08.08.2022 12:06:17 Pid: 13128 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
In Verbindung gebracht
Online
Batteri...
08.08.2022 12:06:17 Pid: 13128 WinNUT : Battery Status => Charged
08.08.2022 12:06:18 Pid: 13128 UPS_Device : Enter Retrieve_UPS_Datas
08.08.2022 12:06:18 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.serial
08.08.2022 12:06:18 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving battery.runtime
08.08.2022 12:06:18 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving output.frequency
08.08.2022 12:06:18 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
08.08.2022 12:06:18 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
08.08.2022 12:06:18 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
08.08.2022 12:06:18 Pid: 13128 UPS_Device : UPS is On Line
08.08.2022 12:06:18 Pid: 13128 WinNUT : Updating UPS data for Form.
08.08.2022 12:06:18 Pid: 13128 WinNUT : Battery Charged
08.08.2022 12:06:18 Pid: 13128 WinNUT : Update Dial
08.08.2022 12:06:18 Pid: 13128 WinNUT : Update Icon
08.08.2022 12:06:18 Pid: 13128 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
In Verbindung gebracht
Online
Batteri...
08.08.2022 12:06:18 Pid: 13128 WinNUT : Battery Status => Charged
08.08.2022 12:06:19 Pid: 13128 UPS_Device : Enter Retrieve_UPS_Datas
08.08.2022 12:06:19 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.serial
08.08.2022 12:06:19 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving battery.runtime
08.08.2022 12:06:19 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving output.frequency
08.08.2022 12:06:19 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
08.08.2022 12:06:19 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
08.08.2022 12:06:19 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
08.08.2022 12:06:19 Pid: 13128 UPS_Device : UPS is On Line
08.08.2022 12:06:19 Pid: 13128 WinNUT : Updating UPS data for Form.
08.08.2022 12:06:19 Pid: 13128 WinNUT : Battery Charged
08.08.2022 12:06:19 Pid: 13128 WinNUT : Update Dial
08.08.2022 12:06:19 Pid: 13128 WinNUT : Update Icon
08.08.2022 12:06:19 Pid: 13128 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
In Verbindung gebracht
Online
Batteri...
08.08.2022 12:06:19 Pid: 13128 WinNUT : Battery Status => Charged
08.08.2022 12:06:20 Pid: 13128 UPS_Device : Enter Retrieve_UPS_Datas
08.08.2022 12:06:20 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.serial
08.08.2022 12:06:20 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving battery.runtime
08.08.2022 12:06:20 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving output.frequency
08.08.2022 12:06:20 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
08.08.2022 12:06:20 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
08.08.2022 12:06:20 Pid: 13128 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
08.08.2022 12:06:20 Pid: 13128 UPS_Device : UPS is On Line
08.08.2022 12:06:20 Pid: 13128 WinNUT : Updating UPS data for Form.
08.08.2022 12:06:20 Pid: 13128 WinNUT : Battery Charged
08.08.2022 12:06:20 Pid: 13128 WinNUT : Update Dial
08.08.2022 12:06:20 Pid: 13128 WinNUT : Update Icon
08.08.2022 12:06:20 Pid: 13128 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.2
In Verbindung gebracht
Online
Batteri...
08.08.2022 12:06:20 Pid: 13128 WinNUT : Battery Status => Charged

gbakeman, 2022-08-09:
Thank you again for your reports @MartinKurtz . Please see the [latest pre-release](https://github.com/nutdotnet/WinNUT-Client/releases) and provide feedback in both issues if the errors have been resolved.

MartinKurtz, 2022-09-30:
Appears to have been fixed

closed 2022-09-30 by MartinKurtz