nutdotnet/WinNUT-Client #18: Increasing battery max in Calibration extends the gauge instead of adjusting scale.

hugalafutro, 2022-08-08:
**WinNUT Version:** v2.1.7740.35837
**Windows OS Version:** 11 x64

**Describe the bug**
I increased the max to 28 as my charged battery sits at around 27.1 and expected the gauge to remain same size and the data on it to "shrink", instead the gauge grew a new notch triggering my imagined ocd how it's all uneven now.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings/Calibration'
2. Increase Battery Max to 28
3. ???
4. See error

**Expected behavior**
The gauge should not change, instead the scale of the data on it should.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 x64 Version 21H2, Build 22000.832, Windows Feature Experience Pack 1000.22000.832.0

hugalafutro, 2022-08-08:
I didn't know the repo moved here, I'll see if it happens with the latest version from here when I get home from work.

gbakeman, 2022-08-08:
I suspect this is still an issue, but thank you for checking anyways. It may be a bug in AGauge but I will try to take a look at this when some of the other crash bugs are fixed.

gbakeman, 2022-11-12:
Just want to update you. I've taken a dive into this issue, and only managed to clean up some code. It looks like the bug exists somewhere in the drawing code of the gauge, but it's a little beyond me at the moment. I was able to get the gauge to draw correctly while tweaking min and max values, I wonder if it may help if you bump up the minimum battery voltage?

hugalafutro, 2022-11-13:
tweaking the value has some effect, but it only disappears when I set min to 20:
[image]

19 and lower makes it look like this:
[image]

gbakeman, 2022-11-28:
Just an update and more of a note to myself after looking into this more: It seems like this problem occurs when `ScaleLinesMajorStepValue` is an odd number of 3 or higher, and the Min or MaxValue is an odd number. Continuing the search.

hugalafutro, 2023-10-20:
Hi I've compltely forgot about this issue, but it the latest prerelease ([2.2.8693](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.2.8693)) it seems to work ok:
[image]

gbakeman, 2023-10-20:
I can't imagine what would've done it; I've spent several hours already trying to get to the bottom of the issue, and could have sworn it was still there. When I jump back into this project, let me test it some more to see if I can't reproduce it.

closed 2024-07-23 by gbakeman

gbakeman, 2024-07-23:
Thanks for your patience on this one, I believe it's fixed now! To summarize, the issue occurs when someone tries to enter a Min or MaxValue which is not divisible by the number of spaces between a major tick mark on the gauge. As far as WinNUT is concerned, as long as you keep the total gauge range (Min and Max) an even number, it should be fine. To help with this, I've made the gauge only display one major tick mark range when an invalid number is entered, like below:

[image]

Hope this helps!

hugalafutro, 2024-07-24:
There was no rush, thanks for finding the time to fix a cosmetic issue!

PR #168 (2024-07-23): AGauge - fix tick mark bug
Updating AGauge dependency, which includes a fix for the tick mark bug (tick marks extending beyond end of gauge when the combination of MinValue, MaxValue and TickMarksMajorStepValue are not multiples of one another).

Correcting references throughout code to new AGauge namespace.

Closes #18.