nutdotnet/WinNUT-Client #21: WinNut crashes as startup v2.2.8255.34401

UltraHKR, 2022-08-12:
Note: NUT is running on OpenWRT, and reports fine....

[image]

inNUT Bug Report
Os Version : 6.2.9200.0
WinNUT Version : 2.2.8255.34401

WinNUT Parameters : 
{
  "Delay": 30000,
  "AutoReconnect": true,
  "MinInputVoltage": 80,
  "MaxInputVoltage": 145,
  "FrequencySupply": 1,
  "MinInputFrequency": 56,
  "MaxInputFrequency": 64,
  "MinOutputVoltage": 108,
  "MaxOutputVoltage": 132,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 24,
  "MaxBattVoltage": 28,
  "MinimizeToTray": true,
  "MinimizeOnStart": true,
  "CloseToTray": true,
  "StartWithWindows": true,
  "UseLogFile": true,
  "Log Level": 3,
  "ShutdownLimitBatteryCharge": 0,
  "ShutdownLimitUPSRemainTime": 0,
  "ImmediateStopAction": true,
  "Follow_FSD": false,
  "TypeOfStop": 2,
  "DelayToShutdown": 0,
  "AllowExtendedShutdownDelay": false,
  "ExtendedShutdownDelay": 0,
  "VerifyUpdate": true,
  "VerifyUpdateAtStart": true,
  "DelayBetweenEachVerification": 1,
  "StableOrDevBranch": 0,
  "LastDateVerification": "10/8/2022 12:55:22"
}

Error Message : 
Exception type: System.Exception
Exception message: ups.mfr : ERR DATA-STALE
Exception stack trace: 
   at WinNUT_Client_Common.UPS_Device.GetUPSVar(String varName, String UPSName, Object Fallback_value)
   at WinNUT_Client_Common.UPS_Device.GetUPSProductInfo()
   at WinNUT_Client_Common.UPS_Device.Connect_UPS()
   at WinNUT_Client_Common.UPS_Device.Reconnect_Socket(Object sender, EventArgs e)
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
12/8/2022 11:39:14 Pid: 18604 WinNUT : Status Icon Changed
12/8/2022 11:39:14 Pid: 18604 WinNUT : New Icon Value For Systray : 1344
12/8/2022 11:39:14 Pid: 18604 WinNUT : New Icon Value For Gui : 1344
12/8/2022 11:39:14 Pid: 18604 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Lost Connect To 10.10.10.50:3493
12/8/2022 11:39:14 Pid: 18604 WinNUT : Battery Status => Unknown
12/8/2022 11:39:14 Pid: 18604 WinNUT : Update Icon
12/8/2022 11:39:14 Pid: 18604 UPS_Device : Reconnection Process Started
12/8/2022 11:39:44 Pid: 18604 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Reconnection In Progress
Try 1 of 30
[... 222 lines cut ...]
12/8/2022 11:45:14 Pid: 18604 String : New Log to CB_Current Log : Lost Connect To 10.10.10.50:3493
12/8/2022 11:45:14 Pid: 18604 WinNUT : Status Icon Changed
12/8/2022 11:45:14 Pid: 18604 WinNUT : New Icon Value For Systray : 1344
12/8/2022 11:45:14 Pid: 18604 WinNUT : New Icon Value For Gui : 1344
12/8/2022 11:45:14 Pid: 18604 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Lost Connect To 10.10.10.50:3493
12/8/2022 11:45:14 Pid: 18604 WinNUT : Battery Status => Unknown
12/8/2022 11:45:14 Pid: 18604 WinNUT : Update Icon
12/8/2022 11:45:14 Pid: 18604 UPS_Device : Reconnection Process Started
12/8/2022 11:45:44 Pid: 18604 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Reconnection In Progress
Try 1 of 30
12/8/2022 11:45:44 Pid: 18604 WinNUT : Update Icon
12/8/2022 11:45:44 Pid: 18604 UPS_Device : Try Reconnect 1 / 30
12/8/2022 11:45:44 Pid: 18604 String : New Log to CB_Current Log : Try Reconnect 1 / 30
12/8/2022 11:45:44 Pid: 18604 UPS_Device : Beginning connection: @10.10.10.50:3493, Name: CDP [AutoReconnect]
12/8/2022 11:45:44 Pid: 18604 Nut_Socket : Attempting TCP socket connection to 10.10.10.50:3493...
12/8/2022 11:45:44 Pid: 18604 Nut_Socket : Connection established and streams ready for 10.10.10.50:3493
12/8/2022 11:45:44 Pid: 18604 Nut_Socket : Attempting authentication...
12/8/2022 11:45:44 Pid: 18604 Nut_Socket : Authenticated successfully.
12/8/2022 11:45:44 Pid: 18604 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
12/8/2022 11:45:44 Pid: 18604 UPS_Device : TCP Socket Created
12/8/2022 11:45:44 Pid: 18604 UPS_Device : Error Result On Retrieving  ups.mfr : ERR DATA-STALE

gbakeman, 2022-08-12:
I've never seen a DATA-STALE error before. Either way, I don't think the program should have to halt when receiving that error. We should maybe just fill the affected field with that message (DATA-STALE) or report to the user and move on.

I will take a look at this when possible. Thank you for the report.

closed 2022-09-09 by gbakeman

gbakeman, 2022-09-09:
@UltraHKR Please see the latest release and let me know if this is still an issue!
https://github.com/nutdotnet/WinNUT-Client/releases

PR #25 (2022-09-09): Large changes to Socket/UPS, DATA-STALE error handling
- New Nut_Exception constructor to allow specifying NUTResponse instead of _Exception_Value
- Shifted some responsibility of error handling out of Socket class and into UPS class. These are some major changes and need testing.
- Merged Create_Socket code into Connect code
- Renamed and then commented-out IsKnownUPS function. Feels like this should be a part of the UPS class, but regardless, we'll know if there's an invalid UPS the moment a command is run for one.
- Simplified & renamed Query_Desc function
- Added a few events to the UPS cl [... cut]