nutdotnet/WinNUT-Client #24: 120 volt grid and multiple ups?

wx4cb, 2022-09-01:
[image]

Running the lastest downloaded version from the site, it connects fine, but we run on 120 here in the states, is there a way to set it so that it doesn't think it's on a 220 volt grid?

Also, i monitor multiple ups' would be nice to be able to select which ups onthe system i'm wanting to look at :D

wx4cb, 2022-09-01:
i think a better way to do the scaling would be to set the voltage ranges to be whatever the max/min voltage is set in the UPS. YES I KNOW you can do it in the "calibration" page, but would be an idea to at least start on a fresh install by reading the values that the ups is sending.

for example, the list vars of the one 1500watt ups i have connected at the minute is below, so for example, if the nominal battery voltage is 24 volts, you should scale it so that it's a min of say 20 and a max of say 28 (should never get to those levels).

Likewise for the mains voltage, (and i have no idea why it says 142 volts for the output cus it's not LOL) but again set a 20 volt swing either side of the nominal voltage.

```
BATTERY VOLTAGE
VAR pve720_1500 battery.voltage "24.0"
VAR pve720_1500 battery.voltage.nominal "24"

Incoming mains voltage
VAR pve720_1500 input.voltage "118.0"
VAR pve720_1500 input.voltage.nominal "120"
VAR pve720_1500 output.voltage "142.0"
```

```
FULL output
VAR pve720_1500 battery.charge "100"
VAR pve720_1500 battery.charge.low "10"
VAR pve720_1500 battery.charge.warning "20"
VAR pve720_1500 battery.mfr.date "CPS"
VAR pve720_1500 battery.runtime "840"
VAR pve720_1500 battery.runtime.low "300"
VAR pve720_1500 battery.type "PbAcid"
VAR pve720_1500 battery.voltage "24.0"
VAR pve720_1500 battery.voltage.nominal "24"
VAR pve720_1500 device.mfr "CPS"
VAR pve720_1500 device.model "LX1500GU"
VAR pve720_1500 device.serial "QAKGN2000988"
VAR pve720_1500 device.type "ups"
VAR pve720_1500 driver.name "usbhid-ups"
VAR pve720_1500 driver.parameter.bus "001"
VAR pve720_1500 driver.parameter.pollfreq "30"
VAR pve720_1500 driver.parameter.pollinterval "2"
VAR pve720_1500 driver.parameter.port "auto"
VAR pve720_1500 driver.parameter.product "LX1500GU"
VAR pve720_1500 driver.parameter.productid "0501"
VAR pve720_1500 driver.parameter.serial "QAKGN2000988"
VAR pve720_1500 driver.parameter.synchronous "no"
VAR pve720_1500 driver.parameter.vendor "CPS"
VAR pve720_1500 driver.parameter.vendorid "0764"
VAR pve720_1500 driver.version "2.7.4"
VAR pve720_1500 driver.version.data "CyberPower HID 0.4"
VAR pve720_1500 driver.version.internal "0.41"
VAR pve720_1500 input.voltage "118.0"
VAR pve720_1500 input.voltage.nominal "120"
VAR pve720_1500 output.voltage "142.0"
VAR pve720_1500 ups.beeper.status "enabled"
VAR pve720_1500 ups.delay.shutdown "20"
VAR pve720_1500 ups.delay.start "30"
VAR pve720_1500 ups.load "46"
VAR pve720_1500 ups.mfr "CPS"
VAR pve720_1500 ups.model "LX1500GU"
VAR pve720_1500 ups.productid "0501"
VAR pve720_1500 ups.realpower.nominal "900"
VAR pve720_1500 ups.serial "QAKGN2000988"
VAR pve720_1500 ups.status "OL"
VAR pve720_1500 ups.test.result "No test initiated"
VAR pve720_1500 ups.timer.shutdown "-60"
VAR pve720_1500 ups.timer.start "-60"
VAR pve720_1500 ups.vendorid "0764"

```

gbakeman, 2022-09-01:
Hi @wx4cb ,

Thank you for making the report. I agree that WinNUT could be more intelligent about how it reads parameters from the UPS device and applies them to the UI. Just so I understand correctly regarding ranges, you're suggesting we should read from the nominal values and apply a +/- range based on that? I think that would make sense - maybe the range can be adjusted by the user, but otherwise I think it should be automatic by default. I think this will be a fairly trivial change to make so I'll rename this issue to be the primary subject.

As for monitoring multiple UPSs, I agree this would be a nice feature. I don't think that will be within the scope of this specific project though, since WinNUT is showing its age and may be in need of a redesign. I'm planning on what a WinNUT 2.0 would look like over at https://github.com/nutdotnet/Coco.Nut/discussions. Feel free to join the discussions and submit your own ideas.

wx4cb, 2022-09-01:
yea pretty much.

usually it's +/- 10 percent so you can usually just use that as a starter from the "nominal" voltage reported from the UPS

```
Wikipedia:
Most of the world population ([Europe](https://en.wikipedia.org/wiki/Europe), [Africa](https://en.wikipedia.org/wiki/Africa), [Asia](https://en.wikipedia.org/wiki/Asia), [Australia](https://en.wikipedia.org/wiki/Australia), [New Zealand](https://en.wikipedia.org/wiki/New_Zealand), and much of [South America](https://en.wikipedia.org/wiki/South_America)) use a supply that is within 6% of 230 V. In the [United Kingdom](https://en.wikipedia.org/wiki/United_Kingdom) and Australia[[5]](https://en.wikipedia.org/wiki/Mains_electricity#cite_note-5) the nominal supply voltage is 230 V +10%/−6% to accommodate the fact that most transformers are in fact still set to 240 V. The 230 V standard has become widespread so that 230 V equipment can be used in most parts of the world with the aid of an [adapter](https://en.wikipedia.org/wiki/Adapter) or a change to the equipment's plug to the standard for the specific country. The [United States](https://en.wikipedia.org/wiki/United_States) and [Canada](https://en.wikipedia.org/wiki/Canada) use a supply voltage of 120 volts ± 6%. [Japan](https://en.wikipedia.org/wiki/Japan), [Taiwan](https://en.wikipedia.org/wiki/Taiwan), [Saudi Arabia](https://en.wikipedia.org/wiki/Saudi_Arabia), [North America](https://en.wikipedia.org/wiki/North_America), [Central America](https://en.wikipedia.org/wiki/Central_America) and some parts of northern South America use a voltage between 100 V and 127 V. However, most of the households in Japan equips [split-phase electric power](https://en.wikipedia.org/wiki/Split-phase_electric_power) like USA, which can supply 200V by using reversed phase at the same time. [Brazil](https://en.wikipedia.org/wiki/Brazil) is unusual in having both 127 V and 220 V systems at 60 Hz and also permitting interchangeable plugs and sockets.[[6]](https://en.wikipedia.org/wiki/Mains_electricity#cite_note-6) [Saudi Arabia](https://en.wikipedia.org/wiki/Saudi_Arabia) and [Mexico](https://en.wikipedia.org/wiki/Mexico) have mixed voltage systems; in residential and light commercial buildings both countries use 127 volts, with 220 volts in commercial and industrial applications. The Saudi government approved plans in August 2010 to transition the country to a totally 230/400 volts system,[[7]](https://en.wikipedia.org/wiki/Mains_electricity#cite_note-7) but Mexico has no plans to transition.
```