nutdotnet/WinNUT-Client #27: Sleep Mode not working

DravenSA, 2022-09-29:
So i have been going through the old repository and this issue was a big one for a lot of windows 10 users, but i never did work out the solution.
I was then told to come to here and try the pre-releases and report back.
Either i am stupid, or my setup is not working correctly, but I have tried several pre releases and nothing will put my PC to sleep, I don't even get the dialog box with the 15 second timer.
Everything connects, all gauges work, when it goes to battery it turns yellow, but the ID status at the bottom never changes, 
I have tired the install files and uninstalled each time from add remove programs, I have tried the zip files and just extracting them to a folder and running it as a standalone. Nothing is working for me.

Is there something i need to maybe need to clear from the registry or something?
I notice every time i run an new version, even if the pervious one was uninstalled, it remembers all my setting like IP address, port number ups name password etc, so I can only assume that that SOMETHING is still in the machine even after an uninstall.

What am I doing wrong
WinNUT-Client on Windows 10 Ver 21H2
NUTserver on Pi 3b+

gbakeman, 2022-09-29:
Hi @DravenSA , thank you for posting here. I thought power mode bugs were fixed, but it sounds like there's more work to do. Settings are currently stored in the registry and uninstalling doesn't remove them. Ideally old settings shouldn't cause a bug like this, so let's get a better understanding of what's going on and maybe I can put out another patch.

Would you please make sure you have the latest pre-release from here, then run it and try to reproduce the problem? After that, generate a bug report from the file menu and upload or paste it here.

Thank you!

Related #2

DravenSA, 2022-09-29:
Thanks for the response, sorry it took so long to get this done, But we have a thing in South Africa called Load Shedding , the power company kills our electricity to our homes for 2- 4 hours at least twice a day right now, maddens I know. 
So you can see how important it is that I can get my PC to sleep, because of this.
anyway this is what I think you want? I am not sure and there was nowhere i saw bug report from the file menu

```
2022/09/29 15:55:59 Pid: 12492 Logger : Log file is initialized at C:\WinNUT-Client\Logs\WinNUT-Client-2022-09-29.log
2022/09/29 15:55:59 Pid: 12492 WinNUT : WinNut Preferences Applied.
2022/09/29 15:55:59 Pid: 12492 WinNUT : Auto-connecting to UPS on startup.
2022/09/29 15:55:59 Pid: 12492 Nut_Socket : Attempting TCP socket connection to 192.168.0.59:3493...
2022/09/29 15:55:59 Pid: 12492 Nut_Socket : Connection established and streams ready for 192.168.0.59:3493
2022/09/29 15:55:59 Pid: 12492 Nut_Socket : Attempting authentication...
2022/09/29 15:55:59 Pid: 12492 Nut_Socket : Authenticated successfully.
2022/09/29 15:55:59 Pid: 12492 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
2022/09/29 15:55:59 Pid: 12492 UPS_Device : TCP Socket Created
2022/09/29 15:55:59 Pid: 12492 WinNUT : Connection to Nut Host Established
2022/09/29 15:56:04 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:09 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:14 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:19 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:24 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:29 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:34 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:39 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:44 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:49 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:54 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:56:59 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:04 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:09 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:14 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:19 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:24 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:29 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:34 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:39 Pid: 12492 UPS_Device : UPS is On Line
2022/09/29 15:57:43 Pid: 12492 Logger : Log file is initialized at C:\WinNUT-Client\Logs\WinNUT-Client-2022-09-29.log
2022/09/29 15:57:43 Pid: 12492 WinNUT : Connection parameters Changed. Force Reconnect
2022/09/29 15:57:43 Pid: 12492 UPS_Device : Processing request to disconnect...
2022/09/29 15:57:43 Pid: 12492 Nut_Socket : NutSocket raising Disconnected event.
2022/09/29 15:57:43 Pid: 12492 UPS_Device : NutSocket raised Disconnected event.
2022/09/29 15:57:43 Pid: 12492 WinNUT : Update all informations displayed to empty values
2022/09/29 15:57:43 Pid: 12492 WinNUT : Update Icon
2022/09/29 15:57:43 Pid: 12492 WinNUT : Status Icon Changed
2022/09/29 15:57:43 Pid: 12492 WinNUT : New Icon Value For Systray : 1216
2022/09/29 15:57:43 Pid: 12492 WinNUT : New Icon Value For Gui : 1216
2022/09/29 15:57:43 Pid: 12492 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
2022/09/29 15:57:43 Pid: 12492 WinNUT : Battery Status => Unknown
2022/09/29 15:57:43 Pid: 12492 WinNUT : Disconnected from Nut Host
2022/09/29 15:57:43 Pid: 12492 String : New Log to CB_Current Log : Disconnected from Nut Host
2022/09/29 15:57:43 Pid: 12492 Nut_Socket : NutSocket has been Disconnected.
2022/09/29 15:57:43 Pid: 12492 UPS_Device : Beginning connection: upsmaster@192.168.0.59:3493, Name: mainpc [AutoReconnect]
2022/09/29 15:57:43 Pid: 12492 Nut_Socket : Attempting TCP socket connection to 192.168.0.59:3493...
2022/09/29 15:57:43 Pid: 12492 Nut_Socket : Connection established and streams ready for 192.168.0.59:3493
2022/09/29 15:57:43 Pid: 12492 Nut_Socket : Attempting authentication...
2022/09/29 15:57:43 Pid: 12492 Nut_Socket : Authenticated successfully.
2022/09/29 15:57:43 Pid: 12492 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
2022/09/29 15:57:43 Pid: 12492 UPS_Device : TCP Socket Created
2022/09/29 15:57:43 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/09/29 15:57:43 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/09/29 15:57:43 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/09/29 15:57:43 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/09/29 15:57:43 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving battery.capacity
2022/09/29 15:57:43 Pid: 12492 WinNUT : mainpc has indicated it's ready to start sending data.
2022/09/29 15:57:43 Pid: 12492 WinNUT : Update Icon
2022/09/29 15:57:43 Pid: 12492 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
2022/09/29 15:57:43 Pid: 12492 WinNUT : Connection to Nut Host Established
[... 456 lines cut ...]
2022/09/29 15:59:20 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/09/29 15:59:20 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/09/29 15:59:20 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
2022/09/29 15:59:20 Pid: 12492 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
[... cut ...]
2022/09/29 15:59:20 Pid: 12492 WinNUT : Update Dial
2022/09/29 15:59:20 Pid: 12492 WinNUT : Update Icon
2022/09/29 15:59:20 Pid: 12492 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/09/29 15:59:20 Pid: 12492 WinNUT : Battery Status => Charging
2022/09/29 15:59:22 Pid: 12492 Pref_Gui : Setting LogControl statuses.
2022/09/29 15:59:24 Pid: 12492 Pref_Gui : Show LogFile

```

Hope it help?

Oh and i am using the pre release WinNUT-Setup-v2.2.8303.msi
Or should i be using something else?

DravenSA, 2022-10-01:
Have my logs not helped? or have i posted the wrong thing?

gbakeman, 2022-10-01:
Sorry about that, I've been trying to setup a CI pipeline for the project but very much appreciate your log.

So it looks like preferences are changed a few times, then the UPS goes on battery approximately around 15:58:15, does that sound right?

I'll need to take a look at the code. Some fairly major changes have occurred around the area of detecting battery events, so that could be what's happening here. I'll try to get to it in the next couple of days and keep you updated.

DravenSA, 2022-10-01:
According to the logs, yeah, looks like it. In the next 2 - 15 minutes we will have the power turned off due to our wonderful power provider, then we will switch over to a generator, I will post the logs again, but i will look for a smaller section for you.

DravenSA, 2022-10-01:
So as predicted, the power went out.

Below is the logs from just before it went off until i get the generator on, so it is a little long, but you can see that the values only really seams to change at switch over.

There is nothing i see in the logs to say battery percentage, which is what i would like my setting to be rather than time.

anyway here are the logs
```
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/01 18:03:49 Pid: 15500 WinNUT : Battery Status => Charged
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
2022/10/01 18:03:54 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
2022/10/01 18:03:54 Pid: 15500 UPS_Device : UPS is On Line
2022/10/01 18:03:54 Pid: 15500 WinNUT : Updating UPS data for Form.
2022/10/01 18:03:54 Pid: 15500 WinNUT : Battery Charged
2022/10/01 18:03:54 Pid: 15500 WinNUT : Update Dial
2022/10/01 18:03:54 Pid: 15500 WinNUT : Update Icon
2022/10/01 18:03:54 Pid: 15500 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/01 18:03:54 Pid: 15500 WinNUT : Battery Status => Charged
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
2022/10/01 18:03:59 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
2022/10/01 18:03:59 Pid: 15500 UPS_Device : UPS is On Battery
2022/10/01 18:03:59 Pid: 15500 WinNUT : Updating UPS data for Form.
2022/10/01 18:03:59 Pid: 15500 WinNUT : Battery Charged
2022/10/01 18:03:59 Pid: 15500 WinNUT : Update Dial
2022/10/01 18:03:59 Pid: 15500 WinNUT : Update Icon
2022/10/01 18:03:59 Pid: 15500 WinNUT : Status Icon Changed
2022/10/01 18:03:59 Pid: 15500 WinNUT : New Icon Value For Systray : 1104
2022/10/01 18:03:59 Pid: 15500 WinNUT : New Icon Value For Gui : 1104
2022/10/01 18:03:59 Pid: 15500 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/01 18:03:59 Pid: 15500 WinNUT : Battery Status => Charging
2022/10/01 18:04:04 Pid: 15500 WinNUT : Restore Main Gui On Mouse Click Notify Icon
2022/10/01 18:04:04 Pid: 15500 WinNUT : Main Gui Has Focus
2022/10/01 18:04:04 Pid: 15500 WinNUT : Update Icon
2022/10/01 18:04:04 Pid: 15500 WinNUT : Main Gui Lose Focus
2022/10/01 18:04:04 Pid: 15500 WinNUT : Update Icon
2022/10/01 18:04:04 Pid: 15500 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/01 18:04:04 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/01 18:04:04 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/01 18:04:04 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/01 18:04:04 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/01 18:04:04 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/01 18:04:04 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving input.frequency
[... 872 lines cut ...]
2022/10/01 18:07:29 Pid: 15500 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/01 18:07:29 Pid: 15500 WinNUT : Battery Status => Charging
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
2022/10/01 18:07:34 Pid: 15500 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
2022/10/01 18:07:34 Pid: 15500 UPS_Device : UPS is On Line
2022/10/01 18:07:34 Pid: 15500 WinNUT : Updating UPS data for Form.
2022/10/01 18:07:34 Pid: 15500 WinNUT : Battery Charged
2022/10/01 18:07:34 Pid: 15500 WinNUT : Update Dial
2022/10/01 18:07:34 Pid: 15500 WinNUT : Update Icon
2022/10/01 18:07:34 Pid: 15500 WinNUT : NotifyIcon Text => 
```

Hope it helps a little more

gbakeman, 2022-10-01:
I'm seeing some conflicting information being presented at exactly the same time. I think the debugging messages could be reworked a little, and I'm also taking the chance to upgrade the UI. Stay tuned & thank you for the help.

gbakeman, 2022-10-10:
Apologies for the long wait on this - I was busy working on a build automation system to make testing a little easier.

I've made some fairly large changes and I'd like to ask you to be my Guinea Pig. It turns out that the stop actions system was disabled quite some time ago, so I've also turned it back on. Please download the latest debug build [here ](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3216319243)and let me know how it goes.

DravenSA, 2022-10-10:
Hi there, Unfortunately still not working.
I set the Shutdown (type of stop = Sleep)  if the battery was lower than 95% (to save the battery) when it registered 71%, there was no pop up to say it was shutting down or sleeping. Bellow is the log
```
2022/10/10 16:31:45 Pid: 21552 Logger : Log file is initialized at E:\Desktop\New folder (6)\Data\Logs\WinNUT-Client-2022-10-10.log
2022/10/10 16:31:45 Pid: 21552 WinNUT : Parameter Dial Input Voltage Need to be Updated
2022/10/10 16:31:45 Pid: 21552 WinNUT : Parameter Dial Input Voltage Updated
2022/10/10 16:31:45 Pid: 21552 WinNUT : Parameter Dial Input Frequency Need to be Updated
2022/10/10 16:31:45 Pid: 21552 WinNUT : Parameter Dial Input Frequency Updated
2022/10/10 16:31:45 Pid: 21552 WinNUT : Parameter Dial Output Voltage Need to be Updated
2022/10/10 16:31:45 Pid: 21552 WinNUT : Parameter Dial Output Voltage Updated
2022/10/10 16:31:45 Pid: 21552 WinNUT : Parameter Dial Voltage Battery Need to be Updated
2022/10/10 16:31:45 Pid: 21552 WinNUT : Parameter Dial Voltage Battery Updated
2022/10/10 16:31:45 Pid: 21552 WinNUT : WinNut Preferences Applied.
2022/10/10 16:31:45 Pid: 21552 String : New Log to CB_Current Log : WinNut Preferences Changed
2022/10/10 16:31:45 Pid: 21552 WinNUT : Loaded Params Complete
2022/10/10 16:31:45 Pid: 21552 WinNUT : NotifyIcons Initialised
2022/10/10 16:31:45 Pid: 21552 WinNUT : Windows 10 Toast Notification Available
2022/10/10 16:31:45 Pid: 21552 WinNUT : Windows App Use Dark Theme
2022/10/10 16:31:45 Pid: 21552 WinNUT : Windows Use Dark Theme
2022/10/10 16:31:45 Pid: 21552 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
2022/10/10 16:31:45 Pid: 21552 WinNUT : Status Icon Changed
2022/10/10 16:31:45 Pid: 21552 WinNUT : New Icon Value For Systray : 1216
2022/10/10 16:31:45 Pid: 21552 WinNUT : New Icon Value For Gui : 1216
2022/10/10 16:31:45 Pid: 21552 WinNUT : Update Icon at Startup
2022/10/10 16:31:45 Pid: 21552 WinNUT : Run Automatic Update
2022/10/10 16:31:45 Pid: 21552 Update_Gui : Verify Update
2022/10/10 16:31:45 Pid: 21552 WinNUT : Main Gui Has Focus
2022/10/10 16:31:45 Pid: 21552 WinNUT : Update Icon
2022/10/10 16:31:45 Pid: 21552 WinNUT : Update Icon
2022/10/10 16:31:45 Pid: 21552 WinNUT : Minimize WinNut On Start
2022/10/10 16:31:45 Pid: 21552 WinNUT : Update Icon
2022/10/10 16:31:45 Pid: 21552 WinNUT : Minimize Main Gui To Notify Icon
2022/10/10 16:31:45 Pid: 21552 WinNUT : Main Gui Lose Focus
2022/10/10 16:31:45 Pid: 21552 WinNUT : Update Icon
2022/10/10 16:31:45 Pid: 21552 WinNUT : Auto-connecting to UPS on startup.
2022/10/10 16:31:45 Pid: 21552 UPS_Device : Beginning connection: upsmaster@192.168.0.59:3493, Name: mainpc [AutoReconnect]
2022/10/10 16:31:45 Pid: 21552 Nut_Socket : Attempting TCP socket connection to 192.168.0.59:3493...
2022/10/10 16:31:45 Pid: 21552 Nut_Socket : Connection established and streams ready for 192.168.0.59:3493
2022/10/10 16:31:45 Pid: 21552 Nut_Socket : Attempting authentication...
2022/10/10 16:31:45 Pid: 21552 Nut_Socket : Authenticated successfully.
2022/10/10 16:31:45 Pid: 21552 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
2022/10/10 16:31:45 Pid: 21552 UPS_Device : TCP Socket Created
2022/10/10 16:31:45 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/10 16:31:45 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/10 16:31:45 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/10 16:31:45 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/10 16:31:45 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving battery.capacity
2022/10/10 16:31:45 Pid: 21552 WinNUT : mainpc has indicated it's ready to start sending data.
2022/10/10 16:31:45 Pid: 21552 WinNUT : Update Icon
2022/10/10 16:31:45 Pid: 21552 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
2022/10/10 16:31:45 Pid: 21552 WinNUT : Connection to Nut Host Established
2022/10/10 16:31:45 Pid: 21552 String : New Log to CB_Current Log : Connection to Nut Host 192.168.0.59:3493 Established
2022/10/10 16:31:45 Pid: 21552 WinNUT : Completed WinNUT_Shown
2022/10/10 16:31:50 Pid: 21552 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/10 16:31:50 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/10 16:31:50 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/10/10 16:31:50 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
2022/10/10 16:31:50 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
2022/10/10 16:31:50 Pid: 21552 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
2022/10/10 16:31:50 Pid: 21552 UPS_Device : UPS statuses have CHANGED, updating...
2022/10/10 16:31:50 Pid: 21552 WinNUT : Handling new UPS status(es)...
2022/10/10 16:31:50 Pid: 21552 WinNUT : Finished handling new UPS statuses: None OL 
2022/10/10 16:31:50 Pid: 21552 WinNUT : Updating UPS data for Form.
2022/10/10 16:31:50 Pid: 21552 WinNUT : Retrieve UPS Informations
2022/10/10 16:31:50 Pid: 21552 WinNUT : Updating battery icons based on charge percent: 100%
2022/10/10 16:31:50 Pid: 21552 WinNUT : Calculated estimated remaining battery time: 02:22:45
[... 446 lines cut ...]
2022/10/10 16:33:24 Pid: 21552 WinNUT : Main Gui Lose Focus
2022/10/10 16:33:24 Pid: 21552 WinNUT : Update Icon
2022/10/10 16:33:24 Pid: 21552 Pref_Gui : Pref Gui Opened.
[... cut ...]
2022/10/10 16:33:27 Pid: 21552 WinNUT : Updating battery icons based on charge percent: 71%
2022/10/10 16:33:27 Pid: 21552 WinNUT : Calculated estimated remaining battery time: 01:35:05
2022/10/10 16:33:27 Pid: 21552 WinNUT : Update Dial
2022/10/10 16:33:27 Pid: 21552 WinNUT : Update Icon
2022/10/10 16:33:27 Pid: 21552 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/10 16:33:27 Pid: 21552 WinNUT : Battery Status => Charging
2022/10/10 16:33:28 Pid: 21552 Pref_Gui : Show LogFile
****
```

gbakeman, 2022-10-11:
I think I see the problem - it looks like the code to activate the shutdown event would never actually be called. I've moved some things around, please download [build 8319](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3223327811) and let me know how that goes. (I'll try to setup my own NUT server so you don't have to test as much...)

DravenSA, 2022-10-11:
Ok so here we have new information 
The pop up appeared
[image]

and so did this
[image]
so i generated the crash report
```
WinNUT Bug Report
Os Version : 10.0.19044.0
WinNUT Version : 2.2.8319.1461

WinNUT Parameters : 
{
  "Delay": 5000,
  "AutoReconnect": "True",
  "MinInputVoltage": 210,
  "MaxInputVoltage": 270,
  "FrequencySupply": 0,
  "MinInputFrequency": 40,
  "MaxInputFrequency": 60,
  "MinOutputVoltage": 210,
  "MaxOutputVoltage": 250,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 12,
  "MaxBattVoltage": 30,
  "MinimizeToTray": "True",
  "MinimizeOnStart": "True",
  "CloseToTray": "True",
  "StartWithWindows": "True",
  "UseLogFile": "True",
  "Log Level": 3,
  "ShutdownLimitBatteryCharge": 95,
  "ShutdownLimitUPSRemainTime": 120,
  "ImmediateStopAction": "False",
  "Follow_FSD": "False",
  "TypeOfStop": 1,
  "DelayToShutdown": 10,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "True",
  "VerifyUpdateAtStart": "True",
  "DelayBetweenEachVerification": 2,
  "StableOrDevBranch": 1,
  "LastDateVerification": "2022/10/01 10:10:32"
}

Error Message : 
Exception type: System.NullReferenceException
Exception message: Object reference not set to an instance of an object.
Exception stack trace: 
   at WinNUT_Client.Shutdown_Gui.Shutdown_Gui_Load(Object sender, EventArgs e) in D:\a\WinNUT-Client\WinNUT-Client\WinNUT_V2\WinNUT-Client\Shutdown_Gui.vb:line 38
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
2022/10/11 09:10:47 Pid: 19868 WinNUT : Update Dial
2022/10/11 09:10:47 Pid: 19868 WinNUT : Update Icon
2022/10/11 09:10:47 Pid: 19868 WinNUT : NotifyIcon Text => 
[... 235 lines cut ...]
Connected
On Battery (75%)
Battery OK
2022/10/11 09:11:52 Pid: 19868 WinNUT : Battery Status => Charging
2022/10/11 09:11:57 Pid: 19868 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/11 09:11:57 Pid: 19868 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/11 09:11:57 Pid: 19868 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/10/11 09:11:57 Pid: 19868 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
2022/10/11 09:11:57 Pid: 19868 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
2022/10/11 09:11:57 Pid: 19868 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
2022/10/11 09:11:57 Pid: 19868 UPS_Device : UPS statuses have not changed since last update, skipping.
2022/10/11 09:11:57 Pid: 19868 WinNUT : Updating UPS data for Form.
2022/10/11 09:11:57 Pid: 19868 WinNUT : Updating battery icons based on charge percent: 75%
2022/10/11 09:11:57 Pid: 19868 WinNUT : Calculated estimated remaining battery time: 01:41:15
2022/10/11 09:11:57 Pid: 19868 WinNUT : Update Dial
2022/10/11 09:11:57 Pid: 19868 WinNUT : Update Icon
2022/10/11 09:11:57 Pid: 19868 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Battery (75%)
Battery OK
2022/10/11 09:11:57 Pid: 19868 WinNUT : Battery Status => Charging

```
Hope this helps

gbakeman, 2022-10-11:
Hey, that's progress!

I went through the code for the Shutdown_Gui and cleaned it up, while also fixing the NRE error you encountered. Go ahead and download the new build [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3230024642). Thank you!

DravenSA, 2022-10-12:
Seems like success, It put the PC to sleep. 
When i woke the PC back up, the Client didn't reconnect to the NUT server though. I had to close the Client and restart it. 
I dont know if that was because i was running it as stand alone verses and install.
not sure if the logs will help you but i have put them here just in case
```
2022/10/12 08:36:28 Pid: 22428 Logger : Log file is initialized at E:\Desktop\New folder (8)\Data\Logs\WinNUT-Client-2022-10-12.log
2022/10/12 08:36:28 Pid: 22428 WinNUT : Parameter Dial Input Voltage Need to be Updated
2022/10/12 08:36:28 Pid: 22428 WinNUT : Parameter Dial Input Voltage Updated
2022/10/12 08:36:28 Pid: 22428 WinNUT : Parameter Dial Input Frequency Need to be Updated
2022/10/12 08:36:28 Pid: 22428 WinNUT : Parameter Dial Input Frequency Updated
2022/10/12 08:36:28 Pid: 22428 WinNUT : Parameter Dial Output Voltage Need to be Updated
2022/10/12 08:36:28 Pid: 22428 WinNUT : Parameter Dial Output Voltage Updated
2022/10/12 08:36:28 Pid: 22428 WinNUT : Parameter Dial Voltage Battery Need to be Updated
2022/10/12 08:36:28 Pid: 22428 WinNUT : Parameter Dial Voltage Battery Updated
2022/10/12 08:36:28 Pid: 22428 WinNUT : WinNut Preferences Applied.
2022/10/12 08:36:28 Pid: 22428 String : New Log to CB_Current Log : WinNut Preferences Changed
2022/10/12 08:36:28 Pid: 22428 WinNUT : Loaded Params Complete
2022/10/12 08:36:28 Pid: 22428 WinNUT : NotifyIcons Initialised
2022/10/12 08:36:28 Pid: 22428 WinNUT : Windows 10 Toast Notification Available
2022/10/12 08:36:28 Pid: 22428 WinNUT : Windows App Use Dark Theme
2022/10/12 08:36:28 Pid: 22428 WinNUT : Windows Use Dark Theme
2022/10/12 08:36:28 Pid: 22428 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
2022/10/12 08:36:28 Pid: 22428 WinNUT : Status Icon Changed
2022/10/12 08:36:28 Pid: 22428 WinNUT : New Icon Value For Systray : 1216
2022/10/12 08:36:28 Pid: 22428 WinNUT : New Icon Value For Gui : 1216
2022/10/12 08:36:28 Pid: 22428 WinNUT : Update Icon at Startup
2022/10/12 08:36:28 Pid: 22428 WinNUT : Run Automatic Update
2022/10/12 08:36:28 Pid: 22428 Update_Gui : Verify Update
2022/10/12 08:36:28 Pid: 22428 WinNUT : WinNUT Client v2.2.8319.36904 completed initialization.
2022/10/12 08:36:28 Pid: 22428 WinNUT : Main Gui Has Focus
2022/10/12 08:36:28 Pid: 22428 WinNUT : Update Icon
2022/10/12 08:36:28 Pid: 22428 WinNUT : Update Icon
2022/10/12 08:36:28 Pid: 22428 WinNUT : Minimize WinNut On Start
2022/10/12 08:36:28 Pid: 22428 WinNUT : Update Icon
2022/10/12 08:36:28 Pid: 22428 WinNUT : Minimize Main Gui To Notify Icon
2022/10/12 08:36:28 Pid: 22428 WinNUT : Main Gui Lose Focus
2022/10/12 08:36:28 Pid: 22428 WinNUT : Update Icon
2022/10/12 08:36:28 Pid: 22428 WinNUT : Auto-connecting to UPS on startup.
2022/10/12 08:36:28 Pid: 22428 UPS_Device : Beginning connection: upsmaster@192.168.0.59:3493, Name: mainpc [AutoReconnect]
2022/10/12 08:36:28 Pid: 22428 Nut_Socket : Attempting TCP socket connection to 192.168.0.59:3493...
2022/10/12 08:36:28 Pid: 22428 Nut_Socket : Connection established and streams ready for 192.168.0.59:3493
2022/10/12 08:36:28 Pid: 22428 Nut_Socket : Attempting authentication...
2022/10/12 08:36:28 Pid: 22428 Nut_Socket : Authenticated successfully.
2022/10/12 08:36:28 Pid: 22428 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
2022/10/12 08:36:28 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/12 08:36:28 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/12 08:36:28 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/12 08:36:28 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/12 08:36:28 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving battery.capacity
2022/10/12 08:36:28 Pid: 22428 WinNUT : mainpc has indicated it's ready to start sending data.
2022/10/12 08:36:28 Pid: 22428 WinNUT : Update Icon
2022/10/12 08:36:28 Pid: 22428 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
2022/10/12 08:36:28 Pid: 22428 WinNUT : Connection to Nut Host Established
2022/10/12 08:36:28 Pid: 22428 String : New Log to CB_Current Log : Connection to Nut Host 192.168.0.59:3493 Established
2022/10/12 08:36:28 Pid: 22428 WinNUT : Completed WinNUT_Shown
2022/10/12 08:36:32 Pid: 22428 WinNUT : Restore Main Gui On Mouse Click Notify Icon
2022/10/12 08:36:32 Pid: 22428 WinNUT : Main Gui Has Focus
2022/10/12 08:36:32 Pid: 22428 WinNUT : Update Icon
2022/10/12 08:36:32 Pid: 22428 WinNUT : Main Gui Lose Focus
2022/10/12 08:36:32 Pid: 22428 WinNUT : Update Icon
2022/10/12 08:36:33 Pid: 22428 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/12 08:36:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/12 08:36:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/10/12 08:36:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
2022/10/12 08:36:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
2022/10/12 08:36:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
[... 279 lines cut ...]
2022/10/12 08:38:23 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
2022/10/12 08:38:23 Pid: 22428 UPS_Device : UPS statuses have not changed since last update, skipping.
2022/10/12 08:38:28 Pid: 22428 UPS_Device : Enter Retrieve_UPS_Datas
[... cut ...]
2022/10/12 08:38:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/10/12 08:38:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
2022/10/12 08:38:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
2022/10/12 08:38:33 Pid: 22428 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
2022/10/12 08:38:33 Pid: 22428 UPS_Device : UPS statuses have not changed since last update, skipping.
2022/10/12 08:38:34 Pid: 22428 Pref_Gui : Show LogFile

```

gbakeman, 2022-10-13:
Thanks again for your testing reports.

I'm happy to report that I was able to setup a NUT server on my dev machine so I can do some simulations of my own and won't have to make you test every change. I think I've fixed both problems you've reported above - I was able to auto hibernate and resume with WinNUT making the connection automatically. [A fresh new build is ready](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3238958352), please give it a go and let me know of the results.

DravenSA, 2022-10-13:
Sleep works a treat, nice and clean.

after waking PC up, i noticed that in the dropdown box it said "[Id 13: 2022/10/13 08:45:54] Try Reconnect 1 / 30" waited over a minute, and it never changed, however the plus side of that is selecting Connection => Reconnect did work.

All my testing so far has been on Sleep, I have not yet tried Hibernate. Would you like me to test that too?

```
2022/10/13 08:43:05 Pid: 26152 Logger : Log file is initialized at E:\Desktop\New folder (9)\Data\Logs\WinNUT-Client-2022-10-13.log
2022/10/13 08:43:05 Pid: 26152 WinNUT : Parameter Dial Input Voltage Need to be Updated
2022/10/13 08:43:05 Pid: 26152 WinNUT : Parameter Dial Input Voltage Updated
2022/10/13 08:43:05 Pid: 26152 WinNUT : Parameter Dial Input Frequency Need to be Updated
2022/10/13 08:43:05 Pid: 26152 WinNUT : Parameter Dial Input Frequency Updated
2022/10/13 08:43:05 Pid: 26152 WinNUT : Parameter Dial Output Voltage Need to be Updated
2022/10/13 08:43:05 Pid: 26152 WinNUT : Parameter Dial Output Voltage Updated
2022/10/13 08:43:05 Pid: 26152 WinNUT : Parameter Dial Voltage Battery Need to be Updated
2022/10/13 08:43:05 Pid: 26152 WinNUT : Parameter Dial Voltage Battery Updated
2022/10/13 08:43:05 Pid: 26152 WinNUT : WinNut Preferences Applied.
2022/10/13 08:43:05 Pid: 26152 String : New Log to CB_Current Log : WinNut Preferences Changed
2022/10/13 08:43:05 Pid: 26152 WinNUT : Loaded Params Complete
2022/10/13 08:43:05 Pid: 26152 WinNUT : NotifyIcons Initialised
2022/10/13 08:43:05 Pid: 26152 WinNUT : Windows 10 Toast Notification Available
2022/10/13 08:43:05 Pid: 26152 WinNUT : Windows App Use Dark Theme
2022/10/13 08:43:05 Pid: 26152 WinNUT : Windows Use Dark Theme
2022/10/13 08:43:05 Pid: 26152 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
2022/10/13 08:43:05 Pid: 26152 WinNUT : Status Icon Changed
2022/10/13 08:43:05 Pid: 26152 WinNUT : New Icon Value For Systray : 1216
2022/10/13 08:43:05 Pid: 26152 WinNUT : New Icon Value For Gui : 1216
2022/10/13 08:43:05 Pid: 26152 WinNUT : Update Icon at Startup
2022/10/13 08:43:05 Pid: 26152 WinNUT : Run Automatic Update
2022/10/13 08:43:05 Pid: 26152 Update_Gui : Verify Update
2022/10/13 08:43:05 Pid: 26152 WinNUT : WinNUT Client v2.2.8321.1952 completed initialization.
2022/10/13 08:43:05 Pid: 26152 WinNUT : Main Gui Has Focus
2022/10/13 08:43:05 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:43:05 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:43:05 Pid: 26152 WinNUT : Minimize WinNut On Start
2022/10/13 08:43:05 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:43:05 Pid: 26152 WinNUT : Minimize Main Gui To Notify Icon
2022/10/13 08:43:05 Pid: 26152 WinNUT : Main Gui Lose Focus
2022/10/13 08:43:05 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:43:05 Pid: 26152 WinNUT : Auto-connecting to UPS on startup.
2022/10/13 08:43:05 Pid: 26152 WinNUT : Client UPS_Connect subroutine beginning.
2022/10/13 08:43:05 Pid: 26152 UPS_Device : Beginning connection: upsmaster@192.168.0.59:3493, Name: mainpc [AutoReconnect]
2022/10/13 08:43:05 Pid: 26152 Nut_Socket : Attempting TCP socket connection to 192.168.0.59:3493...
2022/10/13 08:43:05 Pid: 26152 Nut_Socket : Connection established and streams ready for 192.168.0.59:3493
2022/10/13 08:43:05 Pid: 26152 Nut_Socket : Attempting authentication...
2022/10/13 08:43:05 Pid: 26152 Nut_Socket : Authenticated successfully.
2022/10/13 08:43:05 Pid: 26152 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
2022/10/13 08:43:05 Pid: 26152 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/13 08:43:05 Pid: 26152 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/13 08:43:05 Pid: 26152 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/13 08:43:05 Pid: 26152 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/13 08:43:05 Pid: 26152 UPS_Device : Apply Fallback Value when retrieving battery.capacity
2022/10/13 08:43:05 Pid: 26152 WinNUT : mainpc has indicated it's ready to start sending data.
2022/10/13 08:43:05 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:43:05 Pid: 26152 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
2022/10/13 08:43:05 Pid: 26152 WinNUT : Connection to Nut Host Established
2022/10/13 08:43:05 Pid: 26152 String : New Log to CB_Current Log : Connection to Nut Host 192.168.0.59:3493 Established
2022/10/13 08:43:06 Pid: 26152 WinNUT : Completed WinNUT_Shown
2022/10/13 08:43:10 Pid: 26152 WinNUT : Restore Main Gui On Mouse Click Notify Icon
2022/10/13 08:43:10 Pid: 26152 WinNUT : Main Gui Has Focus
2022/10/13 08:43:10 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:43:10 Pid: 26152 WinNUT : Main Gui Lose Focus
2022/10/13 08:43:10 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:43:10 Pid: 26152 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/13 08:43:10 Pid: 26152 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/13 08:43:10 Pid: 26152 UPS_Device : Apply Fallback Value when retrieving input.frequency
[... 574 lines cut ...]
2022/10/13 08:47:44 Pid: 26152 WinNUT : Main Gui Lose Focus
2022/10/13 08:47:44 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:47:44 Pid: 26152 Pref_Gui : Pref Gui Opened.
2022/10/13 08:47:47 Pid: 26152 Pref_Gui : Setting LogControl statuses.
2022/10/13 08:47:48 Pid: 26152 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/13 08:47:48 Pid: 26152 UPS_Device : Apply Fallback Value when retrieving battery.runtime
[... cut ...]
2022/10/13 08:47:48 Pid: 26152 WinNUT : Updating battery icons based on charge percent: 100%
2022/10/13 08:47:48 Pid: 26152 WinNUT : Calculated estimated remaining battery time: 03:00:57
2022/10/13 08:47:48 Pid: 26152 WinNUT : Update Dial
2022/10/13 08:47:48 Pid: 26152 WinNUT : Update Icon
2022/10/13 08:47:48 Pid: 26152 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/13 08:47:48 Pid: 26152 WinNUT : Battery Status => Charged
2022/10/13 08:47:49 Pid: 26152 Pref_Gui : Show LogFile

```

gbakeman, 2022-10-13:
Glad to hear that sleep mode is working better!

It looks like you caught a rare bug where WinNUT found its self in a strange state (resuming from a suspended state while still connected) and it tried to repair the connection. It managed to make it nearly the whole way but didn't resume updating data for some reason. I can't see a reason why it became stuck like that, but what I did find out is that Windows is not consistent in reporting power state change events to programs, so I've moved some code around to make sure that WinNUT isn't caught in a state like that again.

I don't think you should have to test hibernation as long as sleep is your preferred method, since they're all basically a shutdown event as far as WinNUT is concerned. Once again, thank you for your persistence, and another debug build with the aforementioned fixes awaits you [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3245233964).

DravenSA, 2022-10-13:
So everything is working except reconnection after pc wake up, I don't even get the try reconnect 1/30 like i did last time,
the only way to reconnect was to actually click reconnect again
here is a screen shot of the dropdown box, and ofcourse the log files of the test
[image]

```
2022/10/13 21:57:33 Pid: 9680 Logger : Log file is initialized at E:\Desktop\New folder (10)\Data\Logs\WinNUT-Client-2022-10-13.log
2022/10/13 21:57:33 Pid: 9680 WinNUT : Parameter Dial Input Voltage Need to be Updated
2022/10/13 21:57:33 Pid: 9680 WinNUT : Parameter Dial Input Voltage Updated
2022/10/13 21:57:33 Pid: 9680 WinNUT : Parameter Dial Input Frequency Need to be Updated
2022/10/13 21:57:33 Pid: 9680 WinNUT : Parameter Dial Input Frequency Updated
2022/10/13 21:57:33 Pid: 9680 WinNUT : Parameter Dial Output Voltage Need to be Updated
2022/10/13 21:57:33 Pid: 9680 WinNUT : Parameter Dial Output Voltage Updated
2022/10/13 21:57:33 Pid: 9680 WinNUT : Parameter Dial Voltage Battery Need to be Updated
2022/10/13 21:57:33 Pid: 9680 WinNUT : Parameter Dial Voltage Battery Updated
2022/10/13 21:57:33 Pid: 9680 WinNUT : WinNut Preferences Applied.
2022/10/13 21:57:33 Pid: 9680 String : New Log to CB_Current Log : WinNut Preferences Changed
2022/10/13 21:57:33 Pid: 9680 WinNUT : Loaded Params Complete
2022/10/13 21:57:33 Pid: 9680 WinNUT : NotifyIcons Initialised
2022/10/13 21:57:33 Pid: 9680 WinNUT : Windows 10 Toast Notification Available
2022/10/13 21:57:33 Pid: 9680 WinNUT : Windows App Use Dark Theme
2022/10/13 21:57:33 Pid: 9680 WinNUT : Windows Use Dark Theme
2022/10/13 21:57:33 Pid: 9680 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
2022/10/13 21:57:33 Pid: 9680 WinNUT : Status Icon Changed
2022/10/13 21:57:33 Pid: 9680 WinNUT : New Icon Value For Systray : 1216
2022/10/13 21:57:33 Pid: 9680 WinNUT : New Icon Value For Gui : 1216
2022/10/13 21:57:33 Pid: 9680 WinNUT : Update Icon at Startup
2022/10/13 21:57:33 Pid: 9680 WinNUT : Run Automatic Update
2022/10/13 21:57:33 Pid: 9680 Update_Gui : Verify Update
2022/10/13 21:57:33 Pid: 9680 WinNUT : WinNUT Client v2.2.8321.35676 completed initialization.
2022/10/13 21:57:33 Pid: 9680 WinNUT : Main Gui Has Focus
2022/10/13 21:57:33 Pid: 9680 WinNUT : Update Icon
2022/10/13 21:57:33 Pid: 9680 WinNUT : Update Icon
2022/10/13 21:57:33 Pid: 9680 WinNUT : Minimize WinNut On Start
2022/10/13 21:57:33 Pid: 9680 WinNUT : Update Icon
2022/10/13 21:57:33 Pid: 9680 WinNUT : Minimize Main Gui To Notify Icon
2022/10/13 21:57:33 Pid: 9680 WinNUT : Main Gui Lose Focus
2022/10/13 21:57:33 Pid: 9680 WinNUT : Update Icon
2022/10/13 21:57:33 Pid: 9680 WinNUT : Auto-connecting to UPS on startup.
2022/10/13 21:57:33 Pid: 9680 WinNUT : Client UPS_Connect subroutine beginning.
2022/10/13 21:57:33 Pid: 9680 UPS_Device : Beginning connection: upsmaster@192.168.0.59:3493, Name: mainpc [AutoReconnect]
2022/10/13 21:57:33 Pid: 9680 Nut_Socket : Attempting TCP socket connection to 192.168.0.59:3493...
2022/10/13 21:57:33 Pid: 9680 Nut_Socket : Connection established and streams ready for 192.168.0.59:3493
2022/10/13 21:57:33 Pid: 9680 Nut_Socket : Attempting authentication...
2022/10/13 21:57:33 Pid: 9680 Nut_Socket : Authenticated successfully.
2022/10/13 21:57:33 Pid: 9680 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
2022/10/13 21:57:33 Pid: 9680 UPS_Device : ups.mfr is not supported by server.
2022/10/13 21:57:33 Pid: 9680 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/13 21:57:33 Pid: 9680 UPS_Device : ups.model is not supported by server.
2022/10/13 21:57:33 Pid: 9680 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/13 21:57:33 Pid: 9680 UPS_Device : ups.serial is not supported by server.
2022/10/13 21:57:33 Pid: 9680 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/13 21:57:33 Pid: 9680 UPS_Device : ups.firmware is not supported by server.
2022/10/13 21:57:33 Pid: 9680 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/13 21:57:33 Pid: 9680 UPS_Device : battery.capacity is not supported by server.
2022/10/13 21:57:33 Pid: 9680 UPS_Device : Apply Fallback Value when retrieving battery.capacity
2022/10/13 21:57:33 Pid: 9680 WinNUT : mainpc has indicated it's ready to start sending data.
2022/10/13 21:57:33 Pid: 9680 WinNUT : Update Icon
2022/10/13 21:57:33 Pid: 9680 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
2022/10/13 21:57:33 Pid: 9680 WinNUT : Connection to Nut Host Established
2022/10/13 21:57:33 Pid: 9680 String : New Log to CB_Current Log : Connection to Nut Host 192.168.0.59:3493 Established
2022/10/13 21:57:33 Pid: 9680 WinNUT : Completed WinNUT_Shown
2022/10/13 21:57:35 Pid: 9680 WinNUT : Restore Main Gui On Mouse Click Notify Icon
2022/10/13 21:57:35 Pid: 9680 WinNUT : Main Gui Has Focus
2022/10/13 21:57:35 Pid: 9680 WinNUT : Update Icon
2022/10/13 21:57:35 Pid: 9680 WinNUT : Main Gui Lose Focus
[... 550 lines cut ...]
2022/10/13 22:01:40 Pid: 9680 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/13 22:01:40 Pid: 9680 UPS_Device : input.frequency is not supported by server.
2022/10/13 22:01:40 Pid: 9680 UPS_Device : Apply Fallback Value when retrieving input.frequency
2022/10/13 22:01:40 Pid: 9680 UPS_Device : ups.realpower.nominal is not supported by server.
2022/10/13 22:01:40 Pid: 9680 UPS_Device : Apply Fallback Value when retrieving ups.realpower.nominal
[... cut ...]
2022/10/13 22:01:40 Pid: 9680 WinNUT : Updating battery icons based on charge percent: 100%
2022/10/13 22:01:40 Pid: 9680 WinNUT : Calculated estimated remaining battery time: 03:08:48
2022/10/13 22:01:40 Pid: 9680 WinNUT : Update Dial
2022/10/13 22:01:40 Pid: 9680 WinNUT : Update Icon
2022/10/13 22:01:40 Pid: 9680 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/13 22:01:40 Pid: 9680 WinNUT : Battery Status => Charged
2022/10/13 22:01:41 Pid: 9680 Pref_Gui : Setting LogControl statuses.
2022/10/13 22:01:43 Pid: 9680 Pref_Gui : Show LogFile

```

gbakeman, 2022-10-14:
It looks like the issue was that WinNUT was cancelling the autoreconnect process after just one error while reconnecting. I've fixed that so it shouldn't do it anymore, please try [build 8322 here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3253230532). I'm still seeing broader issues with UI inconsistencies and freezes while a connection is dropped or reconnected, but I think that may be for another bug if at all.

DravenSA, 2022-10-14:
Same this as before unfortunately just with different warning messages in the dropdown
[image]
had to reconnect physically again.
can the reconnect be on a delay? say 60 seconds?

```
2022/10/15 00:17:47 Pid: 24344 Logger : Log file is initialized at E:\Desktop\New folder (11)\Data\Logs\WinNUT-Client-2022-10-15.log
2022/10/15 00:17:47 Pid: 24344 WinNUT : Parameter Dial Input Voltage Need to be Updated
2022/10/15 00:17:47 Pid: 24344 WinNUT : Parameter Dial Input Voltage Updated
2022/10/15 00:17:47 Pid: 24344 WinNUT : Parameter Dial Input Frequency Need to be Updated
2022/10/15 00:17:47 Pid: 24344 WinNUT : Parameter Dial Input Frequency Updated
2022/10/15 00:17:47 Pid: 24344 WinNUT : Parameter Dial Output Voltage Need to be Updated
2022/10/15 00:17:47 Pid: 24344 WinNUT : Parameter Dial Output Voltage Updated
2022/10/15 00:17:47 Pid: 24344 WinNUT : Parameter Dial Voltage Battery Need to be Updated
2022/10/15 00:17:47 Pid: 24344 WinNUT : Parameter Dial Voltage Battery Updated
2022/10/15 00:17:47 Pid: 24344 WinNUT : WinNut Preferences Applied.
2022/10/15 00:17:47 Pid: 24344 String : New Log to CB_Current Log : WinNut Preferences Changed
2022/10/15 00:17:47 Pid: 24344 WinNUT : Loaded Params Complete
2022/10/15 00:17:47 Pid: 24344 WinNUT : NotifyIcons Initialised
2022/10/15 00:17:47 Pid: 24344 WinNUT : Windows 10 Toast Notification Available
2022/10/15 00:17:47 Pid: 24344 WinNUT : Windows App Use Dark Theme
2022/10/15 00:17:47 Pid: 24344 WinNUT : Windows Use Dark Theme
2022/10/15 00:17:47 Pid: 24344 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
2022/10/15 00:17:47 Pid: 24344 WinNUT : Status Icon Changed
2022/10/15 00:17:47 Pid: 24344 WinNUT : New Icon Value For Systray : 1216
2022/10/15 00:17:47 Pid: 24344 WinNUT : New Icon Value For Gui : 1216
2022/10/15 00:17:47 Pid: 24344 WinNUT : Update Icon at Startup
2022/10/15 00:17:47 Pid: 24344 WinNUT : Run Automatic Update
2022/10/15 00:17:47 Pid: 24344 Update_Gui : Verify Update
2022/10/15 00:17:47 Pid: 24344 WinNUT : WinNUT Client v2.2.8322.39776 completed initialization.
2022/10/15 00:17:47 Pid: 24344 WinNUT : Main Gui Has Focus
2022/10/15 00:17:47 Pid: 24344 WinNUT : Update Icon
2022/10/15 00:17:47 Pid: 24344 WinNUT : Update Icon
2022/10/15 00:17:47 Pid: 24344 WinNUT : Minimize WinNut On Start
2022/10/15 00:17:47 Pid: 24344 WinNUT : Update Icon
2022/10/15 00:17:47 Pid: 24344 WinNUT : Minimize Main Gui To Notify Icon
2022/10/15 00:17:47 Pid: 24344 WinNUT : Main Gui Lose Focus
2022/10/15 00:17:47 Pid: 24344 WinNUT : Update Icon
2022/10/15 00:17:47 Pid: 24344 WinNUT : Auto-connecting to UPS on startup.
2022/10/15 00:17:47 Pid: 24344 WinNUT : Client UPS_Connect subroutine beginning.
2022/10/15 00:17:47 Pid: 24344 UPS_Device : Beginning connection: upsmaster@192.168.0.59:3493, Name: mainpc [AutoReconnect]
2022/10/15 00:17:47 Pid: 24344 Nut_Socket : Attempting TCP socket connection to 192.168.0.59:3493...
2022/10/15 00:17:47 Pid: 24344 Nut_Socket : Connection established and streams ready for 192.168.0.59:3493
2022/10/15 00:17:47 Pid: 24344 Nut_Socket : Attempting authentication...
2022/10/15 00:17:47 Pid: 24344 Nut_Socket : Authenticated successfully.
2022/10/15 00:17:47 Pid: 24344 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
2022/10/15 00:17:47 Pid: 24344 UPS_Device : ups.mfr is not supported by server.
2022/10/15 00:17:47 Pid: 24344 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/15 00:17:47 Pid: 24344 UPS_Device : ups.model is not supported by server.
2022/10/15 00:17:47 Pid: 24344 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/15 00:17:47 Pid: 24344 UPS_Device : ups.serial is not supported by server.
2022/10/15 00:17:47 Pid: 24344 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/15 00:17:47 Pid: 24344 UPS_Device : ups.firmware is not supported by server.
2022/10/15 00:17:47 Pid: 24344 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/15 00:17:47 Pid: 24344 UPS_Device : battery.capacity is not supported by server.
2022/10/15 00:17:47 Pid: 24344 UPS_Device : Apply Fallback Value when retrieving battery.capacity
2022/10/15 00:17:47 Pid: 24344 WinNUT : mainpc has indicated it's ready to start sending data.
2022/10/15 00:17:47 Pid: 24344 WinNUT : Update Icon
2022/10/15 00:17:47 Pid: 24344 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
2022/10/15 00:17:47 Pid: 24344 WinNUT : Connection to Nut Host Established
2022/10/15 00:17:47 Pid: 24344 String : New Log to CB_Current Log : Connection to Nut Host 192.168.0.59:3493 Established
2022/10/15 00:17:47 Pid: 24344 WinNUT : Completed WinNUT_Shown
2022/10/15 00:17:52 Pid: 24344 UPS_Device : Enter Retrieve_UPS_Datas
2022/10/15 00:17:52 Pid: 24344 UPS_Device : battery.runtime is not supported by server.
2022/10/15 00:17:52 Pid: 24344 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/15 00:17:52 Pid: 24344 UPS_Device : input.frequency is not supported by server.
[... 666 lines cut ...]
2022/10/15 00:22:54 Pid: 24344 UPS_Device : Apply Fallback Value when retrieving ups.power.nominal
2022/10/15 00:22:54 Pid: 24344 UPS_Device : ups.current.nominal is not supported by server.
2022/10/15 00:22:54 Pid: 24344 UPS_Device : Apply Fallback Value when retrieving ups.current.nominal
2022/10/15 00:22:54 Pid: 24344 UPS_Device : UPS statuses have CHANGED...
2022/10/15 00:22:54 Pid: 24344 UPS_Device : Current statuses: OL
2022/10/15 00:22:54 Pid: 24344 WinNUT : Handling new UPS status(es)...
[... cut ...]
2022/10/15 00:22:54 Pid: 24344 WinNUT : Calculated estimated remaining battery time: 02:23:29
2022/10/15 00:22:54 Pid: 24344 WinNUT : Update Dial
2022/10/15 00:22:54 Pid: 24344 WinNUT : Update Icon
2022/10/15 00:22:54 Pid: 24344 WinNUT : Status Icon Changed
2022/10/15 00:22:54 Pid: 24344 WinNUT : New Icon Value For Systray : 1136
2022/10/15 00:22:54 Pid: 24344 WinNUT : New Icon Value For Gui : 1136
2022/10/15 00:22:54 Pid: 24344 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/15 00:22:54 Pid: 24344 WinNUT : Battery Status => Charged
```

gbakeman, 2022-10-14:
Wow, this has been a tough one. I think the issue is that your PC's (and most, to be honest) networking is not immediately ready when it resumes from a suspended state. I haven't seen this issue since my NUT server is running in a VM and therefor immediately available on resume. So it looks like the core problem is that, when resuming from a suspended state, WinNUT only tries to reconnect once and doesn't launch the reconnection procedure after a connection failure. I've changed it so that it will now do this and attempt the specified number of times to reconnect.

> can the reconnect be on a delay? say 60 seconds?

Actually this was hard coded to 30 seconds in the program, but I had it set to five seconds for testing. I've now set it to five seconds no matter what. This could be a user preference, but I need to dive into and understand localizations before I start adding strings everywhere for the user to see.

Another build is up, [please give it a try](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3253440681)! And thank you for getting back so quickly.

DravenSA, 2022-10-14:
Looks like we have a winner Took a little longer for it to realise that the UPS was on battery, but that isnt really a problem (less than 30 seconds)
as you can see in the screen shot it did an attempt and then it reconnected itself automatically.
[image]
Below are the logs, just in case you want to have a look through them. (Hopefully they are clean and you dont see errors )

Will look forward to proper release, that i can install, and hopefully other will appreciate all the hard work you have done to make the sleep function work.

Well as you might see by the log time, it is after 1 am and i guess i need to sleep at some point.
Thanks again, really appreciate it.
```
2022/10/15 01:02:34 Pid: 22776 Logger : Log file is initialized at E:\Desktop\New folder (12)\Data\Logs\WinNUT-Client-2022-10-15.log
2022/10/15 01:02:34 Pid: 22776 WinNUT : Parameter Dial Input Voltage Need to be Updated
2022/10/15 01:02:34 Pid: 22776 WinNUT : Parameter Dial Input Voltage Updated
2022/10/15 01:02:34 Pid: 22776 WinNUT : Parameter Dial Input Frequency Need to be Updated
2022/10/15 01:02:34 Pid: 22776 WinNUT : Parameter Dial Input Frequency Updated
2022/10/15 01:02:34 Pid: 22776 WinNUT : Parameter Dial Output Voltage Need to be Updated
2022/10/15 01:02:34 Pid: 22776 WinNUT : Parameter Dial Output Voltage Updated
2022/10/15 01:02:34 Pid: 22776 WinNUT : Parameter Dial Voltage Battery Need to be Updated
2022/10/15 01:02:34 Pid: 22776 WinNUT : Parameter Dial Voltage Battery Updated
2022/10/15 01:02:34 Pid: 22776 WinNUT : WinNut Preferences Applied.
2022/10/15 01:02:34 Pid: 22776 String : New Log to CB_Current Log : WinNut Preferences Changed
2022/10/15 01:02:34 Pid: 22776 WinNUT : Loaded Params Complete
2022/10/15 01:02:34 Pid: 22776 WinNUT : NotifyIcons Initialised
2022/10/15 01:02:34 Pid: 22776 WinNUT : Windows 10 Toast Notification Available
2022/10/15 01:02:34 Pid: 22776 WinNUT : Windows App Use Dark Theme
2022/10/15 01:02:34 Pid: 22776 WinNUT : Windows Use Dark Theme
2022/10/15 01:02:34 Pid: 22776 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
2022/10/15 01:02:34 Pid: 22776 WinNUT : Status Icon Changed
2022/10/15 01:02:34 Pid: 22776 WinNUT : New Icon Value For Systray : 1216
2022/10/15 01:02:34 Pid: 22776 WinNUT : New Icon Value For Gui : 1216
2022/10/15 01:02:34 Pid: 22776 WinNUT : Update Icon at Startup
2022/10/15 01:02:34 Pid: 22776 WinNUT : Run Automatic Update
2022/10/15 01:02:34 Pid: 22776 Update_Gui : Verify Update
2022/10/15 01:02:34 Pid: 22776 WinNUT : WinNUT Client v2.2.8322.41316 completed initialization.
2022/10/15 01:02:34 Pid: 22776 WinNUT : Main Gui Has Focus
2022/10/15 01:02:34 Pid: 22776 WinNUT : Update Icon
2022/10/15 01:02:34 Pid: 22776 WinNUT : Update Icon
2022/10/15 01:02:34 Pid: 22776 WinNUT : Minimize WinNut On Start
2022/10/15 01:02:34 Pid: 22776 WinNUT : Update Icon
2022/10/15 01:02:34 Pid: 22776 WinNUT : Minimize Main Gui To Notify Icon
2022/10/15 01:02:34 Pid: 22776 WinNUT : Main Gui Lose Focus
2022/10/15 01:02:34 Pid: 22776 WinNUT : Update Icon
2022/10/15 01:02:34 Pid: 22776 WinNUT : Auto-connecting to UPS on startup.
2022/10/15 01:02:34 Pid: 22776 WinNUT : Client UPS_Connect subroutine beginning.
2022/10/15 01:02:34 Pid: 22776 UPS_Device : Beginning connection: upsmaster@192.168.0.59:3493, Name: mainpc [AutoReconnect]
2022/10/15 01:02:34 Pid: 22776 Nut_Socket : Attempting TCP socket connection to 192.168.0.59:3493...
2022/10/15 01:02:34 Pid: 22776 Nut_Socket : Connection established and streams ready for 192.168.0.59:3493
2022/10/15 01:02:34 Pid: 22776 Nut_Socket : Attempting authentication...
2022/10/15 01:02:34 Pid: 22776 Nut_Socket : Authenticated successfully.
2022/10/15 01:02:34 Pid: 22776 Nut_Socket : NUT server reports VER: 2.7.4 NETVER: 1.2
2022/10/15 01:02:34 Pid: 22776 UPS_Device : ups.mfr is not supported by server.
2022/10/15 01:02:34 Pid: 22776 UPS_Device : Apply Fallback Value when retrieving ups.mfr
2022/10/15 01:02:34 Pid: 22776 UPS_Device : ups.model is not supported by server.
2022/10/15 01:02:34 Pid: 22776 UPS_Device : Apply Fallback Value when retrieving ups.model
2022/10/15 01:02:34 Pid: 22776 UPS_Device : ups.serial is not supported by server.
2022/10/15 01:02:34 Pid: 22776 UPS_Device : Apply Fallback Value when retrieving ups.serial
2022/10/15 01:02:34 Pid: 22776 UPS_Device : ups.firmware is not supported by server.
2022/10/15 01:02:34 Pid: 22776 UPS_Device : Apply Fallback Value when retrieving ups.firmware
2022/10/15 01:02:34 Pid: 22776 UPS_Device : battery.capacity is not supported by server.
2022/10/15 01:02:34 Pid: 22776 UPS_Device : Apply Fallback Value when retrieving battery.capacity
2022/10/15 01:02:34 Pid: 22776 WinNUT : mainpc has indicated it's ready to start sending data.
2022/10/15 01:02:34 Pid: 22776 WinNUT : Update Icon
2022/10/15 01:02:34 Pid: 22776 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
2022/10/15 01:02:34 Pid: 22776 WinNUT : Connection to Nut Host Established
2022/10/15 01:02:34 Pid: 22776 String : New Log to CB_Current Log : Connection to Nut Host 192.168.0.59:3493 Established
2022/10/15 01:02:34 Pid: 22776 WinNUT : Completed WinNUT_Shown
[... 819 lines cut ...]
2022/10/15 01:06:03 Pid: 22776 UPS_Device : battery.runtime is not supported by server.
2022/10/15 01:06:03 Pid: 22776 UPS_Device : Apply Fallback Value when retrieving battery.runtime
2022/10/15 01:06:03 Pid: 22776 UPS_Device : input.frequency is not supported by server.
[... cut ...]
2022/10/15 01:06:03 Pid: 22776 WinNUT : Calculated estimated remaining battery time: 02:54:12
2022/10/15 01:06:03 Pid: 22776 WinNUT : Update Dial
2022/10/15 01:06:03 Pid: 22776 WinNUT : Update Icon
2022/10/15 01:06:03 Pid: 22776 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Connected
On Line
Battery OK
2022/10/15 01:06:03 Pid: 22776 WinNUT : Battery Status => Charged
2022/10/15 01:06:03 Pid: 22776 Pref_Gui : Setting LogControl statuses.
2022/10/15 01:06:05 Pid: 22776 Pref_Gui : Show LogFile
```

gbakeman, 2022-10-14:
That's great to hear! You're welcome, and I appreciate your help on this. I'm going to merge the changes and close this issue, then I should have an official pre-release out pending one or more other issues. Have a good night!

PR #36 (2022-10-14): Re-activate stop actions system and status updates overhaul
It looks like the stop action system was disabled during the client library changes. I've reactivated it.
I've also redone how UPS statuses are updated and processed, in the hopes of making program execution more efficient and bug-Closes #27
commit f6206b9f (2022-10-14): Merge pull request #36 from nutdotnet/sleepmode

Re-activate stop actions system, fix resume issues and status updates overhaul. Closes #27
closed 2022-10-14 by gbakeman