nutdotnet/WinNUT-Client #31: Comments needed in the translation.csv files

MartinKurtz, 2022-09-30:
When making some updates to the german translation file, i noticed that some of the lines definitely need some comments that gives some context and some explanation on what it actually means, because some of the longer lines can have more than one meaning or can even have multiple meanings

I would suggest that longer lines definitely need context and an explanation, short lines may sometimes need some context, so it is clear where the translation will appear, as in other languages "Close" may translate to the equivalent of "Close Window" or "Close Program" in that other language, where for example it could be schließen for closing a window and beenden for closing a program

Such comments would probably be best written in english

gbakeman, 2022-10-20:
Hi @MartinKurtz, thank you for the feedback. I agree that translations maybe need to be gone over and clarified/edited since sometimes they're too long, or containing information that doesn't even belong in a translation (like variable numbers or other information). I admit I still don't completely understand the system @gawindx has setup for collecting the translations, but I think there's a way to simplify it if you don't mind installing another program.

I started writing a developer's guide [here](https://github.com/nutdotnet/WinNUT-Client/wiki/Development#translations) for doing translations in WinNUT. Essentially, you have to install a program called [Multilingual App Toolkit 4.0](https://learn.microsoft.com/en-us/windows/apps/design/globalizing/use-mat) (a Microsoft program) which provides a GUI for editing the translation files of WinNUT. After you install that program, you open the file `WinNUT_V2/WinNUT-Client/MultilingualResources/WinNUT-client.de-DE.xlf` inside it and you'll be able to directly translate virtually any part of WinNUT with it. I can start adding comments where needed, maybe add details to some sentences and simplify others. Do you want to track a list here for resources/strings that you want me to look at?

MartinKurtz, 2022-10-21:
How should i note down lines which need clarification? do you want a CSV file?

MartinKurtz, 2022-10-21:
I also tried to use the multilingual toolkit and despite opening the file, nothing is really happening differently, the developers guide may need some extra steps and instructions on how to use this

gbakeman, 2022-10-21:
> I also tried to use the multilingual toolkit and despite opening the file, nothing is really happening differently, the developers guide may need some extra steps and instructions on how to use this

I think all you need to do is open the toolkit and then open the XLF file, but I will see if I can maybe add some screenshots and additional details to the guide (and also verify that I'm giving good instructions.) If you want to copy lines from the translation CSV into this issue (ideally in the form of a checklist), then I can go back and add additional information. I'm hoping that the Multilingual Toolkit is useful so we can do away with the CSV files entirely and just use the comment fields in the toolkit.

MartinKurtz, 2022-10-21:
I think some screenshots would do wonders

gbakeman, 2022-10-22:
I've made a new tutorial for using the toolkit [here](https://github.com/nutdotnet/WinNUT-Client/wiki/Translations), please give it a try when you can and let me know your thoughts.