nutdotnet/WinNUT-Client #32: Transtation csv (at least the german one) does not seem to affect the right click menu on the systray icon and some other places

MartinKurtz, 2022-09-30:
I noticed this when searching the word "Ausgang" in the german translation csv and it did not appear there, however it is present on the mentioned right click menu in the german version. 
For example the check for updates option in the csv is "Auf Updates prüfen." in the systray it is "Überprüfen sie das Update" just as within the program when you click on Help and then Check for updates.

I downloaded the newest prerelease  this morning:
[image]

Lineflyer, 2022-10-09:
This is similar to the problem I wanted to mention:
In version 2.2.8303 (same as mentioned above) I noticed, that the german translation for the settings menu is "die Einstellungen" although the german translation on branch `Dev-2.2` denotes "Einstellungen", which would be correct:
https://github.com/nutdotnet/WinNUT-Client/blob/Dev-2.2/Translation/de-DE/de-DE.csv#L119

gbakeman, 2022-10-15:
Hi everyone,

I admit I do not understand very well how the translation system works in WinNUT. As far as I'm aware, the translation csv doesn't actually affect any part of the program, rather the `Resources.resx` file is what's used. I'm going to spend some time documenting how translations work in .Net and WinNUT, and see if we can make translations more consistent throughout the program.

gbakeman, 2022-10-19:
So I've learned a lot about how translations work, and I think I've figured out how to apply them to the program. If you could test the latest debug version [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3284896817), please let me know if the translated strings are correctly applied and I'll have a pre-release build up soon.

Lineflyer, 2022-10-19:
Will do. Probably next weekend.

gbakeman, 2022-10-19:
@Lineflyer thank you for offering to take a look. I realized I forgot to merge @MartinKurtz's changes, so I've made a [new debug build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3285469203) for you to take a look when possible.

Lineflyer, 2022-10-20:
Had the chance to test the debug build right now.
I can confirm the translations are up to date with the latest changes.

MartinKurtz, 2022-10-20:
Do you want to create a pre-release with all the changes so they can be tested be people who dont wanna compile themselves, since the latest one is basically a month old?

gbakeman, 2022-10-20:
@MartinKurtz I spent some time setting up Continuous Integration for WinNUT's repository - that means GitHub builds the program whenever I push a change to the repository. These are only debug builds that you download and run from the folder, no install is created at the moment. You can find these in the **Actions** menu at the top of the repository, and I also linked to a few generated builds in this issue discussion. @Lineflyer has validated the changes so I think I will close this issue as completed, but I plan on putting out a pre-release soon (next day or so) and I'd appreciate having your feedback as well once it's out.

Thank you both!

closed 2022-10-20 by gbakeman

MartinKurtz, 2022-10-20:
I will definitely test it and may even contribute some code, once i have a good understanding of the project and the language used, but for now i wasnt able to compile myself a installer from the repo, which is most likely a issue on my end

Lineflyer, 2022-10-20:
> ]...] wanna compile themselves, since the latest one is basically a month old?

For the linked debug build you dont need to compile anything. Can be used by downloading the artifact listed in the link.

gbakeman, 2022-10-20:
> I will definitely test it and may even contribute some code, once i have a good understanding of the project and the language used, but for now i wasnt able to compile myself a installer from the repo, which is most likely a issue on my end

@MartinKurtz I started writing a short how-to for building the project (and installer) I think on the README, let me know if it needs more details. I just want to warn you, I don't think there's much of a future for this version of WinNUT. I'm considering this project "legacy", and I'm only updating to fix bugs, translations and make sure it can exist as a stable version of WinNUT. There's a lot of technical debt and legacy code in need of modernizing. As such I've started another project called [Coco.Nut](https://github.com/nutdotnet/Coco.Nut) to create a modern version of WinNUT with up-to-date UI and .Net frameworks. Please give it a look if you're at all interested, submit ideas/suggestions, and I'd love to have someone contributing code (planning on doing it in C#.)

PR #39 (2022-10-20): Applying translations
I've had to do some research on how we apply translations in WinNUT. I believe I've learned the correct procedure (and documented it [here](https://github.com/nutdotnet/WinNUT-Client/wiki/Development#translations)), therefor I've applied recent de-DE and zh-TW translations to the program.

Closes #32.