nutdotnet/WinNUT-Client #35: WinNUT 2.2.8303 GUI failure when interval=0

deajan, 2022-10-07:
I've upgraded an earlier WinNUT 2.0.x release to latest beta Winnut 2.2.8303.
Now everytime I launch the GUI, I get the follwoing error message, roughly translating to value '0' is not valid for interval.

```
Exception type: System.ArgumentOutOfRangeException
Exception message: La valeur '0' n'est pas valide pour Interval. Interval doit être supérieur à 0.
Nom du paramètre : Interval
Exception stack trace: 
   à System.Windows.Forms.Timer.set_Interval(Int32 value)
   à WinNUT_Client_Common.UPS_Device.set_PollingInterval(Int32 value)
   à WinNUT_Client_Common.UPS_Device..ctor(Nut_Parameter& Nut_Config, Logger& LogFile, Int32 pollInterval)
   à WinNUT_Client.WinNUT.UPS_Connect()
   à WinNUT_Client.WinNUT.WinNUT_Shown(Object sender, EventArgs e)
   à System.Windows.Forms.Form.OnShown(EventArgs e)
   à System.Windows.Forms.Form.CallShownEvent()
   à System.Windows.Forms.Control.InvokeMarshaledCallbackDo(ThreadMethodEntry tme)
   à System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   à System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   à System.Windows.Forms.Control.InvokeMarshaledCallback(ThreadMethodEntry tme)
   à System.Windows.Forms.Control.InvokeMarshaledCallbacks()
```

I've managed to update the interval value directly in registry path `HKEY_CURRENT_USER\SOFTWARE\WinNUT\Connexion\Delay' from 0x00000000 to 0x000001388 (5000 ms I guess ?).
The software runs now.
But what baffles me is that it's possible to set the interval again to zero in the GUI, rendering the GUI unusable on next run.

gbakeman, 2022-10-14:
Thank you for the bug report. Somehow I thought I fixed this, but apparently not. I've just pushed a fix and the debug build should be ready for you by the time you see this message. Please [download it from here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3253354862) and let me know if the issue is fixed.

Thank you!

closed 2022-10-20 by gbakeman

gbakeman, 2022-10-27:
Hi @deajan ,

Thank you for getting back to me to verify the original patch, as well as providing detailed bug reports following that.

I think it's important that WinNUT runs as a standard user so it's good you're testing it as such. Interestingly, someone else just reported that first crash error when running a later debug build. I'm curious if you too were attempting to run the debug build from inside the program files folder? The debug build is designed to be operated from a folder where the user has write access (such as a subfolder in your Downloads folder, for example.) I understand if this is undesirable though, and I've just released an updated debug build in #1 if you have the time to attempt to reproduce that first bug.

As for the second one, I believe that was patched in a later release as you may have seen when you updated to 8331. So thank you again for your testing and reporting efforts, and I hope you'll return if you see anything else!

deajan, 2022-10-28:
@gbakeman Yes, I copied the debug build over the original WinNUT installation in progam files folder.
So far soo good, nothing spectacular except one crash with current build.
I've enabled debug logs in the GUI.
I'll report back if something goes wrong.

Thanks for the time and effort.

PR #37 (2022-10-20): Correct Interval/Delay value
The Interval/Delay value has been incorrectly set in older versions to 0, which causes an ArgumentOutOfRange exception upon connecting the UPS. Make sure this value is corrected when Prefs are updated, and make sure the Prefs form validates the field.

Closes #35.