nutdotnet/WinNUT-Client #38: Initial setup

katiekat641, 2022-10-19:
I am trying to setup WinNUT on my Windows 8 pc to be the slave to my QNAP. I have tried following the instructions in the QNAP forum post https://forum.qnap.com/viewtopic.php?f=151&t=47493

but cannot get it connected. I have SSH into the NAS and added the following to the upsmon.conf:

 MONITOR qnapups@ myQNAP's ip 1 admin 123456 master

[Crash_Report_19-10-2022_14-28-10.txt](https://github.com/nutdotnet/WinNUT-Client/files/9823618/Crash_Report_19-10-2022_14-28-10.txt)

WinNUT keeps encountering an error and closing.  Have attached the log after most recent error.

gbakeman, 2022-10-19:
Hi @katiekat641 ,

Thank you for posting here. It looks like the error you are encountering is the same as in #35. Please download the debug build I've attached to that issue and see if it fixes your error. I'm going to close this issue; please comment over there if the error persists, or comment back here if it's a different error and I'll reopen this issue.

katiekat641, 2022-10-19:
Unfortunately still crashes. Have attached new report.
[Crash_Report_19-10-2022_17-00-10.txt](https://github.com/nutdotnet/WinNUT-Client/files/9824661/Crash_Report_19-10-2022_17-00-10.txt)

gbakeman, 2022-10-19:
Alright, that's a good one! Let me look into it and get back to you.

gbakeman, 2022-10-19:
Update for you: I'm looking through the current code, but I'm mystified how WinNUT isn't throwing more errors. I wonder if I may have fixed a bug or two in some other pending changes. Would you mind waiting for the next pre-release to test again and see if I may have inadvertently fixed this? I have someone verifying some other changes this weekend, then I'm hoping I can push out the pre-release by then. I can ping you here when that's ready.

Sorry for the wait and thank you for your patience.

katiekat641, 2022-10-19:
Sure, no worries. I appreciate your fast replies. Thanks so much.

gbakeman, 2022-10-21:
@katiekat641 thank you for waiting, I was able to push out a new pre-release. Please download it [here](https://github.com/nutdotnet/WinNUT-Client/releases) and post back with a log if the issue is still present.

katiekat641, 2022-10-21:
Success! That seems to have fixed my issues. Thanks so much for your prompt support.

closed 2022-10-21 by katiekat641

gbakeman, 2022-10-21:
You're welcome, glad to hear it was fixed!