nutdotnet/WinNUT-Client #40: Legacy and Future of WinNUT

gbakeman, 2022-10-21:
Welcome, and I'm glad you've found the current home for WinNUT. WinNUT has had a long history as a project dedicated to presenting information from a NUT server to a Windows computer, and offering critical automation to make sure your system is safe from a power outage. This repository was migrated from [gawindx/WinNUT-Client](https://github.com/gawindx/WinNUT-Client) and work has continued to make sure WinNUT is in a stable state for users around the world.

WinNUT was migrated to where it is today in its Windows Forms and Visual Basic .Net form thanks to the dedication and hard work of @gawindx and other contributors. However, WinNUT is collecting technical debt, and the technology it was built upon is aging and nearing deprecation from Microsoft.

## The future of WinNUT
As you can see, WinNUT now lives under the [nutdotnet](https://github.com/nutdotnet) GitHub Organization, which I intend to be the home of all .Net software for NUT and UPS interaction. Work has already been completed to create a modern .Net library (currently at _Minimum Viable Product_ status) with the sole purpose of maintaining a connection to and interacting with a NUT server. All that's left is to build a UI ontop of that library, and so the time has arrived to create the next generation of desktop NUT monitoring software.

### Introducing Coco.Nut
I've already created a new repository with the working name of Coco.Nut (https://github.com/nutdotnet/Coco.Nut), and I intend for this to replace WinNUT entirely. It will be written in Microsoft's C# language (same as the Nut.Net library), and written for a modern cross-platform UI toolkit (most likely MAUI) with the intent that it will run on any traditional personal computer, and potentially mobile devices as well. Please **star the Coco.Nut repository**, follow it and join in the discussions by adding your feedback and ideas for features you'd like to see added into the program.

WinNUT will continue to have bug fixes or important features added to it as necessary. This project is always open to pull requests, but I discourage anyone from putting too much of their time into a new feature. I'm committed to a stable 2.2 release soon, and potentially a final stable 2.3 release with bugfixes thereafter. At that point, the WinNUT repository will enter a legacy state and development will wind down to only accepting translation and bug fixes.

Thank you for your support, and I hope you'll join us for WinNUT's next generation.

closed 2022-10-21 by gbakeman

gbakeman, 2025-08-27:
> Status?

Still alive and being worked on! Other things came up, although I was stuck in a rut trying to troubleshoot why the Multilingual extension takes a long time to install when running in a GitHub runner. If you have any skills in PowerShell, Visual Studio extensions or GitHub automation, [I could use some help](https://github.com/gbakeman/install-vsix/pull/5)!

SebastianOehm, 2026-10-09:
@gbakeman I made a one-shot that is at least useable as a POC over at my [fork ](https://github.com/SebastianOehm/Coco.Nut)under branch [preMain](https://github.com/SebastianOehm/Coco.Nut/tree/preMain) and releases(https://github.com/SebastianOehm/Coco.Nut/releases/tag/v0.1.0-pre.11), do you maybe have a look at it and tell me whether that is in the direction you were thinking? One thing though, I have not used MAUI, but Avalonia instead to include linux if someone wanted to and to have one, same, UI across platforms