nutdotnet/WinNUT-Client #41: UPS Power out showing wrong Wattage

MartinKurtz, 2022-10-21:
Basically what it says in the title, my UPS has a max power out of 1400w so 20 percent arent 136w

I have been trying to add the option to manually set the maximum power myself ,but am currently failing at correctly positioning the field in the Calibration tab of the settings, and have the program calculate the power usage from that and the load percentage.  I have also not managed to find out how you correctly use the .resx files and correctly sync them between each other so once i add a new button/tooltip or whatever such a line is added in all other languages

gbakeman, 2022-10-22:
Just so I understand correctly, this is regarding the **UPS Load** dial? I think it is strange because the calibration menu lets you change what percentage range between 0 and 100 the dial shows, which makes no sense to me.

Besides the `ups.load` variable which does appear to be a calculation given by the UPS its self, we also try to find the output power like so:

1. Check if UPS reports `ups.realpower.nominal`, and multiply it by the load percentage.
2. If it's not provided, then check for `ups.power.nominal` and multiply it by the load percentage and **0.6** (CosPhi value).
3. If that isn't provided, then get `ups.current.nominal` (default to **1** if not provided), and multiply that by the Input Voltage, **0.95** and **0.6**.

As I was looking at your bug report, I also realized that I've broken the UPS Variable list window at some point, so I'll need to fix that before I can troubleshoot further. Once that window is working again, could you compare what your real UPS power consumption is, versus what's reported in the variables window, and what WinNUT displays?

I also wonder if we could relate this to #24 and try to autodetect some of the calibration values instead of needing to manually input them.

MartinKurtz, 2022-12-27:
Allright, sorry for my long abscence, university is stressful.
Ive installed the newest prerelease and WinNUT shows me a solid 134 watts while i am pulling some 260 to 300 watts from the wall and the 134 watts dont change no matter what, exept if my UPs goes on battery power.

I may do some digging through the code, but i probably will not understand the code sufficiently to do anything about it, as my knowledge of it extremely limited and this is a complicated project for me

gbakeman, 2022-12-28:
Ok, before we go further it would be helpful to see the output of the variables list here. I'm busy with a lot of things too and probably won't be able to do much until mid-January at the earliest, so no worries.

MartinKurtz, 2022-12-28:
nutdev1 (/2000VA/)
battery.charge (Battery charge (percent of full)) : 100
battery.voltage (Battery voltage (V)) : 27.60
battery.voltage.high (Description unavailable) : 26.00
battery.voltage.low (Description unavailable) : 20.80
battery.voltage.nominal (Nominal battery voltage (V)) : 24.0
device.mfr (Description unavailable) : 
device.model (Description unavailable) : 2000VA
device.type (Description unavailable) : ups
driver.name (Driver name) : blazer_usb
driver.parameter.pollinterval (Description unavailable) : 5
driver.parameter.port (Description unavailable) : auto
driver.parameter.synchronous (Description unavailable) : no
driver.version (Driver version - NUT release) : 2.7.4
driver.version.internal (Internal driver version) : 0.12
input.current.nominal (Nominal input current (A)) : 8.0
input.frequency (Input line frequency (Hz)) : 50.0
input.frequency.nominal (Nominal input line frequency (Hz)) : 50
input.voltage (Input voltage (V)) : 235.6
input.voltage.fault (Description unavailable) : 234.3
input.voltage.nominal (Nominal input voltage (V)) : 230
output.voltage (Output voltage (V)) : 235.6
ups.beeper.status (UPS beeper status) : enabled
ups.delay.shutdown (Interval to wait after shutdown with delay command (seconds)) : 30
ups.delay.start (Interval to wait before (re)starting the load (seconds)) : 180
ups.firmware (UPS firmware) : 
ups.load (Load on UPS (percent of full)) : 21
ups.mfr (UPS manufacturer) : 
ups.model (UPS model) : 2000VA
ups.productid (Product ID for USB devices) : 0000
ups.status (UPS status) : OL
ups.type (UPS type) : offline / line interactive
ups.vendorid (Vendor ID for USB devices) : 0001

gbakeman, 2022-12-28:
I should have also asked you to paste what the power reading was, but just from reading your variable output, it looks like your ups provides none of the desired variables for calculating the load as states above, besides the load variable its self. I'm still not sure why we don't use that instead. Do you remember if the load was displayed at around 21%?

MartinKurtz, 2022-12-28:
Yeah that is entirely possible,  at the moment it is showing about 29 to 32 percent load and these are the variables, so this corresponds to the ups.load variable:

nutdev1 (/2000VA/)
battery.charge (Battery charge (percent of full)) : 100
battery.voltage (Battery voltage (V)) : 27.60
battery.voltage.high (Description unavailable) : 26.00
battery.voltage.low (Description unavailable) : 20.80
battery.voltage.nominal (Nominal battery voltage (V)) : 24.0
device.mfr (Description unavailable) : 
device.model (Description unavailable) : 2000VA
device.type (Description unavailable) : ups
driver.name (Driver name) : blazer_usb
driver.parameter.pollinterval (Description unavailable) : 5
driver.parameter.port (Description unavailable) : auto
driver.parameter.synchronous (Description unavailable) : no
driver.version (Driver version - NUT release) : 2.7.4
driver.version.internal (Internal driver version) : 0.12
input.current.nominal (Nominal input current (A)) : 8.0
input.frequency (Input line frequency (Hz)) : 50.0
input.frequency.nominal (Nominal input line frequency (Hz)) : 50
input.voltage (Input voltage (V)) : 239.3
input.voltage.fault (Description unavailable) : 214.5
input.voltage.nominal (Nominal input voltage (V)) : 230
output.voltage (Output voltage (V)) : 239.3
ups.beeper.status (UPS beeper status) : enabled
ups.delay.shutdown (Interval to wait after shutdown with delay command (seconds)) : 30
ups.delay.start (Interval to wait before (re)starting the load (seconds)) : 180
ups.firmware (UPS firmware) : 
ups.load (Load on UPS (percent of full)) : 32
ups.mfr (UPS manufacturer) : 
ups.model (UPS model) : 2000VA
ups.productid (Product ID for USB devices) : 0000
ups.status (UPS status) : OL
ups.type (UPS type) : offline / line interactive
ups.vendorid (Vendor ID for USB devices) : 0001

gbakeman, 2022-12-28:
Ok, thank you. I definitely think we need to take a close look at how WinNUT calculates power usage... I'll have to go back and relearn some electricity formulas, but I think we can do something with the fact that your UPS provides voltage and amperage readings.

FynnleyNeko, 2023-09-19:
I am increadibly sorry for necro-bumping this, seeing as its still open I'll go ahead tho.

I am having the same problem, 19% load being shown, under that 182W. But ups.realpower is giving out the real live power draw as 243W. Why is ups.realpower not used if available to show said wattage under the percent?

gbakeman, 2023-09-19:
Hi @FynnleyNeko , no worries and thank you for providing feedback. Could you please also paste the full output of the UPS variables list? That will help me generate a list of useful parameters for the future. Hopefully this will be a relatively easy issue to address.

FynnleyNeko, 2023-09-19:
Sure, my UPS gives out this:

`battery.charge: 96
battery.charge.low: 20
battery.runtime: 1602
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse ECO 1600
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0463
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: no
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 3
outlet.2.status: on
outlet.2.switchable: no
outlet.desc: Main Outlet
outlet.id: 1
outlet.power: 25
outlet.switchable: no
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02
ups.load: 19
ups.mfr: EATON
ups.model: Ellipse ECO 1600
ups.power.nominal: 1600
ups.productid: ffff
ups.realpower: 243
ups.serial: 000000000
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 0463`

Bluscream, 2023-10-10:
https://github.com/home-assistant/core/issues/86307#issuecomment-1432047387

gbakeman, 2023-10-15:
Everyone,

I'm focusing on addressing this issue, still planning on how to do that. Prior to this however, I think the List_Var_Gui form needs an upgrade so we can properly visualize the variables of a UPS. I'm currently working on this in #109 , feel free to follow that.

FynnleyNeko, 2023-10-18:
> @FynnleyNeko It looks like there's no reason to guess the max power since your UPS reports the current power. Do you feel it will be fine to take your `ups.realpower` variable by its self?

Yep! Verified its spot on with one of those current clamps while hooking the UPS up to HomeAssistant for logging and notifications!

gbakeman, 2023-10-19:
All,

Please download the debug build from [here ](https://github.com/nutdotnet/WinNUT-Client/actions/runs/6580236403?pr=112) once it's done building. Notable changes are that the UPS Load calibration fields have been removed from the Prefs GUI (since they only adjusted the max percentage values of the dial, which made no sense) and also adds in a new power calculation mode selection system and associated calculation algorithms. Please let me know how this works for you. If you have problems, then please attach the full log file to your comment.

FynnleyNeko, 2023-10-19:
Verified working in my case, now reads the same as HomeAssistant and Netdata! Thanks!

gbakeman, 2023-10-20:
@FynnleyNeko Thank you for the quick feedback.

@all I'm going to go ahead and merge this and push out a quick pre-release so I can work on some other issues. If anyone is having further issues with the loal/power display, please re-open the issue with a debug log attached.

closed 2023-10-20 by gbakeman

deajan, 2023-11-03:
I am perplex as with my NUT setup. I have a RHEL 9 Nut server which Winnut is connected to.
Whenever I loose the connection to my UPS on the server, Winnut just shows bogus readings. Why does it show 12V batteries charged at 20% when it can't communicate with the UPS ? Shouldn't it show something like "unknown" ? I understand from the logs that those are "fallback" values, but these aren't just correct and misguide the user.

[image]

Also, if I happen to stop and restart nut-driver via `systemctl stop nut-driver; sleep 5; systemctl start nut-driver`, my UPS model dissappears.

Before:
[image]
After:
[image]

Logs attached to the issue.
[WinNUT-Client-2023-11-03.zip](https://github.com/nutdotnet/WinNUT-Client/files/13248955/WinNUT-Client-2023-11-03.zip)

I've updated WinNUT to latest release to make this test.
Btw, thank you for making winnut ;)

gbakeman, 2023-11-04:
> I am perplex as with my NUT setup. I have a RHEL 9 Nut server which Winnut is connected to. Whenever I loose the connection to my UPS on the server, Winnut just shows bogus readings. Why does it [...]

WinNUT has the unfortunate behavior of supplying fallback values as you've noticed, and this is something I've been meaning to correct. I'm not quite sure what the fix will look like; perhaps gauges will have a "disabled" appearance if a value is unavailable. It looks like I currently don't have an issue tracking this. Feel free to open one, or I may try to address this in #24 .

> Also, if I happen to stop and restart nut-driver via `systemctl stop nut-driver; sleep 5; systemctl start nut-driver`, my UPS model dissappears, as well as the consumed watts:
> 
> Before: [image] After: [image]
> 
> If I happen to disconnect / re-connect to the UPS via the WinNUT interface, readings are okay again.
> 
> Logs attached to the issue. [WinNUT-Client-2023-11-03.zip](https://github.com/nutdotnet/WinNUT-Client/files/13248955/WinNUT-Client-2023-11-03.zip)
> 
> I've updated WinNUT to latest release to make this test. Btw, thank you for making winnut ;)

I have a bad feeling this is the result of me trying to reduce the amount of values that are updated every cycle in WinNUT. Would you please open a new issue report with the exact information above?

Thank you!

MartinKurtz, 2023-11-23:
@gbakeman math seems to check out, as its quite literally just a multiplication of the percentage with a maximum value at 100%. 
You may not even need any efficiency calculation, as you can just enter both the value of what comes out of the UPS or what goes in as the max value however you prefer. The power factor is something potentially very different for every UPS, so i wouldnt hardcode that

gbakeman, 2023-11-23:
Excellent, thank you for verifying!

PR #112 (2023-10-20): Correct UPS Load display
The UPS Load dial presents conflicting information, depending on the UPS being monitored. Typically, the load percentage and real value do not line up with what the maximum possible load is. This PR will aim to correct this by:

- Disabling post-initialization modification of the UPS Load dial besides setting displayed values. It's a gauge of a percentage, which is always between 0 and 100 percent so there's no need to modify it outside of the Designer (unless the user wants to change dial tick settings - open to discussion)
- Similarly remove references to the `MinUPS [... cut]