nutdotnet/WinNUT-Client #46: how to run winnut as a windows service

vnoder, 2022-10-23:


gbakeman, 2022-10-23:
We currently don't have the capabilities for running WinNUT as a service. I know it's been on the planning board for quite some time, but I believe it would take quite a bit of work to implement as a feature. I'll leave this issue open, but I'm not sure this is in the future for current WinNUT. I definitely think service operation will be a part of the next generation of WinNUT though.

timedge5, 2023-02-10:
Hello! I have been trying to implement WinNUT as a service using NSSM - a nice program that has worked great in the past for converting apps to services. However, while it seems that while NSSM runs the program fine, none of the settings that I applied are used. When I run the app normally, I can setup something like shutdown on 90% for testing, then minimize it to tray and see it shutdown when UPS hits that. However, if I the exit the app and start the service I can see the app running in the background processes but when the UPS drops below 90% again it doesn't shut down the server. I have tried to find an ini or config file but can't find anything referencing the settings. Where would I find a file that shows the settings being applied in the client? Thanks!

gbakeman, 2023-02-10:
Hi @timedge5 ,

WinNUT stores its settings in the current user's registry hive, so I wonder if NSSM is trying to run WinNUT as a different user and therefor not loading the settings you're creating. I'm not aware of any facilities we currently have to (safely) move settings into the system hive (where NSSM is probably looking for them), but you could try copying the `HKEY_CURRENT_USER\SOFTWARE\WinNUT` hive to its respective location in the system registry, and then try running it as a service. Note that any saved password will likely become invalid when accessed by a different user.

See also

- https://github.com/gawindx/WinNUT-Client/issues/156
- https://github.com/gawindx/WinNUT-Client/issues/54
- https://github.com/gawindx/WinNUT-Client/issues/42

timedge5, 2023-02-11:
@gbakeman Nice that was it. As soon as I point the service to the user that the settings were created on, it runs perfectly. Thank you! This could obviously now be done with setting up a service account and making sure the reg entries are there. Onwards!

hartez, 2026-11-10:
I knocked together a Windows service implementation using [NUTDotNet](https://github.com/nutdotnet/nutdotnet). The code is at https://github.com/hartez/WinNUT-Service, if anyone is looking for a starting point for implementing their own service.