nutdotnet/WinNUT-Client #47: Mass amount if files installing with WinNUT-Setup.msi

DravenSA, 2022-10-23:
So after installing the latest Pre-Release v2.2.8328 I am unhappy to report that there are at least 80 unwanted files in the WinNut-Client Directory. 
I am assuming this is mistake and not a nasty what to get into someones PC 
[image]

DravenSA, 2022-10-23:
Uninstalling WinNut from within Add remove Programs, does not remove WinNut From the system. nor does it remove all there horrible unwanted useless files.

gbakeman, 2022-10-23:
Hi @DravenSA ,

> So after installing the latest Pre-Release v2.2.8328 I am unhappy to report that there are at least 80 unwanted files in the WinNut-Client Directory. I am assuming this is mistake and not a nasty what to get into someones PC

You're right, there's nothing nefarious about those files. Related to #29; there were some hacks in WinNUT to get Toast Notifications working on Windows 10 PCs. I wanted to get WinNUT building on GitHub infrastructure instead of my own PC (which comes with its own benefits of security and integrity), but this necessitated replacing the hacks with a NuGet package [Microsoft.Toolkit.Uwp.Notifications](https://www.nuget.org/packages/Microsoft.Toolkit.Uwp.Notifications). I was not paying attention and honestly didn't know better at the time, but it appears that a consequence of this was that many dependencies were added to the installer. These files are not present in the non-installer releases on WinNUT, which makes me think that these are indeed unnecessary. I'm currently modifying the installer to see if I can push out a correction.

> Uninstalling WinNut from within Add remove Programs, does not remove WinNut From the system. nor does it remove all there horrible unwanted useless files.

As you can probably see, I was having some trouble with the installer builds! 😅 There were a few pre-releases recently where I did not correctly update the program version, which caused the installer to either error out or install duplicate entries of WinNUT. I've since corrected that, and to make sure, I've just uninstalled WinNUT from my system and verified that the WinNUT directory is removed from `C:\Program Files (x86)`. I would suggest that you uninstall every entry of WinNUT that you have in your programs list, manually remove the application directory from your program files folder, then wait while I push out an update soon that should correct the installer and prevent those dependencies from being installed.

Apologies for the file spam and I'd like to ask for your patience while I prepare a fix.

DravenSA, 2022-10-23:
No problem I will use the Debug one for now, LOL the one that works with the sleep function, that we so tirelessly tested :-) So i Still have that, I will use it until the new installer is ready.
Thanks for the understanding and more so for the explanation, much appreciated.

gbakeman, 2022-10-23:
New pre-release [is up](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.2.8331), please give it a try and let me know how it goes on your end. Thank you again for the report.

DravenSA, 2022-10-24:
Installs without all the extra files but got this error
[image]
Just clicked Ok and it still went through, and the shortcut still appeared on the desktop and in the Start menu (windows 10)

gbakeman, 2022-10-30:
Hi @DravenSA,

That's weird, I can't reproduce that error on my system. Would you please try uninstalling and re-installing the latest pre-release?

DravenSA, 2022-10-31:
So i uninstalled, restarted, and re-installed, and this time didnt get the error, so not sure why, i got it the first time. But it is all good.

gbakeman, 2022-10-31:
Good to hear, thank you.

closed 2022-10-31 by gbakeman