nutdotnet/WinNUT-Client #60: Crash on launch

maiko, 2022-11-19:
Hi there,

Update to latest pre-release. 
Crash on launch. 

`Exception type: System.MissingFieldException
Exception message: Champ introuvable : 'WinNUT_Client_Common.WinNUT_Globals.ApplicationData'.
Exception stack trace:
   à WinNUT_Client.WinNUT.WinNUT_PrefsChanged(Boolean isChanged)
   à WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   à System.EventHandler.Invoke(Object sender, EventArgs e)
   à System.Windows.Forms.Form.OnLoad(EventArgs e)
   à System.Windows.Forms.Form.OnCreateControl()
   à System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   à System.Windows.Forms.Control.CreateControl()
   à System.Windows.Forms.Control.WmShowWindow(Message& m)
   à System.Windows.Forms.Control.WndProc(Message& m)
   à System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   à System.Windows.Forms.Form.WmShowWindow(Message& m)
   à System.Windows.Forms.Form.WndProc(Message& m)
   à System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   à System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   à System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
`

maiko, 2022-11-19:
If I uninstall then reinstall WinNUT, I get this : 

[image]

The window stay on screen for a second before closing by itself without showing any traces.

maiko, 2022-11-19:
Latest working version on my computer appear to be this one 

[image]

Édition	Windows 11 Professionnel
Version	22H2
Installé le	‎03-‎10-‎22
Build du système d’exploitation	22621.819
Expérience	Windows Feature Experience Pack 1000.22636.1000.0

gbakeman, 2022-11-19:
Hi, thank you for the report. Does build 8331 not work on your system then?

maiko, 2022-11-19:
> Hi, thank you for the report. Does build 8331 not work on your system then?

Hi ! Thanks for the quick reply :) 
Sorry, I skipped build 8331.
I just tested it and it work. Only the latest seem to be non working. 

[image]

gbakeman, 2022-11-19:
Just saw you updated the picture, looks like it was working. I'll take a deep dive into this a little later. I think I know what change is causing this, but why is an entirely different issue! Are you running WinNUT installed as normal, or the non-installed version?

maiko, 2022-11-19:
> Just saw you updated the picture, looks like it was working. I'll take a deep dive into this a little later. I think I know what change is causing this, but why is an entirely different issue! Are you running WinNUT installed as normal, or the non-installed version?

I use it installed as normal.

maiko, 2022-11-19:
Interrestingly, the non installed version is working. 
[image]

If I compare the non-installed and the installed directory, it appear that some DLLs were not present (selected on the following screenshot) 
[image]

If I copy over the DDLs from the non-installed to the installed directory, I can now launch and use build 8356
[image]

gbakeman, 2022-11-20:
Thank you for investigating this for me. It looks like the installer compiled with some missing dependencies. Could you please try uninstalling, then reinstalling from the fixed installer I've just uploaded to confirm that everything is ok now?

maiko, 2022-11-20:
> Thank you for investigating this for me. It looks like the installer compiled with some missing dependencies. Could you please try uninstalling, then reinstalling from the fixed installer I've just uploaded to confirm that everything is ok now?

Worked great! 
Thanks. 

I'll close the issue =)

closed 2022-11-20 by maiko