nutdotnet/WinNUT-Client #61: Log files still being saved in system root folder

boblo44, 2022-11-23:
It seems that WinNUT log files are now being saved in a folder named <system root>:\Logs instead of at AppData\Roaming as originally happened. Am running WinNUT client v 2.2.8359.21169 on Win10Pro 22H2 Build 19045.2251. Current log is C:\Logs\WinNUT-Client-2022-11-23.log on my system, previous log was C:\WinNUT-Client\Logs\WinNUT-Client-2022-11-23.log.

gbakeman, 2022-11-23:
That's been a persistent issue for sure. Could you please attach a log file, and I'll look into this further?

gbakeman, 2022-11-23:
I can also reproduce this on my end, including in Debug build mode. I'll have some more core changes up soon.

gbakeman, 2022-11-24:
When you get a chance, please download [this debug build](https://github.com/nutdotnet/WinNUT-Client/suites/9478908215/artifacts/449134315) and run it in a temporary location, just to verify that it writes next to the executable. You may have to cleanup your C drive a little, sorry about that.

Once it looks good to you, I'll make a pre-release so it functions normally and writes to the appdata directory.
A

boblo44, 2022-11-24:
I’ve downloaded and unpacked the debug build but not sure how to get it to run. I’ve copied the contents of the .zip to a temporary location on my C: drive but double-clicking the WinNUT.exe has no effect even with Norton AV disabled. Please advise…

 

Bob

gbakeman, 2022-11-24:
That's a little odd... as far as where to run it, I was thinking you could extract it to a folder within your downloads folder, like `User/Downloads/WinNUT-Client`. Could you try that and see if it makes any difference?

boblo44, 2022-11-24:
No problem, I fixed it by unblocking all the files in the download – Windows security had blocked them all! You’ll be pleased to know that the log is now being written to the folder containing the WinNUT client: C:\WinNUT-Client-debugbuild-v2.2.8363.2520\WinNut-Client\Logs\WinNUT-Client-2022-11-24.log in my case.

 

Also, did you test the shutdown process? A while ago I tried running my UPS on battery until it died and didn’t get a smooth shutdown. I’ll play around with that some more before I flag it as an issue.

 

B

gbakeman, 2022-11-24:
Thank you for confirming that!

Battery and power events have been tweaked several times so I think you should see better handling of that. Please let me know if you have any issues though.

I'll merge the changes and consider this issue closed, and a pre-release build should be out next week.

PR #62 (2022-11-28): Attempting to finally solve data directory oddities
Judging from my Debugging output folder, it looks like generating the application's data directory has been broken since after November 11. Not quite sure what commit did it, but here's a summary of what's in this PR:

- General reorganization of the `WinNUT_Globals` Module.
  - Altered some constants to be more consistent, made shortcut properties and general static strings defined within the module itself.
  - As a result, the Init_Globals function is one line.
  - This calls the new GetAppDirectory function. It returns the best directory f [... cut]
closed 2022-11-28 by gbakeman