nutdotnet/WinNUT-Client #63: Does not re-establish connection, but says it has...

f33d13ack, 2022-11-25:
This is the first version I've ever downloaded of the Windows NUT Client, so I'm a new, here. 
I've currently been using an older, WinNUT 2.0.0.4a, for the last few years, which is installed as a service.

v2.2.8359.21169, Windows 10 Pro, 21H2 Build 19044.2251

If I disconnect or disable the ethernet adapter and reconnect it again, or when my VPN starts if the NUT already had an established connection,
Then, once the NUT tries to re-establish connection, and says that it HAS re-established connection...  it really hasn't, and the yellow reconnection icon stays yellow, and also the gauges stay frozen (Pic1), and if I try to get the UPS variable at this time, it throws back an error (Pic2).
I've also tested a power failure and it does not respond, unless I click to Reconnect manually, then the power failure will trigger.

If I click to Reconnect the connection after the VPN is connected... it's fine,
If I disconnect the VPN... it's also fine, as long as I don't disconnect the ethernet connection,
If I restart the VPN... it shows the yellow reconnect icon, says it's established, but again.. it actually hasn't.

Pic1
[image]

Pic2
[image]

gbakeman, 2022-11-28:
Hi @f33d13ack ,

Thank you for the report. I think there may be several bugs at play here. One is that the UI was not properly synchronized with the state of the UPS when the connection is lost. Two is that a variable was left in an error state when the connection was broken. I was just testing WinNUT now and able to reproduce the issues you reported. I think I have the two primary bugs fixed now. Please download the [latest debug build here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/3568196279) and extract/run it in a temporary location, and let me know if you can still reproduce the issues here. If so, please upload a log.

f33d13ack, 2022-11-30:
WinNUT-Client-debugbuild-v2.2.8367.36037

Hi @gbakeman , 

This build seems to be working as intended.  I'm unable to reproduce the issue.

Question though, in the event that the network adapter bugs out and disconnects, which sometimes happens on UDP or TCP protocols in my VPN, is there a way you can add an option in WinNUT, to try to reestablish the connection indefinitely, maybe silently in the background, like every 5 or 10 seconds, rather than only 30 tries?  If I disconnect the network adapter, let WinNUT try 30 times, then reenable the adapter, I have to manually reconnect.  It seems like each try only lasts a second or two, so if I'm not around on the odd occasion this might occur, it's not going to restart unless it's within those 30 tries.  Just a thought.

gbakeman, 2022-11-30:
Glad to hear it, thank you for reporting back!

> is there a way you can add an option in WinNUT, to try to reestablish the connection indefinitely

That's a good idea, I don't see why that shouldn't be an option. I'll make a separate issue for that and see what I can do.

PR #64 (2022-11-30): Improve connection loss handling
Closes #63 

A few issues have been brought up:

- The WinNUT UI is left in an inconsistent state when the connection to a NUT server is unexpectedly lost.
- The `streamInUse` latch variable was left locked when the connection was broken, resulting in future queries failing.
closed 2022-11-30 by gbakeman