nutdotnet/WinNUT-Client #65: Smarter/Quieter/Indefinite Reconnection attempts

gbakeman, 2022-11-30:
It would be useful if WinNUT could attempt to reconnect to the NUT server indefinitely. This would mean that it is quieter about the reconnection attempts. If the tools are available, then it may also be useful if WinNUT can first detect if it's even possible to make a connection first (network adapter is up & route to the NUT server is available), otherwise wait until the conditions are present.

Ref #63
@f33d13ack

s0uls, 2023-01-23:
Hello.

My nas is the UPS master and it auto powers off during night until 9 am to preserver HDD lifespam.
My Desktop PC using win nut client will time out and not reconnect automatically when the UPS is back at 9 am.
Would it be posible to add a connection schedule or a manual time out override ?

gbakeman, 2023-02-05:
> Hello.
> 
> My nas is the UPS master and it auto powers off during night until 9 am to preserver HDD lifespam. My Desktop PC using win nut client will time out and not reconnect automatically when the UPS is back at 9 am. Would it be posible to add a connection schedule or a manual time out override ?

I'm definitely thinking that, at minimum, we have an option (by default) where WinNUT will just indefinitely attempt to make a re-connection to the NUT server if it was previously connected successfully (connection lost event) without any kind of notification. The last item in the log would be a notification that ongoing connection attempts are occurring. How important is scheduling to you? Perhaps we can just do a simple ping instead if you want lower overhead.

s0uls, 2023-02-06:
> I'm definitely thinking that, at minimum, we have an option (by default) where WinNUT will just indefinitely attempt to make a re-connection to the NUT server if it was previously connected [...]

No, schedules are not a must, but as you said indefinitely attempts would bee great.

bernatcg, 2023-02-25:
In my opinion this kind of software should be always connected, and if there is no connection for any reason, keep reconnecting all the time. Maybe the time between tries should be dynamic rather than a fixed interval, but it should be able to reconnect no matter what. This kind of software should be designed to work unattended to work as expected.

yoyoma2, 2024-03-31:
I vote for the **Quieter** part of this issue. My WinNUT client connects to a rpi which hosts the UPS. While the rpi is performing backups (about 1 hr) most services are stopped including NUT services. I have to exit WinNUT before the rpi backup and restart WinNUT after the rpi backup is done and NUT services are available again.

It would be great if a single "Not Connected" and a single "Connected" notification appeared an hour later.

gbakeman, 2024-07-25:
Hi all,

Got some changes for you. You can see a summary in #171, but in short, WinNUT now tries to stay connected indefinitely. You can find a test build [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/10100965530?pr=171), please give it a try and let me know how it goes.

kirsch33, 2024-08-01:
@gbakeman 

I've been running this for a week with no issues. awesome work

gbakeman, 2024-08-01:
@kirsch33 Thanks for the extended test! I'll probably have a pre-release by next week.

yoyoma2, 2024-08-04:
Tried v2.3.8981 while backing up the rpi (nut server) as described in my comment above. It is almost perfect now. Nice work!

On Win10 I got a single notification when the nut server stopped. "Reconnection In Progress\nReconnection..."
To be perfect a second notification an hour later when the nut server restarted would be nice.

When I look at the recent messages drop down list in WinNUT Client there are three messages.

1. Lost Connect to rpi:3493  (should it say "Lost Connect**ion** to rpi:3493"?)
2. Reconnection In Progress  (should the notification say "Lost Connection: Reconnection In Progress" as no "Connection lost" notification was given?)
3. Connection to Nut Host rpi:3493 Established (should this GUI message also be sent as a Windows notification?)

gbakeman, 2024-08-04:
Hey @yoyoma2 thanks for the feedback on this. I've tried artificially disabling my network adapter and re-enabling it a couple of times to simulate the network (or server) going down, and I'm able to see WinNUT display the "Connected" notification when it reconnects. Can you confirm that it's not sitting in your notification drawer?

[image]

If that isn't the case, could you please upload a debug log where you reproduce this scenario?

yoyoma2, 2024-08-04:
Here's what I see on Win10. Only one notification for the entire backup process starting at 7:42 AM ending an hour later.  It was a little surprising to see "reconnecting" without any "Lost connection" text.
[image]

Points of possible interest:

- I'm on Win10 not Win11
- The RaspiBackup script stops the nut server service during the backup but the ethernet connection to the rpi is never broken (different code path than disabling network adapter)
- The WinNUT client GUI has the "Connection to NUT Host rpi:3493 Established" message precisely after the backup but no Windows notification at all
- My "Create LogFile" setting was off (Next backup I'll enable the log file)
- The Miscellan**e**ous tab title has a typo (missing 'e')

So things are actually working perfectly but the re-connection notification is missing.

yoyoma2, 2024-08-15:
I backed up the rpi (which stops the NUT server for an hour) with Debug level log enabled in the WinNUT client. The whole time WinNUT client was closed to systray. Here's are the notifications:
[image]

Description of log:

- Line 2156 (2:45:37 PM the backup starts and we get a reconnection in progress notification)
- Line 4414 (3:47:25 PM the backup ends and there is no notification) **This is bad!**
- Line 8710 (4:08:13 PM I manually turn off logging and Not Connected, Connected notifications appear)

Here is the log.
[WinNUT-Client-2024-08-14.zip](https://github.com/user-attachments/files/16619825/WinNUT-Client-2024-08-14.zip)

The connection being reestablished is somehow not being reported even though it is happening.

gbakeman, 2024-08-23:
Thank you for your detailed work on this. That's very interesting since it looks like WinNUT logging a connected event and even preparing the NotifyIcon (the icon that sits in the Windows taskbar), but the toast just doesn't seem to be triggering. I've created a new issue in #182 so let's continue conversation over there. I've also created a [debug build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/10531599982?pr=183), please repeat the test just as you did now. Thank you for your efforts here!

kirsch33, 2025-06-02:
Just following up. I've been running that test build almost a year and no issues whatsoever. 

This has fixed the only issue I ever truly needed fixed so I'll probably just keep this test build going into oblivion :)

gbakeman, 2025-11-02:
All,

Thanks again for your help on this. [A new Pre-Release build](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.9437) is out that I hope fixes this. I'll close this assuming all is well, but please comment back if issues persist.

PR #171 (2024-08-03): Persistent Connection attempts, quieter notifications
- Remove reconnection attempt limit; WinNUT now attempts to reconnect indefinitely.
- Tweaked the Connection menu; it now has a convenient toggle for persistent connections (auto reconnect setting), renamed the Reconnect command to Connect, and the Connect/Disconnect commands are now enabled or disabled depending on the state of WinNUT.
[image]
- Small tweak to how the WinNUT window title is updated (should be more consistent.)
- Overall notification spam when connecting, reconnecting and disconnecting is more manageable.

Closes #65, also  [... cut]
closed 2025-11-02 by gbakeman