nutdotnet/WinNUT-Client #66: Auto Reconnect testing failures

SSPW, 2022-12-08:
had reconnect issues with old repo version, see https://github.com/gawindx/WinNUT-Client/issues/162

moved to this repo, testing WinNUT-Client-v2.2.8356-no_install on win10

testing auto reconnect for network path break or server restart
for the sake of this issue, i pulled a uplink cable between two switches in client/server network path

with this new version, there are reconnection attempts, and reconnection success, ...

but after auto reconnect success in winnut, there is no periodic traffic.
wireshark shows only shows 4 frames at reconnect.
debug level logs, also show no periodic attempt to get server data reconnect

manually, using menu disconnect and reconnect, reestablished normal periodic traffic flow

i can provide wireshark and debug level logs, if required.

gbakeman, 2022-12-08:
Thank you again for opening an issue here.

It sounds like this is related to #63 , could you please try the latest debug build I've posted in there and see if it changes anything?

SSPW, 2022-12-09:
the latest debug version does fix the reconnection issue i was experiencing. thx

closed 2022-12-09 by SSPW