nutdotnet/WinNUT-Client #67: Portable version uses wrong path for "Start with Windows"

hugalafutro, 2022-12-10:
I have portable WinNUT installed in `C:/_portable/WinNUT`, but when I check the Start with Windows it adds path to the regularly installed version in Program Files.

[image]

gbakeman, 2023-02-05:
Hi @hugalafutro ,

This is odd because that option only adds the path of the currently running executable so it shouldn't matter where it's installed. Do you have two versions of WinNUT installed on your PC (one in the Program Files location, and the other in your `_portable` directory?) Could you try toggling the Start with Windows option to see if it resets the path in the registry?

hugalafutro, 2023-02-06:
Hi it would appear to be leftover form previously installed regular version, although I made sure to uninstall it properly before switching to portable. Anyways, unticking the option removes the key form registry, re-ticking it adds it back in with correct path. So it was probably already there and thus it didn't get changed.

closed 2023-02-06 by hugalafutro