nutdotnet/WinNUT-Client #68: No data from Master

DarkCorner-IT, 2022-12-15:
I have an APC Back-UPS BX2200 connected and configured in pfSense as "Master".
I downloaded WinNUT-Client version 2.2.8356.28290, but it doesn't fetch data from Master.
All ports are open, so the problem is elsewhere.
I also tried to put the pfSense credentials.
Where am I doing wrong?

gbakeman, 2022-12-15:
Hi,

Please provide debug logs so I can assist further.

DarkCorner-IT, 2022-12-15:
Where is log?

gbakeman, 2022-12-15:
If you open the options menu, I think on the miscellaneous tab is a button to open the log file.

DarkCorner-IT, 2022-12-15:
```
15/12/2022 16:20:21 Pid: 14604 Logger : Log file is initialized at W:\Logs\WinNUT-Client-2022-12-15.log
15/12/2022 16:20:21 Pid: 14604 WinNUT : Parameter Dial Voltage Battery Need to be Updated
15/12/2022 16:20:21 Pid: 14604 WinNUT : Parameter Dial Voltage Battery Updated
15/12/2022 16:20:21 Pid: 14604 WinNUT : WinNut Preferences Applied.
15/12/2022 16:20:21 Pid: 14604 String : New Log to CB_Current Log : WinNut Preferences Changed
15/12/2022 16:20:23 Pid: 14604 WinNUT : Main Gui Has Focus
15/12/2022 16:20:23 Pid: 14604 WinNUT : Update Icon
15/12/2022 16:20:25 Pid: 14604 WinNUT : Force Reconnect from menu
15/12/2022 16:20:25 Pid: 14604 UPS_Device : Processing request to disconnect...
15/12/2022 16:20:25 Pid: 14604 UPS_Device : NutSocket raised Disconnected event.
15/12/2022 16:20:25 Pid: 14604 WinNUT : Update all informations displayed to empty values
15/12/2022 16:20:25 Pid: 14604 WinNUT : Update Icon
15/12/2022 16:20:25 Pid: 14604 WinNUT : Status Icon Changed
15/12/2022 16:20:25 Pid: 14604 WinNUT : New Icon Value For Systray : 1216
15/12/2022 16:20:25 Pid: 14604 WinNUT : New Icon Value For Gui : 1152
15/12/2022 16:20:25 Pid: 14604 WinNUT : NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
15/12/2022 16:20:25 Pid: 14604 WinNUT : Battery Status => Unknown
15/12/2022 16:20:25 Pid: 14604 WinNUT : Disconnected from Nut Host
15/12/2022 16:20:25 Pid: 14604 String : New Log to CB_Current Log : Disconnected from Nut Host
15/12/2022 16:20:25 Pid: 14604 WinNUT : Client UPS_Connect subroutine beginning.
15/12/2022 16:20:25 Pid: 14604 UPS_Device : Beginning connection: @192.168.101.1:3493, Name: BackUPS_BX2200MI [AutoReconnect]
15/12/2022 16:20:25 Pid: 14604 Nut_Socket : Attempting TCP socket connection to 192.168.101.1:3493...
15/12/2022 16:20:46 Pid: 14604 WinNUT : Something went wrong connecting to UPS BackUPS_BX2200MI. IsConnected: False, IsAuthenticated: False
15/12/2022 16:20:46 Pid: 14604 String : New Log to CB_Current Log : Connection to Nut Host 192.168.101.1:3493 Failed: Impossibile stabilire la connessione. Risposta non corretta della parte connessa dopo l'intervallo di tempo oppure mancata risposta dall'host collegato [::ffff:192.168.101.1]:3493
15/12/2022 16:20:46 Pid: 14604 WinNUT : Main Gui Lose Focus
15/12/2022 16:20:46 Pid: 14604 WinNUT : Update Icon
15/12/2022 16:20:54 Pid: 14604 WinNUT : Main Gui Has Focus
15/12/2022 16:20:54 Pid: 14604 WinNUT : Update Icon
15/12/2022 16:21:00 Pid: 14604 WinNUT : Open Pref Gui From Menu
15/12/2022 16:21:00 Pid: 14604 Pref_Gui : Load Pref Gui
15/12/2022 16:21:00 Pid: 14604 WinNUT : Main Gui Lose Focus
15/12/2022 16:21:00 Pid: 14604 WinNUT : Update Icon
15/12/2022 16:21:00 Pid: 14604 Pref_Gui : Pref Gui Opened.
15/12/2022 16:21:03 Pid: 14604 Pref_Gui : Setting LogControl statuses.
15/12/2022 16:21:09 Pid: 14604 Pref_Gui : Show LogFile
```

gbakeman, 2022-12-15:
The username and password fields are NUT specific and not related to pfsense.

The error:

> Impossibile stabilire la connessione. Risposta non corretta della parte connessa dopo l'intervallo di tempo oppure mancata risposta dall'host collegato

Roughly translates to

> Unable to establish connection. Incorrect response from the connected party after the time interval or no response from the connected host

The error above says that the NUT client is unable to make a connection to the NUT server. You'll need to do some troubleshooting between your host and the WinNUT server as this issue appears to be beyond WinNUT.

DarkCorner-IT, 2022-12-15:
On pfSense there is only one user and the password is correct. If I open an SSH session, I successfully get to his console.
On the other hand, as I said above, it is not clear why to query a port I have to log into the firewall administration.

DarkCorner-IT, 2022-12-15:
Perhaps fingerprinting with login is needed, but I don't know how to authorize it with this program.

gbakeman, 2022-12-15:
NUT server does not use your pfsense login, these are two different things. The NUT credentials are stored in a configuration file along with other parameters for the NUT server. You'll need to refer to the documentation for pfsense's NUT package to find that.

All of this is beside the point, though. WinNUT is indicating that it cannot communicate with the NUT server. This is before usernames and passwords even come in to play. You need to do some troubleshooting to understand where the problem lies.

- On the pfsense host, verify upsd is up and listening on the correct network interface
-  telnet 127.0.0.1:3493 to verify the NUT server is accepting connections and responding to commands
- On your PC, scan port 3493 on pfsense to verify it is open

DarkCorner-IT, 2022-12-16:
On pfSense the port 3493 test is open.
On Win10 Pro I disabled Windows Defender
Also I'm using in WinNUT-client both admin and local-monitor user with their passwords found in upsd.users file.

gbakeman, 2022-12-16:
Please also try opening a telnet session from your client (Windows 10 Pro) to the pfsense host on port 3493 and let me know what response you get from sending random characters.

DarkCorner-IT, 2022-12-18:
The Master configuration was incorrect.
Now WinNUT-Client detects the status of the UPS. Many thanks.

How do I now run a Windows batch file when an event occurs?
For example the UPS goes on battery or when the battery has reached a low charge level

gbakeman, 2022-12-19:
I'm glad to hear it was fixed!

We currently don't support running external commands when an event happens. I'm hesitant to add new features since I'd like to make a new version of WinNUT, but feel free to open a new issue as a feature request for custom events.

closed 2022-12-19 by gbakeman