nutdotnet/WinNUT-Client #69: Ability to overwrite UPS value (Enhancement)

MartinKurtz, 2022-12-27:
Basically, the ability to tell WinNUT that my UPS can output 1400 watt in my case, so that it can calculate an actual wattage from the percentage and this. Some other values could also be useful, but this one came to mind, as i just had a power outage.

gbakeman, 2023-02-05:
Thanks for the report @MartinKurtz . Just want to link a few related issues here: #24 , and I think your #41 report may accomplish the same thing as this? Let me know if these are actually different issues and I'll reopen.

MartinKurtz, 2023-02-07:
I dont think #24 is related, as this and #41 is showing the wrong wattage, not autodetecting ranges and #41 is the issue this is a cheap solution for, so completely up to you

gbakeman, 2023-02-07:
Good points, overwriting UPS values could be useful outside of those issues too. I'll reopen this but will try to focus on the earlier issues first, hopefully then I'll get to this one later.

Bluscream, 2023-10-10:
Any news on this?

gbakeman, 2023-10-15:
Hi @Bluscream , nothing yet unfortunately, although this is turning out to be a popular issue. I've just fixed a couple of other functionality-breaking issues that I may push out a release for first, then I think I'll come back and prioritize these issues.