nutdotnet/WinNUT-Client #70: PC not shutting down properly

MartinKurtz, 2022-12-27:
I have recently had a power outage and my pc was on. During this i observed, that the UPS battery charge dropped below the 50% i specified to shut down my computer without the PC actually shutting down.

The charge got down to 38% and no shutdown was initiated.

I have however recently upgraded to a newer version of WinNUT, but i dont think youve done any work on that part of WinNUT since then.

[image]
These are the settings used

MartinKurtz, 2022-12-27:
Sadly testing isnt something i can do too often, as some devices that are on my UPS need to be running usually and i cant just shut them down quickly and power them back on, as they have pretty complex shutdown procedures and turning everything back on is a pain in the arse and takes a while, but whan i get a chance or there is a outage, i will let you know what i find with thew new version

gbakeman, 2022-12-28:
This sounds like something that was fixed recently, but it would helpful if you could retest on the latest pre-release and capture logs to confirm that the issue is present. You don't need to shutdown your entire UPS to test this - you can set the percentage to something like 99% so only your PC powers off.

MartinKurtz, 2022-12-28:
I set it to 100% and that did work within the 30 second window before things start shutting down, so it seems that this was fixed indeed

closed 2022-12-28 by MartinKurtz