nutdotnet/WinNUT-Client #73: Cyber Power CP600 Not Recognized

AlanEisen, 2023-01-19:
I've installed WinNut (2.2.8359.21169) on my Windows 10 machine. I'm connecting it to a NUT server running on Asustor NAS. The NAS is USB connected to a CyberPower CP600. I can connect to the NAS, but the Manufacturer; Name; Serial; Firmware are blank. Additionally, the various sensors are incorrect. I tried to calibrate them, but not clear that that fixed anything, particularly the Battery Charge which is showing 20%.

I had just replaced the battery; it's been plugged in for a while. I used the HomeAssistant integration and it connected quickly; recognizes the model; and shows as fully charged (100%). 

I can't figure out if there's a configuration problem on windows or if it's just that the software doesn't recognize this model.

Thanks,

Alan

gbakeman, 2023-02-05:
Thank you for the report @AlanEisen. Would you please first update to the latest pre-release version that was just released today, then provide output from the Variables List window and a short log captured while running WinNUT?

AlanEisen, 2023-02-07:
Thank you for the response. I'm pretty new to Github and do not see how to install v2.2.8436. I can download the code files, but don't see an install option. Can you provide some guidance?

AlanEisen, 2023-02-07:
Apologies . . . finally figured it out.  I'm now at 2.2.8436.29819. I don't see any change in the dials. When I go to the Variables List window I get an Error.  "Error encountered trying to get variables from the UPS: UNKNOWNUPS (ERR UNKNOWN-UPS) Query: LIST VAR ASUSNAS"  

[image]

gbakeman, 2023-02-07:
It sounds like the name of the UPS specified is incorrect and therefor can't find it on the NUT server. Double check and make sure the name is spelled correctly as it's specified in the ups.conf file on your NAS (if you can find it.) We definitely need to communicate an error like this better so I'd consider this a feature request for sure.

One other thing I'm seeing - it looks like your gauges are calibrated with fairly tight mins and maxes. I'd suggest adding about +/- 10% on either end for the voltages and frequencies.

AlanEisen, 2023-02-07:
Thanks. I've tried many variations of the UPS and couldn't find one to work. (I had the name of the NAS network server in there.)  I could not find the ups.conf file on the NAS. So I'm not sure where to go from here.

I did adjust the gauges.

Thanks,
Alan

gbakeman, 2023-02-07:
If you can, open a telnet session to your NUT server (another client like PuTTY will work great) and run the command LIST UPS, this should show you what name you need.

AlanEisen, 2023-02-07:
I did that from the home directory and didn't find a UPS file at all.

AlanEisen, 2023-02-07:
I've configured the NUT integration into HomeAssistant. That connected immediately. Is there information from that connection that would help configure WinNUT?

gbakeman, 2023-02-07:
Potentially, I'm not familiar with home assistant. All you're really looking for is the name given to the UPS.

To be clear, what you'd need to do is use a terminal client like telnet, and connect to the NUT server at its correct port (3493 IIRC?) and then run the command LIST UPS.

AlanEisen, 2023-02-08:
I tried that but I'm not finding anything.

gbakeman, 2023-02-08:
A couple of things:

- You need to connect directly to the NUT server/upsd (not the machine that's running NUT), so it would be _your ip:3492_ (I'm guessing at the port number, can't remember off of the top of my head)
- The correct command you need to run is `LIST UPS`.

AlanEisen, 2023-02-09:
I used PUTTY to try to access the NUT server (IP addr/3493). However, I never got any response. I tried an SSH, Telnet, rLogin, and BareSSH. One of those accepted commands, but never recognized list or its variants.

gbakeman, 2023-02-09:
Sorry for the difficulties! I have to wonder if ASUS may be using a customized version of upsd that's breaking a few things. I've seen Synology do the same although the differences weren't really noticeable.

So just to be totally clear, here's the config I use to connect to my test upsd:

[image]

Then when the terminal opens (you won't get a welcome message), you run the command as follows and should see some output like this:

[image]

**An important thing to note** (and I didn't realize this before), but commands are _case sensitive_.

gbakeman, 2023-02-09:
Odd... Can you run `VER` for me? At the very least that should tell us what we're dealing with here.

AlanEisen, 2023-02-09:
VER gives me the same "Unknown Command" error.

gbakeman, 2023-02-09:
Wow. Ok, sounds like we're dealing with something fairly custom.

I've just come across [this](https://www.reddit.com/r/asustor/comments/rulq9b/the_network_ups_support_in_adm_sucks/) discussion which may help. It sounds like the UPS name is supposed to be _asustor_, and server config files can be found in either `/etc/ups` or `/usr/etc.base`. Not sure if actually changing the config files changes anything, though.

Let me know if that helps.

AlanEisen, 2023-02-10:
Wow! That really helped. I changed the name of the UPS and immediately got recognized - with the correct battery charge. Now I need to finish configuring WinNUT.

[image]

gbakeman, 2023-02-10:
Glad to hear it's working! Again, sorry for the trouble.

I think one take away from this is that WinNUT should better communicate what the issue is - the fact that the UPS name was unknown/incorrect was not immediately obvious, and should probably be front and center to the user. Does this sound right based on your experience?

AlanEisen, 2023-02-11:
I think that makes sense, if it's clear to the software that that is the issue.

Thank you so much for your assistance diligence in resolving this problem.

Alan