nutdotnet/WinNUT-Client #76: Hypervisor Awareness

GitRDone2022, 2023-01-31:
Feature Request.

Hoping to be able to identify which hypervisor a VM is on while the software is running.  At specific intervals, detect the hypervisor name and switch to the correct NUT server.  vmware/esxi, proxmox/kvm, xen/xcp-ng, hyper-v, etc. could auto migrate the VM to a different host or data center with different UPS.  Maybe, the software would need a list of hypervisors associated with a specific NUT server?

Thanks

gbakeman, 2023-02-03:
I originally started contributing to WinNUT because I wanted to add more of a business aspect, like managing physical infrastructure as well as virtual. To be honest, I have very little idea into how redundancy works with virtual environments, so I'll need to study that more before implementing.

I definitely want this to be a part of the [next iteration of WinNUT](https://github.com/nutdotnet/WinNUT-Client/issues/40), but WinNUT is in need of some structural upgrades just due to the way .NET has been changing. WinNUT has a rich history of administrative utility especially in businesses, and I think it's time to take it to the next level. Please consider contributing to Coco.nut; ideas like this are great to put on a planning board for the 1.0 release.

gbakeman, 2023-02-05:
Also just want to clarify a few things about your idea:

- Is WinNUT running inside a VM, and we want it to be able to detect the hypervisor its VM is running on?
- So if the VM switches to another hypervisor, WinNUT becomes aware of this?
- Is WinNUT monitoring a UPS connected to the Hypervisor then?
- It sounds like this is much more of a datacenter management type of feature... I'm not sure if this is really within the scope of the WinNUT project. Perhaps if some sort of virtual/hypervisor infrastructure management software comes along,

GitRDone2022, 2023-02-06:
Yes, I agree this might be the wrong place for this. I don't see any way to figure out the hypervisor from the VM.

I've been researching this idea more and it seems like most people handle it from the hypervisor.  
So, exsi, proxmox have a linux nut client.   Then you can create actions that force a VM shutdown if the battery is low.

https://docs.deeztek.com/books/vmware/page/vmware-esxi-6x-nut-client-installation-and-configuration
https://www.kreaweb.be/diy-home-server-2021-software-proxmox-ups/
https://www.itgeekrambling.co.uk/hyper-v-using-ups-hyper-v-server-graceful-shutdown-email-alerts/

gbakeman, 2023-02-06:
I still think the idea of a .Net hypervisor management program is enticing, whether it's a full desktop app or maybe even a .Net Core webapp container. You might be on to something!

closed 2023-02-06 by gbakeman