nutdotnet/WinNUT-Client #78: Setting short date format in windows makes WinNut crash

HotelAlpha4, 2023-02-13:
I have my short date set to yyyy-MM-dd as it removes the confusion of the default dd/MM/yyyy format given occasional reversed date formats. WinNut is confirmed to crash with any short date format that starts with a year. The day/month/year formats all work, as does the month/date/year formats, but they leave a developer guessing the meaning of any short date. I am hoping this is a quick fix, but for the meantime, dd/MMM/yyyy works fine and achieves a similar result.

gbakeman, 2023-02-13:
Could you please provide a log and/or exact error message of the crash, along with version of WinNUT?

alebambi81, 2023-07-01:
Hello,
I guess I have a similar issue.
I've recently changed the regional format on my computer (from Italian to US) and WinNUT always crashes at the startup with this error log:

Exception type: System.FormatException
Exception message: String was not recognized as a valid DateTime.
Exception stack trace:
at System.DateTimeParse.Parse(String s, DateTimeFormatInfo dtfi, DateTimeStyles styles)
at System.Convert.ToDateTime(String value, IFormatProvider provider)
at System.String.System.IConvertible.ToDateTime(IFormatProvider provider)
at System.Convert.ToDateTime(Object value)
at WinNUT_Client.Update_Gui.VerifyUpdate()
at WinNUT_Client.Update_Gui.SetVisibleCore(Boolean value)
at System.Windows.Forms.Control.set_Visible(Boolean value)
at WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
at System.EventHandler.Invoke(Object sender, EventArgs e)
at System.Windows.Forms.Form.OnLoad(EventArgs e)
at System.Windows.Forms.Form.OnCreateControl()
at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
at System.Windows.Forms.Control.CreateControl()
at System.Windows.Forms.Control.WmShowWindow(Message& m)
at System.Windows.Forms.Control.WndProc(Message& m)
at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
at System.Windows.Forms.ContainerControl.WndProc(Message& m)
at System.Windows.Forms.Form.WmShowWindow(Message& m)
at System.Windows.Forms.Form.WndProc(Message& m)
at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

The date has changed from dd/mm/yyyy to mm/dd/yyyy and the time from 24h to 12h.
I tried to uninstall and reinstall it but I always get the crash.
How can I reset or change the WinNUT's date and time format?

Thanks

Alessandro

gbakeman, 2023-08-05:
@HotelAlpha4 and @alebambi81 , could you please try the latest pre-release 8614 and provide a full crash report?

alebambi81, 2023-08-05:
Hello,
thanks for the update.
Unfortunately it still crashes, here's the log

[CrashReport_2023-08-05T17.54.05.txt](https://github.com/nutdotnet/WinNUT-Client/files/12267627/CrashReport_2023-08-05T17.54.05.txt)

gbakeman, 2023-08-05:
@alebambi81 Thank you for getting back to me so quickly. I believe I've fixed the issue, and I have a [debug build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/5771704025?pr=98) building right now. Would you please download the debug build once it's finished, and upload a log here even if there's no problem?

alebambi81, 2023-08-05:
Hi, it works now, thanks!
Here's the log
[WinNUT-Client-2023-08-05.log](https://github.com/nutdotnet/WinNUT-Client/files/12267697/WinNUT-Client-2023-08-05.log)

gbakeman, 2023-08-05:
I'm not seeing the error being triggered there, which is not what I was expecting. Since it sounds like it's fixed though, I'll send out another pre-release so you should be able to update in an hour or so. Thank you for your help!

PR #98 (2023-08-05): Fix FormatException in Update_Gui
Prevent crashing by wrapping `Convert.ToDateTime` in Try-Catch block (default operation will continue with update), and in the future, store dates in a standard format for parsing.

Closes #78
closed 2023-08-05 by gbakeman