nutdotnet/WinNUT-Client #79: WinNUT Crash, dont know why!

Dirk23, 2023-02-14:
Hi,

i installed WinNUT on a Server 2016 and after configuring the application it suddenly crashes. I installed it on other Servers before and hat no issues. I attach the Crashlog, as they maybe of interest.

[Crash_Report_14-02-2023_11-36-29.txt](https://github.com/nutdotnet/WinNUT-Client/files/10731504/Crash_Report_14-02-2023_11-36-29.txt)
[Crash_Report_14-02-2023_11-36-30.txt](https://github.com/nutdotnet/WinNUT-Client/files/10731505/Crash_Report_14-02-2023_11-36-30.txt)

Regards

Dirk

gbakeman, 2023-02-15:
Hi @Dirk23 ,

Thank you for opening this bug report. It looks like there may be a couple of things going on here. Most importantly though, the error seems to happen when you try generating the bug report - not good. Let me see if I can work on the error reporter and have you test a debug build.

Just to be clear, do you have other instances of WinNUT currently working? Or is this the only instance you have?

Dirk23, 2023-02-15:
I have other instances runnig on serval other Servers.

Dirk23, 2023-06-13:
> @Dirk23 Apologies for the long wait on this. As is typical of me, I went down some very long rabbit holes and lost sight of the more pressing issues. I've released a new debug build that hopes to at least address the crash when you try generating a bug report. Please download the debug build [here](https://github.com/nutdotnet/WinNUT-Client/suites/13157572469/artifacts/715546161), extract it to a temporary folder (in your Downloads folder for example), run it, then provide the log when you see the crash again. Thank you!

Hi. Sadly it still crashes.
[CrashReport_2023-06-13T14.42.54.txt](https://github.com/nutdotnet/WinNUT-Client/files/11734928/CrashReport_2023-06-13T14.42.54.txt)

Dirk23, 2023-06-13:
The Server is a Synology NAS.

gbakeman, 2023-07-04:
Please [download the latest debug build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/5458178353) and extract to a temporary location to test. If you please, provide the full log even if the test is successful (should be in the Data directory, which you can find in your settings menu in the log section).

Thank you!

Dirk23, 2023-07-05:
This version is connecting without any issues.

Here is the requested Log:

```
07/05/2023 07:07:32 [6268, Logger]: Windows NUT Client 2.2.8585.36725 Log file init
==== History of 5 previous events ====
[1] 07/05/2023 07:07:32 [6268, Nothing]: Successfully created or opened requested data directory for WinNUT.
requestedDir: C:\Program Files (x86)\WinNUT-client\WinNUT-Client
[2] 07/05/2023 07:07:32 [6268, MyApplication]: MyApplication_Startup complete.
[3] 07/05/2023 07:07:32 [6268, WinNUT]: Initialisation Params Complete
[4] 07/05/2023 07:07:32 [6268, WinNUT]: Beginning WinNUT_PrefsChanged subroutine.
[5] 07/05/2023 07:07:32 [6268, Logger]: Windows NUT Client 2.2.8585.36725 Log file init
==== Begin Live Log ====
07/05/2023 07:07:32 [6268, WinNUT]: Parameter Dial Input Voltage Need to be Updated
07/05/2023 07:07:32 [6268, WinNUT]: Parameter Dial Input Voltage Updated
07/05/2023 07:07:32 [6268, WinNUT]: Parameter Dial Input Frequency Need to be Updated
07/05/2023 07:07:32 [6268, WinNUT]: Parameter Dial Input Frequency Updated
07/05/2023 07:07:32 [6268, WinNUT]: Parameter Dial Output Voltage Need to be Updated
07/05/2023 07:07:32 [6268, WinNUT]: Parameter Dial Output Voltage Updated
07/05/2023 07:07:32 [6268, WinNUT]: Parameter Dial Voltage Battery Need to be Updated
07/05/2023 07:07:32 [6268, WinNUT]: Parameter Dial Voltage Battery Updated
07/05/2023 07:07:32 [6268, WinNUT]: WinNut Preferences Applied.
07/05/2023 07:07:32 [6268, String]: New Log to CB_Current Log : WinNut-Einstellungen geändert
07/05/2023 07:07:32 [6268, WinNUT]: Loaded Params Complete
07/05/2023 07:07:32 [6268, WinNUT]: NotifyIcons Initialised
07/05/2023 07:07:32 [6268, WinNUT]: Windows 10 Toast Notification Not Available. Required Version: 10.0.18362.0, Current: 10.0.14393.0
07/05/2023 07:07:32 [6268, WinNUT]: Windows App Use Dark Theme
07/05/2023 07:07:32 [6268, WinNUT]: Windows Use Dark Theme
07/05/2023 07:07:32 [6268, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Nicht verbunden
07/05/2023 07:07:32 [6268, WinNUT]: Status Icon Changed
07/05/2023 07:07:32 [6268, WinNUT]: New Icon Value For Systray : 1216
07/05/2023 07:07:32 [6268, WinNUT]: New Icon Value For Gui : 1216
07/05/2023 07:07:32 [6268, WinNUT]: Update Icon at Startup
07/05/2023 07:07:32 [6268, WinNUT]: WinNUT Client v2.2.8585.36725 completed initialization.
07/05/2023 07:07:32 [6268, WinNUT]: Main Gui Has Focus
07/05/2023 07:07:32 [6268, WinNUT]: Update Icon
07/05/2023 07:07:32 [6268, WinNUT]: Update Icon
07/05/2023 07:07:32 [6268, WinNUT]: Minimize WinNut On Start
07/05/2023 07:07:32 [6268, WinNUT]: Update Icon
07/05/2023 07:07:32 [6268, WinNUT]: Minimize Main Gui To Notify Icon
07/05/2023 07:07:32 [6268, WinNUT]: Main Gui Lose Focus
07/05/2023 07:07:32 [6268, WinNUT]: Update Icon
07/05/2023 07:07:32 [6268, WinNUT]: Auto-connecting to UPS on startup.
07/05/2023 07:07:32 [6268, WinNUT]: Client UPS_Connect subroutine beginning.
07/05/2023 07:07:32 [6268, UPS_Device]: Beginning connection: upsmon@192.168.110.25:3493, Name: UPS [AutoReconnect]
07/05/2023 07:07:32 [6268, Nut_Socket]: Attempting TCP socket connection to 192.168.110.25:3493...
07/05/2023 07:07:32 [6268, Nut_Socket]: Connection established and streams ready for 192.168.110.25:3493
07/05/2023 07:07:32 [6268, Nut_Socket]: Attempting authentication...
07/05/2023 07:07:32 [6268, Nut_Socket]: Error while attempting to log in: INVALIDARGUMENT (ERR INVALID-ARGUMENT)
Query: LOGIN
07/05/2023 07:07:32 [6268, Nut_Socket]: NUT server reports VER: DSM7-1-1-42930-workplus-version2-repack-42930-220712 NETVER: 1.2
07/05/2023 07:07:32 [6268, UPS_Device]: battery.capacity is not supported by server.
07/05/2023 07:07:32 [6268, UPS_Device]: Apply Fallback Value when retrieving battery.capacity
07/05/2023 07:07:32 [6268, UPS_Device]: output.frequency.nominal is not supported by server.
07/05/2023 07:07:32 [6268, UPS_Device]: Apply Fallback Value when retrieving output.frequency.nominal
07/05/2023 07:07:32 [6268, WinNUT]: UPS has indicated it's ready to start sending data.
07/05/2023 07:07:32 [6268, WinNUT]: Update Icon
07/05/2023 07:07:32 [6268, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Verbunden
07/05/2023 07:07:32 [6268, WinNUT]: Connection to Nut Host Established
07/05/2023 07:07:32 [6268, String]: New Log to CB_Current Log : Verbindung zum Nut Host 192.168.110.25:3493 hergestellt
07/05/2023 07:07:32 [6268, WinNUT]: Completed WinNUT_Shown
07/05/2023 07:07:33 [6268, UPS_Device]: Enter Retrieve_UPS_Datas
07/05/2023 07:07:33 [6268, UPS_Device]: output.frequency is not supported by server.
07/05/2023 07:07:33 [6268, UPS_Device]: Apply Fallback Value when retrieving output.frequency
07/05/2023 07:07:33 [6268, UPS_Device]: input.frequency is not supported by server.
[... 166 lines cut ...]
07/05/2023 07:07:38 [6268, UPS_Device]: input.voltage is not supported by server.
07/05/2023 07:07:38 [6268, UPS_Device]: Apply Fallback Value when retrieving input.voltage
07/05/2023 07:07:38 [6268, UPS_Device]: output.voltage is not supported by server.
07/05/2023 07:07:38 [6268, UPS_Device]: Apply Fallback Value when retrieving output.voltage
07/05/2023 07:07:38 [6268, UPS_Device]: ups.load is not supported by server.
07/05/2023 07:07:38 [6268, UPS_Device]: Apply Fallback Value when retrieving ups.load
07/05/2023 07:07:38 [6268, UPS_Device]: ups.realpower.nominal is not supported by server.
07/05/2023 07:07:38 [6268, UPS_Device]: Apply Fallback Value when retrieving ups.realpower.nominal
[... cut ...]
07/05/2023 07:07:38 [6268, WinNUT]: Updating UPS data for Form.
07/05/2023 07:07:38 [6268, WinNUT]: Updating battery icons based on charge percent: 100%
07/05/2023 07:07:38 [6268, WinNUT]: Calculated estimated remaining battery time: 00:12:26
07/05/2023 07:07:38 [6268, WinNUT]: Update Dial
07/05/2023 07:07:38 [6268, WinNUT]: Update Icon
07/05/2023 07:07:38 [6268, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Verbunden
Im Netzbetrieb
Akku OK
07/05/2023 07:07:38 [6268, WinNUT]: Battery Status => Charged
```

gbakeman, 2023-07-05:
Thank you for capturing that. I'm especially interested in this area:

```
[6268, Nut_Socket]: Error while attempting to log in: INVALIDARGUMENT (ERR INVALID-ARGUMENT)
```

As well as

```
NUT server reports VER: DSM7-1-1-42930-workplus-version2-repack-42930-220712 NETVER: 1.2
```

I think I may need to make additional changes to how Nut_Socket works, if only to get a better picture of what's going on there. The DSM builds of the NUT server have shown some odd behaviors before, so I wonder if there may have been a bug introduced. All I can say is I've never seen errors like this before.

The original issue stemmed from the LOGOUT command. Before I close this, could I make one more build and have you do another test on it? Many thanks in advance.

gbakeman, 2023-08-02:
The issue closed because I merged the pull request with the main branch. Don't worry about doing another test; I own a synology device so I'll do further testing on my own time. I don't think there's much else here that needs to be fixed so I'll be making a pre-release soon. Thank you again for your efforts!

closed 2023-08-02 by gbakeman