nutdotnet/WinNUT-Client #80: When the client fails to connect to the UPS (because of access denied), it then also fails to generate a bug report

miguno, 2023-02-18:
In my case a forbidden IP address on which the client was running caused the client to fail its connection to the UPS. This was expected. What was not expected was that the client app then failed to generate a bug report.

Here is the original error message, when the client's IP address was not yet permitted to connect to the UPS:

```
Exception type: WinNUT_Client_Common.NutException
Exception message: ACCESSDENIED (ERR ACCESS-DENIED)
Query: LOGOUT
Exception stack trace:
   at WinNUT_Client_Common.Nut_Socket.Query_Data(String Query_Msg)
   at WinNUT_Client_Common.Nut_Socket.Disconnect(Object silent, Object forceful)
   at WinNUT_Client_Common.UPS_Device.Disconnect(Boolean cancelReconnect, Boolean silent, Boolean forceful)
   at WinNUT_Client.WinNUT.UPSDisconnect()
   at WinNUT_Client.WinNUT.Menu_Reconnect_Click(Object sender, EventArgs e)
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStripItem.FireEventInteractive(EventArgs e, ToolStripItemEventType met)
   at System.Windows.Forms.ToolStripItem.FireEvent(EventArgs e, ToolStripItemEventType met)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```

The actual problem described in this issue is, however, that the client then also failed to generate a bug report:

```
Exception type: System.InvalidOperationException
Exception message: Collection was modified; enumeration operation may not execute.
Exception stack trace:
   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.Dictionary`2.Enumerator.MoveNext()
   at WinNUT_Client.My.MyApplication.Generate_Button_Click(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```

The attached screenshots show the same information.

[image]

[image]

gbakeman, 2023-02-27:
Thank you for the detailed report! I just want to mention that the Generate Report bug sounds related to #79 , so I'm going to focus on that first if you won't mind testing some debug builds soon.

miguno, 2023-02-27:
Thanks. Works for me!

gbakeman, 2023-02-27:
So just to be sure, I'm assuming that this is happening on Pre-Release 2.2.8436?

miguno, 2023-02-27:
Yes, this was the latest available release (tag) when I downloaded it.

myself248, 2023-04-30:
This is happening to me too. I reported the initial crash in #82 but then the while-generating-the-report crash is here:

Exception type: System.InvalidOperationException
Exception message: Collection was modified; enumeration operation may not execute.
Exception stack trace:
   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.Dictionary`2.Enumerator.MoveNext()
   at WinNUT_Client.My.MyApplication.Generate_Button_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

I am using 2.2.8436.

gbakeman, 2023-05-25:
Everyone,

Apologies for the long wait. I've released a new debug build that hopes to at least address the crash when you try generating a bug report. Please download the debug build [here](https://github.com/nutdotnet/WinNUT-Client/suites/13157572469/artifacts/715546161), extract it to a temporary folder (in your Downloads folder for example), run it, then provide the log when you see the crash again. Thank you!

gbakeman, 2023-05-26:
I've tried to replicate this crash by connecting to a NUT server that I shouldn't be allowed to connect to (by way of hosts.deny/allow), but I can't seem to reproduce it. I did clean up the authentication/login code a little, so I'd like everyone experiencing this issue to please [download the latest debug build](https://github.com/nutdotnet/WinNUT-Client/suites/13183908144/artifacts/717579858) and try running it. If you still experience the error, please describe in detail how you're producing it. Thanks everyone!

andsoitgoes, 2023-06-20:
> I've tried to replicate this crash by connecting to a NUT server that I shouldn't be allowed to connect to (by way of hosts.deny/allow), but I can't seem to reproduce it. I did clean up the [...]

Just wondering if there's anything that's come of this? Do I need to upload further debug logs or are the ones I submitted sufficient?

gbakeman, 2023-07-02:
> Just wondering if there's anything that's come of this? Do I need to upload further debug logs or are the ones I submitted sufficient?

I had made some changes that I thought would help with the crashing, but I had confirmation that that is not the case in #79. I'm actually wondering if this is a duplicate of #79, and I realize I forgot to check what NUT server everyone is connecting to here. Do you all connect to a Synology NUT server by any chance?

gbakeman, 2023-07-04:
All,

Please [download the latest debug build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/5458178353) and extract to a temporary location to test. If you would please, provide the full log even if the test is successful (should be in the Data directory, which you can find in your settings menu in the log section). Also I would like to know exactly what the NUT server is that you're connecting to.

Thank you!

myself248, 2023-07-15:
Hey, this one works! Okay, the NUT sever I'm connecting to is on a Synology NAS, and it has a dialog for entering the IP addresses of permitted hosts. Evidently my desktop's IP was stable for a very long time but then changed, and the server was rejecting it, which I suspect might've been causing the crashes, but I didn't think to check the server config because "nothing had changed". Hah. 

Now that this build doesn't immediately crash, it still wouldn't _connect_, so I went looking for other causes, found the server config, and added my desktop's current IP. (I'll go fix this with static DHCP later.) Now it connects! 

[WinNUT-Client-2023-07-15.log](https://github.com/nutdotnet/WinNUT-Client/files/12062080/WinNUT-Client-2023-07-15.log)

closed 2023-08-02 by gbakeman

gbakeman, 2023-08-02:
All,

The issue closed as a result of me merging the pull request with the main branch. I think the primary problem(s) here have been resolved so I'm working towards a pre-release now. I think we still need to do a better job handling `ACCESS-DENIED` errors, so if you're interested, please subscribe to issue #96 for further updates. And stay tuned for a pre-release in the near future.

andsoitgoes, 2023-08-31:
> All,
>
> The issue closed as a result of me merging the pull request with the main branch. I think the primary problem(s) here have been resolved so I'm working towards a pre-release now. I think we [...]

Thank you!! It's so nice to have this back up and working!!

I had to go through the whole setup again because I'd totally forgotten what the heck I needed to do, tested it again and it fully connected!