nutdotnet/WinNUT-Client #82: Crash after setting connection

GOUKI9999, 2023-03-02:
win11 22H2 22621.1344 /v2.2.8436 /.net v4.0.30319 &3.5 &3.0

Exception type: WinNUT_Client_Common.NutException
Exception message: ACCESSDENIED (ERR ACCESS-DENIED)
Query: LOGOUT
Exception stack trace:
   在 WinNUT_Client_Common.Nut_Socket.Query_Data(String Query_Msg) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\Nut_Socket.vb:行号 225
   在 WinNUT_Client_Common.Nut_Socket.Disconnect(Object silent, Object forceful) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\Nut_Socket.vb:行号 144
   在 WinNUT_Client_Common.UPS_Device.Disconnect(Boolean cancelReconnect, Boolean silent, Boolean forceful) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb:行号 223
   在 WinNUT_Client.WinNUT.UPSDisconnect() 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:行号 343
   在 WinNUT_Client.WinNUT.Menu_Reconnect_Click(Object sender, EventArgs e) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:行号 791
   在 System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   在 System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   在 System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   在 System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   在 System.Windows.Forms.ToolStripItem.FireEventInteractive(EventArgs e, ToolStripItemEventType met)
   在 System.Windows.Forms.ToolStripItem.FireEvent(EventArgs e, ToolStripItemEventType met)
   在 System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   在 System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   在 System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   在 System.Windows.Forms.Control.WndProc(Message& m)
   在 System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   在 System.Windows.Forms.ToolStrip.WndProc(Message& m)
   在 System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   在 System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   在 System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   在 System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Uninstall the old version (gawindx/WinNUT-Client) , del the reg manually

change the setting ,click the connection ,then crash

myself248, 2023-04-30:
I think I'm having the same error, here's my trace: 

Exception type: WinNUT_Client_Common.NutException
Exception message: ACCESSDENIED (ERR ACCESS-DENIED)
Query: LOGOUT
Exception stack trace:
   at WinNUT_Client_Common.Nut_Socket.Query_Data(String Query_Msg)
   at WinNUT_Client_Common.Nut_Socket.Disconnect(Object silent, Object forceful)
   at WinNUT_Client_Common.UPS_Device.Disconnect(Boolean cancelReconnect, Boolean silent, Boolean forceful)
   at WinNUT_Client.WinNUT.UPSDisconnect()
   at WinNUT_Client.WinNUT.Menu_Reconnect_Click(Object sender, EventArgs e)
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStripItem.FireEventInteractive(EventArgs e, ToolStripItemEventType met)
   at System.Windows.Forms.ToolStripItem.FireEvent(EventArgs e, ToolStripItemEventType met)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

It also crashes again while generating the bug report, as seen in #80, I will post my second trace over there

gbakeman, 2023-05-26:
Hi everyone,

This does appear to be a duplicate of #80 so please follow along there for further information.

Thank you!

closed 2023-05-26 by gbakeman