nutdotnet/WinNUT-Client #83: WinNUT has started crashing for... no reason?

andsoitgoes, 2023-03-24:
Hi - so I was trying the old WinNUT client a while back, but I couldn't get it to even start without errors. 

I uninstalled the client, and then found this one which was actually recently updated, and I got FURTHER to where I could actually access the client, but the moment I hit "connect" I get an error thrown, I'm including a few of the crash logs.

Thanks in advance! 

[Crash_Report_23-03-2023_23-27-00.txt](https://github.com/nutdotnet/WinNUT-Client/files/11059059/Crash_Report_23-03-2023_23-27-00.txt)
[Crash_Report_23-03-2023_23-28-13.txt](https://github.com/nutdotnet/WinNUT-Client/files/11059061/Crash_Report_23-03-2023_23-28-13.txt)

gbakeman, 2023-03-24:
I think this is related to #79 . I'm working hard on another project but want to come back to this soon. Please stay tuned; I'll leave this open in case you're experiencing a different issue but hopefully #79 will address this.

andsoitgoes, 2023-05-12:
Any updates here? Hoping I can get WinNUT back up and running, I've tried so many other options with no avail :(

gbakeman, 2023-05-12:
Hi @andsoitgoes ,

I apologize- I fell down yet another rabbit hole and have been neglecting the more important crash bugs like this. I'm reprioritizing, and will start with #79 to see if that fixes the issue you're experiencing.

Sorry for the wait, I'll be in touch.

andsoitgoes, 2023-05-13:
Dude no need to apologize!! Thank you for your commitment, I really appreciate it!!

gbakeman, 2023-05-25:
@andsoitgoes Thanks for your understanding and for sticking with us. I've released a new debug build that hopes to at least address the crash when you try generating a bug report. Please download the debug build [here](https://github.com/nutdotnet/WinNUT-Client/suites/13157572469/artifacts/715546161), extract it to a temporary folder (in your Downloads folder for example), run it, then provide the log when you see the crash again. Thank you!

andsoitgoes, 2023-05-26:
[CrashReport_2023-05-26T01.05.36.txt](https://github.com/nutdotnet/WinNUT-Client/files/11572929/CrashReport_2023-05-26T01.05.36.txt)

There you are!

Thank you again!

gbakeman, 2023-05-26:
Excellent, thanks for getting back to me so quickly.

So I'm not sure about #79 as that seems slightly different, but this does appear to be a **duplicate of #80** so I'd like to ask that you subscribe to that bug and I'll update with further fixes in there. For now though I'll close this as a duplicate and we can reopen later if it turns out being different.

closed 2023-05-26 by gbakeman