nutdotnet/WinNUT-Client #84: EV Certificate signing

deajan, 2023-03-30:
Hello,

I am a WinNUT user for a couple of months now, thanks for the effort.
Just one thing, how about signing the executables and installer with a code signing certificate, so you don't get the Windows defender shield and other antivirus alerts ?

As open source developper myself, I do posses such a certificate, and could sign "general release" executables for you if needed.

deajan, 2023-03-30:
Hmm...
> executable signing with a certificate from Microsoft

AFAIK Microsoft does not distribute free certs, am I right ?
I am talking EV certificates that cost hundreds per year, eg https://learn.microsoft.com/en-us/windows-hardware/drivers/install/authenticode

gbakeman, 2023-03-30:
Sorry for not being clear before-

I'm looking into joining the Microsoft Partner program (free, I hope), and subsequently the [Windows Desktop Application](https://learn.microsoft.com/en-us/windows/win32/appxpkg/windows-desktop-application-program) program. My thinking is that this allows WinNUT to send telemetry and crash events back to Microsoft so I'll be able to review and identify fixes more quickly. My understanding is that this program also provides a certificate to sign the executable, so I'm hoping that that takes care of this. Please let me know if I'm misunderstanding something!

deajan, 2023-03-31:
No problem, but keep in mind that I only can offer to sign the builds for you, as EV certificates come with hardware private  keys, non shareable.
So the point would be to sign official releases only.

Keep me posted.

gbakeman, 2023-04-13:
Hi @deajan ,

I'm moving further along with setting up CI/CD, and wanted to come back to you for more details on your offer. Would you mind describing how this signing process would work? I'd like to have everything isolated to the build environment so signing would happen there if at all possible.

deajan, 2023-04-17:
Hi,

Well the thing with EV certificates is that you can't sign executables without having the actual hardware token, which is unique.
I offered to sign releases only, since this is a manual operation and doesn't happen everyday.
This would mean you publish an executable, I sign it and give it back to you.
AFAIK there's no other way of doing this with EV certificates.

For your CI, you could use a test code signing certificate, and sign with `signtool.exe` that comes with the windows SDK.

deajan, 2023-04-18:
Unless your CI is local, you won't be able to use your signing key (EV or OV since 3rd April) since it needs a physical USB token.
The problem with the sigstore is well discussed (here)[https://github.com/sigstore/fulcio/issues/250#issuecomment-983595336]

Anyway, if someday you need to sign with an authenticode trusted certificate, I'll be around.