nutdotnet/WinNUT-Client #90: Monitor more than one UPS

forlornidealist, 2023-06-06:
Feature request:
We use NUT for servers with redundant power supplies, connected to redundant UPS and power. The NUT project itself provides for a way to monitor multiple UPS, but it appears that WinNUT-Client does not provide an option for this. Basically we would like the ability to be able to monitor two (or more) UPS, and to only shutdown if both are at a specific battery level.

On a side note, thank you for picking up this project!

closed 2023-06-06 by forlornidealist

gbakeman, 2023-06-06:
Hi @forlornidealist ,

Thanks for taking the time to post. And I think you read correctly that multi-UPS monitoring will likely be a feature in a next iteration of WinNUT (Coco.Nut/etc.)

I am especially interested in your use case with multiple/redundant power supplies, and business grade/server infrastructure at large. I envision a future for WinNUT where it exists as a non-specific infrastructure management system. It would of course involve power monitoring, but also coordinate and prioritize resources under atypical power conditions. And this could of course still be as simple as the desktop client we know today, but I don't think a desktop client would be appropriate for business resource management. I'm thinking more along the lines of a virtualized/containerized service.

I'm very inexperienced in these areas. I want the next iteration of WinNUT to be built on a modern foundation with strong flexibility and resiliency. My hope is that this will involve C# as that's my current favorite language, but past that I'm not sure what the future looks like. Hoping to get your thoughts on this.

forlornidealist, 2023-06-16:
Hi @gbakeman ,

I tend to agree with you that a desktop client may not be best. I have no experience with containers outside of Linux, and to be honest wasn't even really aware that they were a thing on Windows, but from a brief look they do seem to have .NET Core containers that would seem to fit the bill. I will give it some thought and see if I can think of anything that would help with regard to any potential architecture of this.