nutdotnet/WinNUT-Client #91: Improve project visibility by way of SEO, word of mouth

gbakeman, 2023-06-08:
Many people are still finding their way to the old WinNUT repository and creating issues, which has a side effect of increasing activity and further raising the old repo's profile.

Absent of having control over the old repository, we should look into ways to boost the visibility of this repository on search engines and websites.

Common search terms

- winnut
- winnut client

uncledulgaria, 2023-12-21:
May I suggest getting listed on this page :- https://networkupstools.org/projects.html ?
At the bottom it says "If you know of a project which should be listed, please send in the URL. ", but I've looked through several of their pages, and can't figure how to contact someone specifically about such a request, otherwise I'd have done it.
I have used the old v1 winnut for years on everything from Win NT4 to Win7, but only came across this site today.  Most of my searches (winnut for Win11) found the old repository before this one.
I hope this is of some help.

gbakeman, 2023-12-22:
> May I suggest getting listed on this page :- https://networkupstools.org/projects.html ?

Haha, I've been all over their site, and I must've glanced across that page years ago and it never registered with me to look into it. I've even been on the NUT developers [mailing list](https://alioth-lists.debian.net/cgi-bin/mailman/listinfo/nut-upsdev) for about just as long. I'll definitely look into this one, thank you.

> I have used the old v1 winnut for years on everything from Win NT4 to Win7, but only came across this site today.

That's amazing, thank you for sticking with this project. I'd love to have more of your feedback about what you think of it currently and for the future.

> Most of my searches (winnut for Win11) found the old repository before this one. I hope this is of some help.

I'm not an expert on GitHub SEO, but I think I recall reading that the visibility of a repository is generally determined by it's activity and popularity. Whenever possible, I try to redirect people interacting with the old WinNUT repository over here. I'm guessing the best we can really do is get the word out. Hyperlink to this repository from other websites and that should generally push us up to the top. Thank you for the ideas and input!

cbturner46, 2024-05-31:
I've just posted a discussion on https://community.synology.com/ which I hope will help.  Go there and search for WinNUT by recent date.  Let me know if any thing else to say, or comment yourself.  Cheers.

gbakeman, 2024-05-31:
Splendid post @cbturner46! You may want to update the link to the project to https://github.com/nutdotnet/WinNUT-Client that way it goes straight here. Thank you for putting the effort into that guide and spreading the word.