nutdotnet/WinNUT-Client #94: Crash on Connection

s3dbw, 2023-07-11:
Latest Beta Version crashing when trying to connect. Crash report uploaded, any thoughts?

Thanks

[Crash Report.txt](https://github.com/nutdotnet/WinNUT-Client/files/12017109/Crash.Report.txt)

gbakeman, 2023-07-11:
It sounds like this issue is a duplicate of #80 , potentially #79 . Please take a look in there for new test builds and see if those help.

miles267, 2023-08-01:
How do you download this debug build?  I'm having same issue on a Windows Server 2019 machine.  Just cannot find the files to download.

gbakeman, 2023-08-01:
@miles267 On the build page [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/5458178353), at the bottom is an "artifact" which you can download. Extract it and then run the WinNUT executable inside. I hope I can get a new pre-release build up soon so people don't have to download these artifacts.

s3dbw, 2023-08-06:
Latest version v2.2.8617 is working for me, so fix appears to be working, thanks

gbakeman, 2023-08-06:
Thank you for letting me know.

closed 2023-08-06 by gbakeman