nutdotnet/WinNUT-Client #100: Transfer settings from Windows Registry to .NET Application Settings, redo Prefs_Gui

gbakeman, 2023-08-10:
# Settings
WinNUT settings/preferences are stored in the user registry hive. To take advantage of more easily manageable and integrated system, we can use the [.NET Application Settings](https://learn.microsoft.com/en-us/visualstudio/ide/managing-application-settings-dotnet) system. This will be a multi-part project:

- [ ] Modify the startup routines to check for existing registry entries. If they exist, prompt the user to import the keys. At the end of the process, ask the user if they'd wish to delete the keys from their registry. Flip a setting indicating that the import process is complete.
- [ ] Modify all project references to the registry keys, and use the new settings system.

# Prefs_Gui
The `Prefs_Gui` Form is being redone in combination with the Settings modification. This will ideally include new Forms design principles such as domain separation and data binding.

- [ ] New Update tab, in combination with #97

DUN-Cosmos, 2023-10-28:
It would be nice to add a backup & restore feature for the setting.

gbakeman, 2023-10-28:
> It would be nice to add a backup & restore feature for the setting.

Thank you for the comment. I'm definitely planning on an import and full backup system (which will likely create an importable .reg file), but I'm not so sure about a restore feature. I want this to be a one-and-done situation, so going forward WinNUT will rely on the new Settings system exclusively and the old Registry system will no longer exist in WinNUT. What do you think about this?

DUN-Cosmos, 2023-10-30:
For the old Registry system, as you say a exported .reg file will be plainly sufficient.

I don't know how the .NET Application Settings work so I can't tall is this need a restore feature.

gbakeman, 2024-01-31:
For anyone monitoring this: please give the [latest build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7731903456?pr=114) a try. I believe all major work on this feature is completed. I've tested the import, backup and deletion features a few times on my own machine, but I'd like to hear from at least one other person confirming success. If I don't hear from anyone in a week or so, I think I'll go ahead and push out a pre-release.

gbakeman, 2024-02-04:
Going ahead and deploying.