nutdotnet/WinNUT-Client #101: CPS wrong output voltage

DfrostX, 2023-09-30:
Hi,

I have the wrong output voltage for my CP1500PFCLCD in WiNUT. It display 137v instead of the expected 121-122v (Canada)

Searching online I found out that his is a known bug that seems to originate from usbhid-ups. It has been documented on other tool's Git:

https://github.com/networkupstools/nut/issues/1338

https://github.com/networkupstools/nut/issues/581

https://alioth-lists.debian.net/pipermail/nut-upsuser/2017-June/010685.html

And fixed https://github.com/networkupstools/nut/issues/439#issuecomment-1213568961

Is there anything that can be done to implement a similar fix/workaround?

gbakeman, 2023-09-30:
Hi @DfrostX ,

I'm just grasping why the issue is occurring from reading those threads. It sounds like the value is being reported incorrectly to NUT's software, or it's interpreting a value incorrectly... I'm not quite sure which. Either way, we can only rely on what the NUT server reports to us when we query the server. Perhaps WinNUT could do a trick where we look at the reported UPS model, then from there either subtract 32 from the reported voltage like someone else said, or just ignore the value entirely. WinNUT is currently programmed to substitute default values when an important parameter is missing (which I don't think is very good practice) so that may be another option.

DfrostX, 2023-09-30:
From what I read it was fixed in NUT 2.8.0

In my case WiNut is reading from a Synology NAS with the UPS connected via USB. So I guess the issue lies with the NAS driver. I guess the fix would need to come from Synology

gbakeman, 2023-09-30:
> ... I guess the issue lies with the NAS driver. I think the fix would need to come from Synology.

I'm afraid that may be the case. If I understood the issue better, I might have been able to make a workaround. Synology seems to pop up here with odd bugs, so hopefully they're on it. Let's leave this bug open until you can confirm a fix.

Grumpy-Old-Engineer, 2024-09-24:
I've got exactly the same problem.... and I just updated to DSM 7.2.2.  Did Synology ever say anything positive, or indicate that they would do anything?

gbakeman, 2024-09-24:
@SHC22 It sounds like you're experiencing a combination of this issue and #136. The gauges will display their minimum value if the true variable is unavailable. I'd suggest calibrating to the minimum of 0V then it should display properly.

@Grumpy-Old-Engineer Best wishes that you find a good answer. I'd also suggest posting in the [Synology Community](https://community.synology.com/) and/or [contacting the NUT people](https://networkupstools.org/support.html) for help.

gbakeman, 2024-09-24:
@SHC22 No mistake on your part! Your NUT server does not provide battery voltage or output voltage (confirm in File -> UPS Variables). This will require further troubleshooting. There are a few issues and discussions here on the topic, but essentially you may need to update your NUT server and/or change to a different driver.

gbakeman, 2024-09-26:
All,

If anyone is experiencing this bug on a Synology device, @Grumpy-Old-Engineer [has confirmed](https://github.com/nutdotnet/WinNUT-Client/discussions/189#discussioncomment-10757468) that Synology is running old software but an update from them is uncertain at this point. I'd encourage anyone experiencing this problem to use an updated instance of the NUT/upsd server software in a different environment (container, package, etc.) until either Synology releases an update or it becomes apparent that WinNUT needs a workaround.

Grumpy-Old-Engineer, 2024-10-01:
The follow-up response I received from Synology was **not** good:

> I have received an update from our Senior Engineers, and they won't be able to upgrade the current NUT Version. However, they recommend to submit a feature request so it can be taken in consideration and update it in future DSM Versions.
> 
> I've successfully submitted this request on your behalf. I can't guarantee any kind of timeline for when this might be implemented, but in the meantime you can sign up to be automatically notified of DSM, SRM, and application updates by email to see if we implement the feature in future updates: https://account.synology.com/
> 
> After signing into your Synology Account, go to Profile->Newsletters to subscribe to emails from the relevant category. If there is a different feature you would like to request in the future you can put in a request through the following form without the need for a support ticket: https://www.synology.com/form/inquiry/feature
> 
> Please do not submit this request again as all requests go to the same location.

Maybe if _everyone_ with an UPS issue on Synology submits a request to have the 12 year old version of the driver DSM is using replaced with the current version we might get some kind of result.

gbakeman, 2024-10-01:
Thanks for the update @Grumpy-Old-Engineer. I'll see if I can submit a request on my own Syno account.

jimklimov, 2024-12-18:
Note from NUT upstream:

PR https://github.com/networkupstools/nut/pull/2718 is presumed to have fixed the broken CPS input/output voltage reports (although it did not focus on cases of "voltage too high" reports directly).

PR https://github.com/networkupstools/nut/pull/1245 is presumed to have fixed the broken CPS transfer voltage reports, at least the high one (and might have fixed or not the output voltages that were too high).

All of those values were reported reasonable in https://github.com/networkupstools/nut/issues/1512 and https://github.com/networkupstools/nut/pull/2718 testing, on the devices available at the time.