nutdotnet/WinNUT-Client #103: Rotating symbol on cancel button when hover in settings menu

masterwishx, 2023-10-08:
installe latest version.
i have Rotating symbol on cancel button when hover in settings menu.

its not captured by screenshot , but meaned this cancel button : 
[image]

gbakeman, 2023-10-08:
Hi @masterwishx ,

This doesn't sound like something WinNUT would do... could you please find a way to capture an image of this icon?

masterwishx, 2023-10-08:
> Hi @masterwishx ,
> 
> This doesn't sound like something WinNUT would do... could you please find a way to capture an image of this symbol?

[image]

gbakeman, 2023-10-08:
Sorry, I completely misunderstood what you meant. You're totally right, the cursor turns into the waiting/spinning circle cursor when I hover over my Cancel button. It looks like the Cancel button mistakenly has the `Cursor` property set to `WaitCursor` when I think it should be `Default`.

I'm currently working on a Preferences GUI replacement/upgrade in #102, so unless this is affecting program functionality, then go ahead and subscribe to that issue and I'll close this one in the meantime. Does this sound good?

masterwishx, 2023-10-08:
Sure, I'm happy this app have been updating, I was using old version 2.0, only yesterday found this new version

closed 2023-10-08 by gbakeman