nutdotnet/WinNUT-Client #104: No auto connection (reconnect) on windows boot

masterwishx, 2023-10-08:
When windows bootted no connection , becouse i think no internet connection on windows start.
but setting reconnect is not working (5 sec).
When make reconnect manual, all working.

[image]

[image]

[image]

Maybe somehow blocked by Norton Antivir but included in exclude list

gbakeman, 2023-10-08:
Does it appear that WinNUT is still trying to connect automatically when you start it, or is it not even doing that?

Could you please try the following steps:

- Close WinNUT
- Open it again
- Wait about 10 seconds
- If not connected now, then connect manually
- Close WinNUT again

Then upload the last log file by opening the folder (found in the options menu in the miscellaneous tab) and clicking and dragging the file into the GitHub message box.

masterwishx, 2023-10-08:
> Does it appear that WinNUT is still trying to connect automatically when you start it, or is it not even doing that?
> 
> Could you please try the following steps:

its doing nothing.

if i close then run again its OK.
i think problem on boot only , when Windows start and like internet connection not avalible on start and take some time for it appear ...

gbakeman, 2023-10-08:
Let me have you do this then: in the miscellaneous tab, set the logging level to Debug, make sure logging is enabled, then click ok and exit WinNUT. Then restart your computer, and allow WinNUT a small amount of time to try to connect. After that, please find the log file and upload it here.

masterwishx, 2023-10-08:
> Let me have you do this then: in the miscellaneous tab, set the logging level to Debug, make sure logging is enabled, then click ok and exit WinNUT. Then restart your computer, and allow WinNUT a small amount of time to try to connect. After that, please find the log file and upload it here.

Strange , but this time after reboot its connected.  
i only enabled and then disabled (run app like administrator in program shortcut).
i will try it again some time if will no problem i will close issue )
also i can disable inet connection and check it if will be reconnection running

[image]

masterwishx, 2023-10-08:
also i saw error about login:   

[image]

i have slave user and password on ups nut server
but in app wrote leave blank on user and password , i think it was also in old 2.0 version.
So its connecting without username and password somehow ?

gbakeman, 2023-10-08:
> Strange , but this time after reboot its connected. i only enabled and then disabled (run app like administrator in program shortcut). i will try it again some time if will no problem i will close [...]

WinNUT currently saves settings in the user registry hive (another issue to track moving this into a file is #100 ), so I wonder if starting it with administrative privileges is causing it to work with two different sets of settings. WinNUT is designed to be run as a standard user (non-administrator) only so please make sure it isn't running as admin.

The login routine is a bit of a curious one with the NUT software and one I don't understand completely. I believe it's intended to tell NUT that you're depending on a UPS for power, so you don't actually need to login in order to read UPS data which is what people do 99% of the time anyways. I think this also gets confused with the Synology server which requires an I.P whitelist in order to connect or read any data from the UPS. All of this to say, WinNUT probably shouldn't have to login at all so I may need to look into that.

Thank you for offering the continue testing this issue. If I could ask you in the future, please upload the log file by dragging it directly into the issue so the entire text file is uploaded (or paste it in a code block.) That will be easier for me to work with than a picture :)

masterwishx, 2023-10-08:
> WinNUT is designed to be run as a standard user (non-administrator) only so please make sure it isn't running as admin.

i only enabled checked and then disabled , i saw no difference only when i enabled  programs dpi :  
it was a little strange smaller view but then also disabled.

[image]

> Thank you for offering the continue testing this issue. If I could ask you in the future, please upload the log file by dragging it directly into the issue so the entire text file is uploaded (or paste it in a code block.) That will be easier for me to work with than a picture :)

i will be glad to help from what i can. im using NUT plugin in Unraid server.
OK i will upload log if it will be needed

gbakeman, 2023-10-08:
Thank you - please double check that the WinNUT settings are correct, then try another reboot with admin and DPI settings disabled.

masterwishx, 2023-10-08:
Yes I disabled them after checked how it work with it, I will check again at morning boot tomorrow  and will let know

masterwishx, 2023-10-09:
Sending the log as at morning was no auto reconnection when was no internet on boot.
then i manually reconnected .

`[38] 09.10.2023 8:52:01 [14692, String]: New Log to CB_Current Log : Соединение с сервером NUT 192.168.0.199:3493 Отказ: Сделана попытка выполнить операцию на сокете при отключенной сети [::ffff:192.168.0.199]:3493`
[WinNUT-Client-2023-10-09.log](https://github.com/nutdotnet/WinNUT-Client/files/12843163/WinNUT-Client-2023-10-09.log)

masterwishx, 2023-10-09:
i checked ,if i disable inet when app is running , then auto reconnect is working until i enabled inet and all OK.
But in morning windows boot there was no reconnection like you can see in log i sended befor ...

masterwishx, 2023-10-09:
in 17:19 i booted win also no connection ,when i reconnect manually, norton popup, hoverer i excluded WinNut:
also sending log again :   

```
UPS_Device]: Beginning connection: @192.168.0.199:3493, Name: Eaton9E2000i [AutoReconnect]
09.10.2023 17:19:17 [13400, Nut_Socket]: Attempting TCP socket connection to 192.168.0.199:3493...
09.10.2023 17:19:17 [13400, WinNUT]: Something went wrong connecting to UPS Eaton9E2000i. IsConnected: False, IsAuthenticated: False
09.10.2023 17:19:17 [13400, String]: New Log to CB_Current Log : Соединение с сервером NUT 192.168.0.199:3493 Отказ: Сделана попытка выполнить операцию на сокете при отключенной сети [::ffff:192.168.0.199]:3493
09.10.2023 17:19:17 [13400, WinNUT]: Completed WinNUT_Shown

```
[WinNUT-Client-2023-10-09.log](https://github.com/nutdotnet/WinNUT-Client/files/12847359/WinNUT-Client-2023-10-09.log)

[image]

UltrashRicco, 2023-10-14:
Hello, I am new to WinNut. I am facing a similar issue where after a shutdown, after the UPS shuts down and powers back up, my Windows machine boots faster than my router, and faster than the machine running the NUT server. When Winnut starts and finds no network, it just appear to cease to try (because the requested IP is unreachable). I am not even sure it goes through the 30 connection attempts. In any case, it does not always happen but when it does, it takes a manual reconnect.

I would suggest increasing the number of connection attempts, or to make it configurable (up to infinite reconnect attempts).

Thanks!

masterwishx, 2023-10-15:
> Hello, I am new to WinNut. I am facing a similar issue where after a shutdown, after the UPS shuts down and powers back up, my Windows machine boots faster than my router, and faster than the [...]

As no answer from developer until now, I can suggest to use old original version 2.0, it's have auto reconnect on boot.
Or to make delay on boot for Winnut in Reg Organizer.

masterwishx, 2023-10-15:
Until it will be fixed here...

UltrashRicco, 2023-10-15:
Thanks for the feedback. I did think of delaying the start of WinNUT (by 30-60 seconds, should be enough) but I have not figured out how yet. Thank you for suggesting Reg Organizer, I did not know about it, however I would personnally prefer not having to use a third-party solution for this.
I can see that the WinNUT client is not a service, but an .exe that is launched at startup. I have not investigated much, but a quick search shows that both can easily be launched with a delay. In the case of a program, it can be done with the built-in Windows task scheduler quite easily, apparently, and with no third-party solution.
This however would likely not address my usecase, as I do not shutdown the computer upon UPS low battery when working on battery power only, as upon resuming the session, WinNUT is already running. It would work fine after a shutdown and subsequent reboot though, so it can still be used as a workaround.

gbakeman, 2023-10-15:
Hi all,

Sorry for disappearing, I had some other things come up. I am able to reproduce the issue on my machine so let me see if I can have a debug build up for you.

gbakeman, 2023-10-15:
This ended up being a very simple fix. I forgot I had disabled the auto-retry function, but only when WinNUT starts for the first time. Sorry for the trouble, everyone. Please try the debug artifact [here ](https://github.com/nutdotnet/WinNUT-Client/actions/runs/6524969018?pr=107) if you want (and please confirm here if it is indeed fixed), otherwise I'll try to get a few other fixes done before I push out another pre-release.

Thanks for your patience!

UltrashRicco, 2023-10-15:
Well, that was quick, thanks!
I tried the build: I rebooted with the computer disconnected from the local network. From what I can see, WinNUT does carry out its 30 connection attempts, then stops and waits for a manual connection. I am attaching the log and a screenshot.
I would love to contribute more, but I am affraid coding is beyond my abilities.
[WinNUT-Client-2023-10-15.log](https://github.com/nutdotnet/WinNUT-Client/files/12909996/WinNUT-Client-2023-10-15.log)
[image]

gbakeman, 2023-10-15:
Hi @UltrashRicco ,

This is considered normal behavior for the auto-reconnection routine, at least when WinNUT is unable to make a connection to the server. Were you expecting a different behavior?

UltrashRicco, 2023-10-15:
Hi! I understand, than you.
I would expect a configurable ammount of retries, up to infinity, as I suggested above, as getting information from the UPS and performing a graceful shutdown is critical in many scenarios: https://github.com/nutdotnet/WinNUT-Client/issues/104#issuecomment-1763309290

gbakeman, 2023-10-15:
Good point, we currently do not allow configuration of the reconnection attempts. You may want to follow #65 since I think that may address your concerns.

UltrashRicco, 2023-10-15:
@gbakeman ah, indeed, thank you, I did have a look through the open issues but did not find #65 and thought I would contribute here instead, sorry for the possible confusion!

gbakeman, 2023-10-15:
No worries at all, thank you for providing the feedback.

PR #107 (2023-10-15): Re-enable retry connection on first start
I had disabled the auto-retry functionality at start because the program hangs as it tries to connect. We'll need to address multithreaded app functionality to truly fix this. In the mean time, WinNUT will now try to automatically reconnect at start when connection issues are encountered.

Closes #104
closed 2023-10-15 by gbakeman