nutdotnet/WinNUT-Client #105: Exit not Working in Win 11

masterwishx, 2023-10-08:
cant exit from app . nothing happening. only can exit by kill app from Ctrl+Alt+del menu
When press exit on icon :   

[image]

or when press Exit in App :  

[image]

masterwishx, 2023-10-11:
i found that exit by mistake ,take function of close to tray .
if i disable function close to tray in menu in settings, then exit working normal . i can exit from program.

[image]

masterwishx, 2023-10-11:
When i disable close to systray from menu and can exit:

[image]

UltrashRicco, 2023-10-14:
Same behaviour here, thank you for the tip consisting in NOT checking "close to tray", at least I can close Winnut if necessary, without having to kill the process.

The expected behaviour would be that when right-clicking on the tray icon and selecting "exit", Winnut should actually exit. Currently, it would appear that it keeps minimizing to the tray.

Thanks!

gbakeman, 2023-10-15:
Hi all,

Another quick fix for this one. After applying it, exiting seems to work normally now. You can try a debug build [here ](https://github.com/nutdotnet/WinNUT-Client/actions/runs/6525104308?pr=108) once it finishes building, and I'm planning on making this a part of the next pre-release.

UltrashRicco, 2023-10-15:
@gbakeman I tested the build and it does work as expected now, thank you! :)

gbakeman, 2023-10-15:
Excellent, thank you for the feedback. I'll merge this and hopefully have a release out today.

closed 2023-10-15 by gbakeman

masterwishx, 2023-10-15:
Thanks for fixing this and auto reconnect bugs. 

Also after update I saw Logs folder moved to program folder instead in AppData/Roaming folder.

gbakeman, 2023-10-16:
> Thanks for fixing this and auto reconnect bugs.
> 
> Also after update I saw Logs folder moved to program folder instead in AppData/Roaming folder.

I noticed that, too. :( I think I may have broken something with the new build configuration. You can track a fix in #110

gbakeman, 2023-10-16:
@masterwishx I wrote up an explanation in the other bug, hope that answers your question.

PR #108 (2023-10-15): Fix unable to exit with tray closing functionality
When Minimize and/or Close to tray is enabled, any calls to Application.Exit are cancelled and WinNUT instead simply hides its self as if it were being minimized/closed to the tray. The application now checks if the user meant to minimize/close it to the tray, otherwise it will exit normally.

Closes #105