nutdotnet/WinNUT-Client #120: UPS load wattage calbrate

alexmb2020, 2023-11-27:
Hi thank You for windows client!

Win NUT ver. 2.2.8719.24624 
Win 11, 22H2

Hi, 
Is any way to calibrate wattage for UPS load? Need to multiply by 3 to have correct wattage, % is correct. UPS load is greyout in the calibration menu.
[image]
[image]

Thank You!
Alex

gbakeman, 2023-11-27:
Hi @alexmb2020 , thank you for moving your post over here. I actually just spent some work on the Load calculation/display in #41, so it sounds like your UPS may have a special case that I need to address. The Calibration menu only affects dial ranges, and I figured that the dial should always be a percentage between 0 and 100%, so I therefor disabled the customization there. You're saying that the percentage of max load is correct, but the displayed wattage is one third of what it should be. Would you please post the output of the UPS Variable List from the **File** menu?

Thank you!

alexmb2020, 2023-11-27:
UPS (CPS/OR2200PFCRT2U/Unknown)
battery.charge (Description unavailable) : 100
battery.charge.low (Description unavailable) : 10
battery.charge.warning (Description unavailable) : 20
battery.mfr.date (Description unavailable) : CPS
battery.runtime (Description unavailable) : 2280
battery.runtime.low (Description unavailable) : 300
battery.type (Description unavailable) : PbAcid
battery.voltage (Description unavailable) : 24.0
battery.voltage.nominal (Description unavailable) : 24
device.mfr (Description unavailable) : CPS
device.model (Description unavailable) : OR2200PFCRT2U
device.type (Description unavailable) : ups
driver.name (Description unavailable) : usbhid-ups
driver.parameter.pollfreq (Description unavailable) : 30
driver.parameter.pollinterval (Description unavailable) : 5
driver.parameter.port (Description unavailable) : auto
driver.parameter.synchronous (Description unavailable) : no
driver.version (Description unavailable) : DSM7-2-1-NewModel-repack-64570-230831
driver.version.data (Description unavailable) : CyberPower HID 0.4
driver.version.internal (Description unavailable) : 0.41
input.voltage (Description unavailable) : 116.0
input.voltage.nominal (Description unavailable) : 120
output.voltage (Description unavailable) : 116.0
ups.beeper.status (Description unavailable) : disabled
ups.delay.shutdown (Description unavailable) : 20
ups.delay.start (Description unavailable) : 30
ups.load (Description unavailable) : 26
ups.mfr (Description unavailable) : CPS
ups.model (Description unavailable) : OR2200PFCRT2U
ups.productid (Description unavailable) : 0601
ups.realpower.nominal (Description unavailable) : 516
ups.status (Description unavailable) : OL
ups.test.result (Description unavailable) : No test initiated
ups.timer.shutdown (Description unavailable) : -60
ups.timer.start (Description unavailable) : -60
ups.vendorid (Description unavailable) : 0764

gbakeman, 2023-11-27:
Thank you for providing that. That's interesting, because I see your `ups.realpower.nominal` variable is available and I'm assuming that _516_ is the correct amount of wattage for the time this was taken? WinNUT should have automatically detected that. Could you please attach your log file from WinNUT? You can find it in the "Miscellanous" tab, by clicking the document with a blue 'i'.

alexmb2020, 2023-11-27:
Its too big is it ok to add it to post?

Please see below

 

11/27/2023 10:44:34 AM [29096, Logger]: Windows NUT Client 2.2.8719.24624 Log file init

==== History of 200 previous events ====

[1] 11/27/2023 10:44:20 AM [29096, WinNUT]: Updating UPS data for Form.

[2] 11/27/2023 10:44:20 AM [29096, WinNUT]: Updating battery icons based on charge percent: 100%

[3] 11/27/2023 10:44:20 AM [29096, WinNUT]: Calculated estimated remaining battery time: 00:39:00

[4] 11/27/2023 10:44:20 AM [29096, WinNUT]: Update Dial

[5] 11/27/2023 10:44:20 AM [29096, WinNUT]: Update Icon

[6] 11/27/2023 10:44:20 AM [29096, WinNUT]: NotifyIcon Text => 

WinNUT Client - 2.2

Connected

On Line

Battery OK

[7] 11/27/2023 10:44:20 AM [29096, WinNUT]: Battery Status => Charged

[8] 11/27/2023 10:44:21 AM [29096, WinNUT]: Open Pref Gui From Menu

[9] 11/27/2023 10:44:21 AM [29096, Pref_Gui]: Load Pref Gui

[10] 11/27/2023 10:44:21 AM [29096, WinNUT]: Main Gui Lose Focus

[11] 11/27/2023 10:44:21 AM [29096, WinNUT]: Update Icon

[12] 11/27/2023 10:44:21 AM [29096, Pref_Gui]: Pref Gui Opened.

[13] 11/27/2023 10:44:21 AM [29096, UPS_Device]: Enter Retrieve_UPS_Datas

[14] 11/27/2023 10:44:21 AM [29096, UPS_Device]: output.frequency is not supported by server.

[15] 11/27/2023 10:44:21 AM [29096, UPS_Device]: Apply Fallback Value when retrieving output.frequency

[16] 11/27/2023 10:44:21 AM [29096, UPS_Device]: input.frequency is not supported by server.

[17] 11/27/2023 10:44:21 AM [29096, UPS_Device]: Apply Fallback Value when retrieving input.frequency

[18] 11/27/2023 10:44:21 AM [29096, UPS_Device]: UPS statuses have not changed since last update, skipping.

[19] 11/27/2023 10:44:21 AM [29096, WinNUT]: Updating UPS data for Form.

[20] 11/27/2023 10:44:21 AM [29096, WinNUT]: Updating battery icons based on charge percent: 100%

[21] 11/27/2023 10:44:21 AM [29096, WinNUT]: Calculated estimated remaining battery time: 00:39:00

[22] 11/27/2023 10:44:21 AM [29096, WinNUT]: Update Dial

[23] 11/27/2023 10:44:21 AM [29096, WinNUT]: Update Icon

[24] 11/27/2023 10:44:21 AM [29096, WinNUT]: NotifyIcon Text => 

WinNUT Client - 2.2

Connected

[... 456 lines cut ...]
[199] 11/27/2023 10:44:34 AM [29096, WinNUT]: Beginning WinNUT_PrefsChanged subroutine.

[200] 11/27/2023 10:44:34 AM [29096, Logger]: Windows NUT Client 2.2.8719.24624 Log file init

==== Begin Live Log ====

11/27/2023 10:44:34 AM [29096, WinNUT]: Connection parameters Changed. Force Reconnect

11/27/2023 10:44:34 AM [29096, WinNUT]: Disconnected from Nut Host

11/27/2023 10:44:34 AM [29096, WinNUT]: Client UPS_Connect subroutine beginning.

11/27/2023 10:44:34 AM [29096, Nut_Socket]: Attempting TCP socket connection to NetStation:3493...

11/27/2023 10:44:36 AM [29096, Nut_Socket]: Connection established and streams ready for NetStation:3493

11/27/2023 10:44:36 AM [29096, Nut_Socket]: Attempting authentication...

11/27/2023 10:44:36 AM [29096, Nut_Socket]: NUT server reports VER: DSM7-2-1-NewModel-repack-64570-230831 NETVER: 1.2

11/27/2023 10:44:36 AM [29096, UPS_Device]: Using NominalPowerCalc method to calculate power usage.

11/27/2023 10:44:36 AM [29096, WinNUT]: Connection to Nut Host Established

11/27/2023 10:44:36 AM [29096, WinNUT]: WinNut Preferences Applied.

alexmb2020, 2023-11-27:
> Thank you - this confirms that WinNUT is getting load information from the ups.realpower.nominal and ups.load variables. I noticed that your UPS model says it supports about 1400W of power, although the nominal realpower variables says 516. I think that's why the reported power is about a third of what you're expecting. I'm just trying to make sure I fully understand the problem so I can make a fix that benefits as many people as possible.
>
> I think the first order of business is understanding why it reports the nominal realpower as 516. This could be a bug in the UPS firmware, or a bug with the driver NUT's upsd uses. Either way, it looks like your exact model of UPS isn't officially supported so you may want to open an issue <https://github.com/networkupstools/nut/issues>  with the NUT people to have them take a closer look at this.
>
> In the meantime, I know a simple solution would be to just override/manually specify the max power of the UPS. I'll create a new issue for something like this and provide a link to you.

Hi, Thank You for your time and support

 

 

 

Alex

gbakeman, 2023-12-02:
I realized we have an issue open that can address this; please subscribe to #69 as that should address what you need. In the meantime, please consider opening that issue with the NUT people as that may give you a more direct solution to the problem you're experiencing.