nutdotnet/WinNUT-Client #123: Since version 2.2.8693 WinNut Constantly reconnects

jandari1, 2023-12-06:
Im using NUT server on Tomato firmware router. Last working version was v2.2.8690. After that WinNut gets data for a second then recconects in a loop.

```
5.12. 2023 21:50:22 [27356, Logger]: Windows NUT Client 2.2.8719.24624 Log file init
==== History of 200 previous events ====
[1] 5. 12. 2023 21:50:07 [27356, WinNUT]: Battery Status => Unknown
[2] 5. 12. 2023 21:50:07 [27356, WinNUT]: Disconnected from Nut Host
[3] 5. 12. 2023 21:50:07 [27356, String]: New Log to CB_Current Log : Disconnected from Nut Host
[4] 5. 12. 2023 21:50:07 [27356, UPS_Device]: Socket has reported a Broken event.
[5] 5. 12. 2023 21:50:07 [27356, WinNUT]: Notify user of lost connection
[6] 5. 12. 2023 21:50:07 [27356, String]: New Log to CB_Current Log : Lost Connect To 192.168.1.1:3493
[7] 5. 12. 2023 21:50:07 [27356, WinNUT]: Update all informations displayed to empty values
[8] 5. 12. 2023 21:50:07 [27356, WinNUT]: Status Icon Changed
[9] 5. 12. 2023 21:50:07 [27356, WinNUT]: New Icon Value For Systray : 1344
[10] 5. 12. 2023 21:50:07 [27356, WinNUT]: New Icon Value For Gui : 1344
[11] 5. 12. 2023 21:50:07 [27356, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Lost Connect To 192.168.1.1:3493
[12] 5. 12. 2023 21:50:07 [27356, WinNUT]: Battery Status => Unknown
[13] 5. 12. 2023 21:50:07 [27356, WinNUT]: Update Icon
[14] 5. 12. 2023 21:50:07 [27356, UPS_Device]: Reconnection Process Started
[15] 5. 12. 2023 21:50:12 [27356, Pref_Gui]: Setting LogControl statuses.
[16] 5. 12. 2023 21:50:12 [27356, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Reconnection In Progress
Try 1 of 30
[17] 5. 12. 2023 21:50:12 [27356, WinNUT]: Update Icon
[18] 5. 12. 2023 21:50:12 [27356, UPS_Device]: Try Reconnect 1 / 30
[19] 5. 12. 2023 21:50:12 [27356, String]: New Log to CB_Current Log : Try Reconnect 1 / 30
[20] 5. 12. 2023 21:50:12 [27356, UPS_Device]: Beginning connection: qnap@192.168.1.1:3493, Name: qnapups [AutoReconnect]
[21] 5. 12. 2023 21:50:12 [27356, Nut_Socket]: Attempting TCP socket connection to 192.168.1.1:3493...
[22] 5. 12. 2023 21:50:12 [27356, Nut_Socket]: Connection established and streams ready for 192.168.1.1:3493
[23] 5. 12. 2023 21:50:12 [27356, Nut_Socket]: Attempting authentication...
[24] 5. 12. 2023 21:50:12 [27356, Nut_Socket]: Error while attempting to log in: INVALIDARGUMENT (ERR INVALID-ARGUMENT)
Query: LOGIN
[25] 5. 12. 2023 21:50:12 [27356, Nut_Socket]: NUT server reports VER: 2.8.0 NETVER: 
[26] 5. 12. 2023 21:50:12 [27356, UPS_Device]: ups.realpower is not supported by server.
[27] 5. 12. 2023 21:50:12 [27356, UPS_Device]: Unhandled error while getting ups.realpower
[28] 5. 12. 2023 21:50:12 [27356, UPS_Device]: Using NominalPowerCalc method to calculate power usage.
[29] 5. 12. 2023 21:50:12 [27356, UPS_Device]: battery.capacity is not supported by server.
[30] 5. 12. 2023 21:50:12 [27356, UPS_Device]: Apply Fallback Value when retrieving battery.capacity
[31] 5. 12. 2023 21:50:12 [27356, UPS_Device]: output.frequency.nominal is not supported by server.
[32] 5. 12. 2023 21:50:12 [27356, UPS_Device]: Apply Fallback Value when retrieving output.frequency.nominal
[33] 5. 12. 2023 21:50:12 [27356, WinNUT]: qnapups has indicated it's ready to start sending data.
[34] 5. 12. 2023 21:50:12 [27356, WinNUT]: Update Icon
[35] 5. 12. 2023 21:50:12 [27356, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Connected
[36] 5. 12. 2023 21:50:12 [27356, WinNUT]: Connection to Nut Host Established
[37] 5. 12. 2023 21:50:12 [27356, String]: New Log to CB_Current Log : Connection to Nut Host 192.168.1.1:3493 Established
[38] 5. 12. 2023 21:50:12 [27356, UPS_Device]: Nut Host Reconnected
[39] 5. 12. 2023 21:50:12 [27356, WinNUT]: qnapups has indicated it's ready to start sending data.
[40] 5. 12. 2023 21:50:12 [27356, WinNUT]: Update Icon
[41] 5. 12. 2023 21:50:12 [27356, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Connected
[42] 5. 12. 2023 21:50:12 [27356, WinNUT]: Connection to Nut Host Established
[43] 5. 12. 2023 21:50:12 [27356, String]: New Log to CB_Current Log : Connection to Nut Host 192.168.1.1:3493 Established
[44] 5. 12. 2023 21:50:13 [27356, UPS_Device]: Enter Retrieve_UPS_Datas
[45] 5. 12. 2023 21:50:13 [27356, UPS_Device]: output.frequency is not supported by server.
[46] 5. 12. 2023 21:50:13 [27356, UPS_Device]: Apply Fallback Value when retrieving output.frequency
[47] 5. 12. 2023 21:50:13 [27356, UPS_Device]: input.frequency is not supported by server.
[48] 5. 12. 2023 21:50:13 [27356, UPS_Device]: Apply Fallback Value when retrieving input.frequency
[49] 5. 12. 2023 21:50:13 [27356, UPS_Device]: output.voltage is not supported by server.
[50] 5. 12. 2023 21:50:13 [27356, UPS_Device]: Apply Fallback Value when retrieving output.voltage
[51] 5. 12. 2023 21:50:13 [27356, UPS_Device]: Something went wrong in Retrieve_UPS_Datas: System.FormatException: Input string was not in a correct format.
   at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseInt32(String s, NumberStyles style, NumberFormatInfo info)
   at WinNUT_Client_Common.UPS_Device.GetPowerUsage()
   at WinNUT_Client_Common.UPS_Device.Retrieve_UPS_Datas()
[52] 5. 12. 2023 21:50:13 [27356, UPS_Device]: Processing request to disconnect...
[53] 5. 12. 2023 21:50:13 [27356, WinNUT]: Update all informations displayed to empty values
[54] 5. 12. 2023 21:50:13 [27356, WinNUT]: Update Icon
[55] 5. 12. 2023 21:50:13 [27356, WinNUT]: Status Icon Changed
[56] 5. 12. 2023 21:50:13 [27356, WinNUT]: New Icon Value For Systray : 1216
[57] 5. 12. 2023 21:50:13 [27356, WinNUT]: New Icon Value For Gui : 1216
[58] 5. 12. 2023 21:50:13 [27356, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
[59] 5. 12. 2023 21:50:13 [27356, WinNUT]: Battery Status => Unknown
[60] 5. 12. 2023 21:50:13 [27356, WinNUT]: Disconnected from Nut Host
[61] 5. 12. 2023 21:50:13 [27356, String]: New Log to CB_Current Log : Disconnected from Nut Host
[62] 5. 12. 2023 21:50:13 [27356, UPS_Device]: Socket has reported a Broken event.
[63] 5. 12. 2023 21:50:13 [27356, WinNUT]: Notify user of lost connection
[64] 5. 12. 2023 21:50:13 [27356, String]: New Log to CB_Current Log : Lost Connect To 192.168.1.1:3493
[65] 5. 12. 2023 21:50:13 [27356, WinNUT]: Update all informations displayed to empty values
[66] 5. 12. 2023 21:50:13 [27356, WinNUT]: Status Icon Changed
[67] 5. 12. 2023 21:50:13 [27356, WinNUT]: New Icon Value For Systray : 1344
[68] 5. 12. 2023 21:50:13 [27356, WinNUT]: New Icon Value For Gui : 1344
[69] 5. 12. 2023 21:50:13 [27356, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Lost Connect To 192.168.1.1:3493
[70] 5. 12. 2023 21:50:13 [27356, WinNUT]: Battery Status => Unknown
[71] 5. 12. 2023 21:50:13 [27356, WinNUT]: Update Icon
[72] 5. 12. 2023 21:50:13 [27356, UPS_Device]: Reconnection Process Started
[73] 5. 12. 2023 21:50:18 [27356, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Reconnection In Progress
```

gbakeman, 2023-12-06:
Hi,

Could you please do the following for me:

- Set WinNUT's log level to Debug, in the Settings menu
- Exit WinNUT
- Clear all remaining log files in WinNUT's AppData folder
- Restart WinNUT
- Allow it to cycle through three connect/disconnect cycles
- Close WinNUT again

After that, please attach the entire log file to your next comment here. Thank you!

gbakeman, 2023-12-07:
Thank you for providing that - this confirms that something strange is happening in this area:

```
6. 12. 2023 19:33:02 [11228, UPS_Device]: Apply Fallback Value when retrieving output.voltage
6. 12. 2023 19:33:02 [11228, UPS_Device]: Something went wrong in Retrieve_UPS_Datas: System.FormatException: Input string was not in a correct format.
   at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseInt32(String s, NumberStyles style, NumberFormatInfo info)
   at WinNUT_Client_Common.UPS_Device.GetPowerUsage()
   at WinNUT_Client_Common.UPS_Device.Retrieve_UPS_Datas()
```

I'll need some time to create a debug build for you, but if you could please test when I do then I'd really appreciate it.

dpvDuncan, 2023-12-12:
Hello,

I have the same problem: it was working great with v2.2.8690, but starting with v2.2.8693, it stopped working with the same symptoms...

My NUT server is an Raspberry Pi Zero running RaspBian 12, connected to a Technoware Era Plus 1100 UPS via USB

Here are my logs:

```
12/12/2023 17:47:50 [1124, Logger]: Windows NUT Client 2.2.8719.24624 Log file init
==== History of 4 previous events ====
[1] 12/12/2023 17:47:50 [1124, MyApplication]: MyApplication_Startup complete.
[2] 12/12/2023 17:47:50 [1124, WinNUT]: Initialisation Params Complete
[3] 12/12/2023 17:47:50 [1124, WinNUT]: Beginning WinNUT_PrefsChanged subroutine.
[4] 12/12/2023 17:47:50 [1124, Logger]: Windows NUT Client 2.2.8719.24624 Log file init
==== Begin Live Log ====
12/12/2023 17:47:50 [1124, WinNUT]: Parameter Dial Input Voltage Need to be Updated
12/12/2023 17:47:50 [1124, WinNUT]: Parameter Dial Input Voltage Updated
12/12/2023 17:47:50 [1124, WinNUT]: Parameter Dial Input Frequency Need to be Updated
12/12/2023 17:47:50 [1124, WinNUT]: Parameter Dial Input Frequency Updated
12/12/2023 17:47:50 [1124, WinNUT]: Parameter Dial Output Voltage Need to be Updated
12/12/2023 17:47:50 [1124, WinNUT]: Parameter Dial Output Voltage Updated
12/12/2023 17:47:50 [1124, WinNUT]: Parameter Dial Voltage Battery Need to be Updated
12/12/2023 17:47:50 [1124, WinNUT]: Parameter Dial Voltage Battery Updated
12/12/2023 17:47:50 [1124, WinNUT]: WinNut Preferences Applied.
12/12/2023 17:47:50 [1124, String]: New Log to CB_Current Log : Préférences de WinNUT Modifiées
12/12/2023 17:47:50 [1124, WinNUT]: Loaded Params Complete
12/12/2023 17:47:50 [1124, WinNUT]: NotifyIcons Initialised
12/12/2023 17:47:50 [1124, WinNUT]: Windows 10 Toast Notification Available
12/12/2023 17:47:50 [1124, WinNUT]: Windows App Use Dark Theme
12/12/2023 17:47:50 [1124, WinNUT]: Windows Use Dark Theme
12/12/2023 17:47:50 [1124, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Non connecté
12/12/2023 17:47:50 [1124, WinNUT]: Status Icon Changed
12/12/2023 17:47:50 [1124, WinNUT]: New Icon Value For Systray : 1216
12/12/2023 17:47:50 [1124, WinNUT]: New Icon Value For Gui : 1216
12/12/2023 17:47:50 [1124, WinNUT]: Update Icon at Startup
12/12/2023 17:47:50 [1124, WinNUT]: Run Automatic Update
12/12/2023 17:47:50 [1124, Update_Gui]: Verify Update
12/12/2023 17:47:50 [1124, WinNUT]: WinNUT Client v2.2.8719.24624 completed initialization.
12/12/2023 17:47:50 [1124, WinNUT]: Main Gui Has Focus
12/12/2023 17:47:50 [1124, WinNUT]: Update Icon
12/12/2023 17:47:50 [1124, WinNUT]: Update Icon
12/12/2023 17:47:50 [1124, WinNUT]: Minimize WinNut On Start
12/12/2023 17:47:50 [1124, WinNUT]: Update Icon
12/12/2023 17:47:50 [1124, WinNUT]: Minimize Main Gui To Notify Icon
12/12/2023 17:47:50 [1124, WinNUT]: Main Gui Lose Focus
12/12/2023 17:47:50 [1124, WinNUT]: Update Icon
12/12/2023 17:47:50 [1124, WinNUT]: Auto-connecting to UPS on startup.
12/12/2023 17:47:50 [1124, WinNUT]: Client UPS_Connect subroutine beginning.
12/12/2023 17:47:50 [1124, UPS_Device]: Beginning connection: dpvduncan@192.168.1.3:3493, Name: tecnoware_era_plus_1100 [AutoReconnect]
12/12/2023 17:47:50 [1124, Nut_Socket]: Attempting TCP socket connection to 192.168.1.3:3493...
12/12/2023 17:47:50 [1124, Nut_Socket]: Connection established and streams ready for 192.168.1.3:3493
12/12/2023 17:47:50 [1124, Nut_Socket]: Attempting authentication...
12/12/2023 17:47:50 [1124, Nut_Socket]: Error while attempting to log in: INVALIDARGUMENT (ERR INVALID-ARGUMENT)
Query: LOGIN
12/12/2023 17:47:50 [1124, Nut_Socket]: NUT server reports VER: 2.8.0 NETVER: 
12/12/2023 17:47:50 [1124, UPS_Device]: ups.mfr is not supported by server.
12/12/2023 17:47:50 [1124, UPS_Device]: Apply Fallback Value when retrieving ups.mfr
12/12/2023 17:47:50 [1124, UPS_Device]: ups.model is not supported by server.
12/12/2023 17:47:50 [1124, UPS_Device]: Apply Fallback Value when retrieving ups.model
12/12/2023 17:47:50 [1124, UPS_Device]: ups.serial is not supported by server.
12/12/2023 17:47:50 [1124, UPS_Device]: Apply Fallback Value when retrieving ups.serial
12/12/2023 17:47:50 [1124, UPS_Device]: ups.firmware is not supported by server.
12/12/2023 17:47:50 [1124, UPS_Device]: Apply Fallback Value when retrieving ups.firmware
12/12/2023 17:47:50 [1124, UPS_Device]: ups.realpower is not supported by server.
12/12/2023 17:47:50 [1124, UPS_Device]: Unhandled error while getting ups.realpower
12/12/2023 17:47:50 [1124, UPS_Device]: ups.realpower.nominal is not supported by server.
12/12/2023 17:47:50 [1124, UPS_Device]: Unhandled error while getting ups.realpower.nominal
[... 325 lines cut ...]
12/12/2023 17:48:51 [1124, UPS_Device]: ups.realpower is not supported by server.
12/12/2023 17:48:51 [1124, UPS_Device]: Unhandled error while getting ups.realpower
12/12/2023 17:48:51 [1124, UPS_Device]: ups.realpower.nominal is not supported by server.
12/12/2023 17:48:51 [1124, UPS_Device]: Unhandled error while getting ups.realpower.nominal
12/12/2023 17:48:51 [1124, UPS_Device]: Using VoltAmpCalc method to calculate power usage.
12/12/2023 17:48:51 [1124, UPS_Device]: battery.capacity is not supported by server.
12/12/2023 17:48:51 [1124, UPS_Device]: Apply Fallback Value when retrieving battery.capacity
12/12/2023 17:48:51 [1124, UPS_Device]: output.frequency.nominal is not supported by server.
12/12/2023 17:48:51 [1124, UPS_Device]: Apply Fallback Value when retrieving output.frequency.nominal
[... cut ...]
12/12/2023 17:48:51 [1124, String]: New Log to CB_Current Log : Connexion au Serveur NUT 192.168.1.3:3493 établie
12/12/2023 17:48:51 [1124, UPS_Device]: Nut Host Reconnected
12/12/2023 17:48:51 [1124, WinNUT]: tecnoware_era_plus_1100 has indicated it's ready to start sending data.
12/12/2023 17:48:51 [1124, WinNUT]: Update Icon
12/12/2023 17:48:51 [1124, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Connecté
12/12/2023 17:48:51 [1124, WinNUT]: Connection to Nut Host Established
12/12/2023 17:48:51 [1124, String]: New Log to CB_Current Log : Connexion au Serveur NUT 192.168.1.3:3493 établie
```

gbakeman, 2023-12-12:
Hi all, thank you for the reports. This one is a bit of a mystery to me. I've added some debugging code so when you hit the error next time, it will print out additional information into the log file. It also won't disconnect anymore, instead the power calculation will just be zero.

I have that debug build cooking [over here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7187508222). Once the download is available, please go ahead and run it, and report back with a few things for me:

- The extended error information from the logfile
- Full output from the UPS Variable List in the File menu

Thank you!

jandari1, 2023-12-13:
The debug version is working now

```
12/13/2023 08:56:42 [29860, Logger]: Windows NUT Client 2.2.8746.38778 Log file init
==== History of 4 previous events ====
[1] 12/13/2023 08:56:41 [29860, MyApplication]: MyApplication_Startup complete.
[2] 12/13/2023 08:56:42 [29860, WinNUT]: Initialisation Params Complete
[3] 12/13/2023 08:56:42 [29860, WinNUT]: Beginning WinNUT_PrefsChanged subroutine.
[4] 12/13/2023 08:56:42 [29860, Logger]: Windows NUT Client 2.2.8746.38778 Log file init
==== Begin Live Log ====
12/13/2023 08:56:42 [29860, WinNUT]: Parameter Dial Input Voltage Need to be Updated
12/13/2023 08:56:42 [29860, WinNUT]: Parameter Dial Input Voltage Updated
12/13/2023 08:56:42 [29860, WinNUT]: Parameter Dial Input Frequency Need to be Updated
12/13/2023 08:56:42 [29860, WinNUT]: Parameter Dial Input Frequency Updated
12/13/2023 08:56:42 [29860, WinNUT]: Parameter Dial Output Voltage Need to be Updated
12/13/2023 08:56:42 [29860, WinNUT]: Parameter Dial Output Voltage Updated
12/13/2023 08:56:42 [29860, WinNUT]: Parameter Dial Voltage Battery Need to be Updated
12/13/2023 08:56:42 [29860, WinNUT]: Parameter Dial Voltage Battery Updated
12/13/2023 08:56:42 [29860, WinNUT]: WinNut Preferences Applied.
12/13/2023 08:56:42 [29860, String]: New Log to CB_Current Log : WinNut Preferences Changed
12/13/2023 08:56:42 [29860, WinNUT]: Loaded Params Complete
12/13/2023 08:56:42 [29860, WinNUT]: NotifyIcons Initialised
12/13/2023 08:56:42 [29860, WinNUT]: Windows 10 Toast Notification Available
12/13/2023 08:56:42 [29860, WinNUT]: Windows App Use Light Theme
12/13/2023 08:56:42 [29860, WinNUT]: Windows Use Dark Theme
12/13/2023 08:56:42 [29860, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Not Connected
12/13/2023 08:56:42 [29860, WinNUT]: Status Icon Changed
12/13/2023 08:56:42 [29860, WinNUT]: New Icon Value For Systray : 1216
12/13/2023 08:56:42 [29860, WinNUT]: New Icon Value For Gui : 1152
12/13/2023 08:56:42 [29860, WinNUT]: Update Icon at Startup
12/13/2023 08:56:42 [29860, WinNUT]: WinNUT Client v2.2.8746.38778 completed initialization.
12/13/2023 08:56:42 [29860, WinNUT]: Main Gui Has Focus
12/13/2023 08:56:42 [29860, WinNUT]: Update Icon
12/13/2023 08:56:42 [29860, WinNUT]: Update Icon
12/13/2023 08:56:42 [29860, WinNUT]: Minimize WinNut On Start
12/13/2023 08:56:42 [29860, WinNUT]: Update Icon
12/13/2023 08:56:42 [29860, WinNUT]: Minimize Main Gui To Notify Icon
12/13/2023 08:56:42 [29860, WinNUT]: Main Gui Lose Focus
12/13/2023 08:56:42 [29860, WinNUT]: Update Icon
12/13/2023 08:56:42 [29860, WinNUT]: Auto-connecting to UPS on startup.
12/13/2023 08:56:42 [29860, WinNUT]: Client UPS_Connect subroutine beginning.
12/13/2023 08:56:42 [29860, UPS_Device]: Beginning connection: @192.168.1.1:3493, Name: QNAPUPS [AutoReconnect]
12/13/2023 08:56:42 [29860, Nut_Socket]: Attempting TCP socket connection to 192.168.1.1:3493...
12/13/2023 08:56:42 [29860, Nut_Socket]: Connection established and streams ready for 192.168.1.1:3493
12/13/2023 08:56:42 [29860, Nut_Socket]: Attempting authentication...
12/13/2023 08:56:42 [29860, Nut_Socket]: Error while attempting to log in: USERNAMEREQUIRED (ERR USERNAME-REQUIRED)
Query: LOGIN
12/13/2023 08:56:42 [29860, Nut_Socket]: NUT server reports VER: 2.8.0 NETVER: 
12/13/2023 08:56:42 [29860, UPS_Device]: Retrieving basic UPS product information...
12/13/2023 08:56:42 [29860, UPS_Device]: Determining best method to calculate power usage...
12/13/2023 08:56:42 [29860, UPS_Device]: ups.realpower is not supported by server.
12/13/2023 08:56:42 [29860, UPS_Device]: Unhandled error while getting ups.realpower
12/13/2023 08:56:42 [29860, UPS_Device]: Using NominalPowerCalc method to calculate power usage.
12/13/2023 08:56:42 [29860, UPS_Device]: battery.capacity is not supported by server.
12/13/2023 08:56:42 [29860, UPS_Device]: Apply Fallback Value when retrieving battery.capacity
12/13/2023 08:56:42 [29860, UPS_Device]: output.frequency.nominal is not supported by server.
12/13/2023 08:56:42 [29860, UPS_Device]: Apply Fallback Value when retrieving output.frequency.nominal
12/13/2023 08:56:42 [29860, UPS_Device]: Completed retrieval of basic UPS product information.
12/13/2023 08:56:42 [29860, WinNUT]: QNAPUPS has indicated it's ready to start sending data.
12/13/2023 08:56:42 [29860, WinNUT]: Update Icon
12/13/2023 08:56:42 [29860, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Connected
12/13/2023 08:56:42 [29860, WinNUT]: Connection to Nut Host Established
12/13/2023 08:56:42 [29860, String]: New Log to CB_Current Log : Connection to Nut Host 192.168.1.1:3493 Established
12/13/2023 08:56:42 [29860, WinNUT]: Completed WinNUT_Shown
12/13/2023 08:56:43 [29860, UPS_Device]: Enter Retrieve_UPS_Datas
12/13/2023 08:56:43 [29860, UPS_Device]: output.frequency is not supported by server.
[... 223 lines cut ...]
device.type (Description unavailable) : ups
driver.name (Driver name) : apcupsd-ups
driver.parameter.pollinterval (Description unavailable) : 5
driver.parameter.port (Description unavailable) : localhost
driver.parameter.synchronous (Description unavailable) : auto
driver.version (Driver version - NUT release) : 2.8.0
driver.version.internal (Internal driver version) : 0.6
input.sensitivity (Input power sensitivity) : M
input.transfer.high (High voltage transfer point (V)) : 288.0
input.transfer.low (Low voltage transfer point (V)) : 176.0
input.transfer.reason (Reason for last transfer to battery) : Automatic or explicit self test
[... cut ...]
input.voltage.nominal (Nominal input voltage (V)) : 230
ups.date (Internal UPS clock date) : 2023-12-13
ups.firmware (UPS firmware) : 958.g3 .I USB FW:g3
ups.id (UPS system identifier) : RT-AC68U
ups.load (Load on UPS (percent of full)) : 19.0
ups.mfr (UPS manufacturer) : APC
ups.model (UPS model) : Back-UPS RS 1600MI
ups.realpower.nominal (UPS real power rating (W)) : 960.0
ups.serial (UPS serial number) : 5B2209T21145
ups.status (UPS status) : OL
ups.test.result (Results of last self test) : NO
ups.time (Internal UPS clock time) : 08:58:55
```

dpvDuncan, 2023-12-15:
Hello @gbakeman , the latest debug build (WinNUT-Client-debugbuild-v2.2.8747.34564) allows me to use NinNUT again, but the error seems to remain (see logs attached) and the load value indicates 0 even though the load percentage indicates 37%:
[image]

```
12/15/2023 17:03:00 [20428, WinNUT]: Connection parameters Changed. Force Reconnect
12/15/2023 17:03:00 [20428, WinNUT]: Running Client disconnect subroutine.
12/15/2023 17:03:00 [20428, UPS_Device]: Processing request to disconnect...
12/15/2023 17:03:00 [20428, WinNUT]: Update all informations displayed to empty values
12/15/2023 17:03:00 [20428, WinNUT]: Update Icon
12/15/2023 17:03:00 [20428, WinNUT]: Status Icon Changed
12/15/2023 17:03:00 [20428, WinNUT]: New Icon Value For Systray : 1216
12/15/2023 17:03:00 [20428, WinNUT]: New Icon Value For Gui : 1216
12/15/2023 17:03:00 [20428, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Non connecté
12/15/2023 17:03:00 [20428, WinNUT]: Battery Status => Unknown
12/15/2023 17:03:00 [20428, WinNUT]: Disconnected from Nut Host
12/15/2023 17:03:00 [20428, String]: New Log to CB_Current Log : Déconnectée du serveur NUT
12/15/2023 17:03:00 [20428, WinNUT]: Client UPS_Connect subroutine beginning.
12/15/2023 17:03:00 [20428, UPS_Device]: Beginning connection: @192.168.1.3:3493, Name: tecnoware_era_plus_1100
12/15/2023 17:03:00 [20428, Nut_Socket]: Attempting TCP socket connection to 192.168.1.3:3493...
12/15/2023 17:03:00 [20428, Nut_Socket]: Connection established and streams ready for 192.168.1.3:3493
12/15/2023 17:03:00 [20428, Nut_Socket]: Attempting authentication...
12/15/2023 17:03:00 [20428, Nut_Socket]: Error while attempting to log in: USERNAMEREQUIRED (ERR USERNAME-REQUIRED)
Query: LOGIN
12/15/2023 17:03:00 [20428, Nut_Socket]: NUT server reports VER: 2.8.0 NETVER: 
12/15/2023 17:03:00 [20428, UPS_Device]: Retrieving basic UPS product information...
12/15/2023 17:03:00 [20428, UPS_Device]: ups.mfr is not supported by server.
12/15/2023 17:03:00 [20428, UPS_Device]: Apply Fallback Value when retrieving ups.mfr
12/15/2023 17:03:00 [20428, UPS_Device]: ups.model is not supported by server.
12/15/2023 17:03:00 [20428, UPS_Device]: Apply Fallback Value when retrieving ups.model
12/15/2023 17:03:00 [20428, UPS_Device]: ups.serial is not supported by server.
12/15/2023 17:03:00 [20428, UPS_Device]: Apply Fallback Value when retrieving ups.serial
12/15/2023 17:03:00 [20428, UPS_Device]: ups.firmware is not supported by server.
12/15/2023 17:03:00 [20428, UPS_Device]: Apply Fallback Value when retrieving ups.firmware
12/15/2023 17:03:00 [20428, UPS_Device]: Determining best method to calculate power usage...
12/15/2023 17:03:00 [20428, UPS_Device]: ups.realpower is not supported by server.
12/15/2023 17:03:00 [20428, UPS_Device]: Unhandled error while getting ups.realpower
12/15/2023 17:03:00 [20428, UPS_Device]: ups.realpower.nominal is not supported by server.
12/15/2023 17:03:00 [20428, UPS_Device]: Unhandled error while getting ups.realpower.nominal
12/15/2023 17:03:00 [20428, UPS_Device]: Using VoltAmpCalc method to calculate power usage.
12/15/2023 17:03:00 [20428, UPS_Device]: battery.capacity is not supported by server.
12/15/2023 17:03:00 [20428, UPS_Device]: Apply Fallback Value when retrieving battery.capacity
12/15/2023 17:03:00 [20428, UPS_Device]: output.frequency.nominal is not supported by server.
12/15/2023 17:03:00 [20428, UPS_Device]: Apply Fallback Value when retrieving output.frequency.nominal
12/15/2023 17:03:00 [20428, UPS_Device]: Completed retrieval of basic UPS product information.
12/15/2023 17:03:00 [20428, WinNUT]: tecnoware_era_plus_1100 has indicated it's ready to start sending data.
12/15/2023 17:03:00 [20428, WinNUT]: Update Icon
12/15/2023 17:03:00 [20428, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Connecté
12/15/2023 17:03:00 [20428, WinNUT]: Connection to Nut Host Established
12/15/2023 17:03:00 [20428, String]: New Log to CB_Current Log : Connexion au Serveur NUT 192.168.1.3:3493 établie
12/15/2023 17:03:00 [20428, WinNUT]: Parameter Dial Voltage Battery Need to be Updated
12/15/2023 17:03:00 [20428, WinNUT]: Parameter Dial Voltage Battery Updated
12/15/2023 17:03:00 [20428, WinNUT]: WinNut Preferences Applied.
12/15/2023 17:03:00 [20428, String]: New Log to CB_Current Log : Préférences de WinNUT Modifiées
12/15/2023 17:03:00 [20428, WinNUT]: Main Gui Has Focus
12/15/2023 17:03:00 [20428, WinNUT]: Update Icon
12/15/2023 17:03:09 [20428, WinNUT]: Force Reconnect from menu
12/15/2023 17:03:09 [20428, WinNUT]: Running Client disconnect subroutine.
12/15/2023 17:03:09 [20428, UPS_Device]: Processing request to disconnect...
12/15/2023 17:03:09 [20428, WinNUT]: Update all informations displayed to empty values
12/15/2023 17:03:09 [20428, WinNUT]: Update Icon
12/15/2023 17:03:09 [20428, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Non connecté
12/15/2023 17:03:09 [20428, WinNUT]: Battery Status => Unknown
12/15/2023 17:03:09 [20428, WinNUT]: Disconnected from Nut Host
12/15/2023 17:03:09 [20428, String]: New Log to CB_Current Log : Déconnectée du serveur NUT
[... 50 lines cut ...]
12/15/2023 17:03:19 [20428, UPS_Device]: UPS statuses have CHANGED...
12/15/2023 17:03:19 [20428, UPS_Device]: Current statuses: OL
12/15/2023 17:03:19 [20428, WinNUT]: Handling new UPS status(es)...
12/15/2023 17:03:19 [20428, WinNUT]: Updating UPS data for Form.
12/15/2023 17:03:19 [20428, WinNUT]: Updating battery icons based on charge percent: 100%
[... cut ...]
WinNUT Client - 2.2
Connecté
Sur Secteur
Batterie OK
12/15/2023 17:03:19 [20428, WinNUT]: Battery Status => Charged
12/15/2023 17:03:23 [20428, WinNUT]: Open Pref Gui From Menu
12/15/2023 17:03:23 [20428, Pref_Gui]: Load Pref Gui
12/15/2023 17:03:23 [20428, WinNUT]: Main Gui Lose Focus
12/15/2023 17:03:23 [20428, WinNUT]: Update Icon
12/15/2023 17:03:23 [20428, Pref_Gui]: Pref Gui Opened.
12/15/2023 17:03:26 [20428, Pref_Gui]: Setting LogControl statuses.
12/15/2023 17:03:27 [20428, Pref_Gui]: Show LogFile
```

gbakeman, 2023-12-16:
It sounds like I still haven't fully understood the problem! 😄

I'm going to take a deeper look at this and ask for your patience while I work on a few things. Please stay tuned for another update.

gbakeman, 2023-12-17:
I've made an attempt to standardize some code that I think was causing the problem. I'm not quite sure how to change my computer's locale to correctly reproduce the issue, so thank you for testing these changes for me.

Please give the new build [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7240341344?pr=125) a try and let me know if you see any problems. In the future if you want to upload a log, please click and drag the log into your comment on GitHub - that way it's uploaded as an attachment. Thank you!

dpvDuncan, 2023-12-20:
Your last build seems to work correctly:

[image]

Thank you ^_^

[WinNUT-Client-2023-12-20.log](https://github.com/nutdotnet/WinNUT-Client/files/13728593/WinNUT-Client-2023-12-20.log)

gbakeman, 2023-12-20:
Excellent, thank you everyone. I'll have a pre-release update out soon.

closed 2023-12-20 by gbakeman