nutdotnet/WinNUT-Client #124: Connection retry interval too small

pvanderp, 2023-12-12:
When using a NAS as a NUT server, if the NAS is rebooted for new firmware installation, WinNUT will time-out retrying the connection within the installation-reboot interval. This leaves several clients unprotected.

Suggestion is to increase the wait time in between retries. You could also double the wait time for each attempt.

The complete shutdown-install-reboot procedure for a NAS can take up to 5 or 10 minutes.

gbakeman, 2023-12-12:
Hi @pvanderp ,

I know this is something people have been requesting. I think #65 addresses this. If you agree, go ahead and subscribe to that issue and provide feedback, and let's close this one.

pvanderp, 2023-12-13:
Duplicate of #65

closed 2023-12-13 by pvanderp