nutdotnet/WinNUT-Client #128: WinNUT always reconnects after startup with NUT server from OPNsense | INVALIDARGUMENT | FormatException

codiflow, 2024-01-25:
- [x] Completed below form
- [x] Attached debug log file
- [x] Copy output in File -> UPS Variable

**WinNUT Version: 2.2.8719.24624** 
**Windows OS Version: Windows Server 2016 1607 (Build 14393.6529)** 

**Describe the bug**
WinNUT permanently disconnects from NUT running on OPNsense.
WinNUT is not reconnecting to the NUT service anymore **which has worked just fine a few months ago** (I did not track this down immediately as it worked again sometimes). It tries to connect and after the "poll intervall" period it tries to reconnect – see the logs.

The setup is as follows:
The UPS is connected via USB to the OPNsense firewall. The APCUPSd is the main UPS server on this firewall also providing a netserver instance. The NUT server on OPNsense internally connects to the APCUPSd netserver daemon and provides a NUT server for other devices. This way we can use NUT and APCUPSd in parallel in our network.

Unfortunately I don't have any logs on OPNsense side which could provide additional hints.

There's also a Synology NAS which is connected to the same NUT server and it just works fine – no issues.

**Expected behavior**
Connection should be established and remain stable.

**Additional context**
Debug log from WinNUT:
```
WinNUT Client - 2.2
Verbindung zu 192.168.0.1:3493 verloren
25.01.2024 13:15:29 [2752, WinNUT]: Battery Status => Unknown
25.01.2024 13:15:29 [2752, WinNUT]: Update Icon
25.01.2024 13:15:29 [2752, UPS_Device]: Reconnection Process Started
25.01.2024 13:15:34 [2752, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Wiederverbindung läuft
Versuch 1 von 30
25.01.2024 13:15:34 [2752, WinNUT]: Update Icon
25.01.2024 13:15:34 [2752, UPS_Device]: Try Reconnect 1 / 30
25.01.2024 13:15:34 [2752, String]: New Log to CB_Current Log : Versuche Wiederverbindung 1 / 30
25.01.2024 13:15:34 [2752, UPS_Device]: Beginning connection: admin@192.168.0.1:3493, Name: ups [AutoReconnect]
25.01.2024 13:15:34 [2752, Nut_Socket]: Attempting TCP socket connection to 192.168.0.1:3493...
25.01.2024 13:15:34 [2752, Nut_Socket]: Connection established and streams ready for 192.168.0.1:3493
25.01.2024 13:15:34 [2752, Nut_Socket]: Attempting authentication...
25.01.2024 13:15:34 [2752, Nut_Socket]: Error while attempting to log in: INVALIDARGUMENT (ERR INVALID-ARGUMENT)
Query: LOGIN
25.01.2024 13:15:34 [2752, Nut_Socket]: NUT server reports VER: 2.8.1 NETVER: 
25.01.2024 13:15:34 [2752, UPS_Device]: ups.realpower is not supported by server.
25.01.2024 13:15:34 [2752, UPS_Device]: Unhandled error while getting ups.realpower
25.01.2024 13:15:34 [2752, UPS_Device]: Using NominalPowerCalc method to calculate power usage.
25.01.2024 13:15:34 [2752, UPS_Device]: battery.capacity is not supported by server.
25.01.2024 13:15:34 [2752, UPS_Device]: Apply Fallback Value when retrieving battery.capacity
25.01.2024 13:15:34 [2752, UPS_Device]: output.frequency.nominal is not supported by server.
25.01.2024 13:15:34 [2752, UPS_Device]: Apply Fallback Value when retrieving output.frequency.nominal
25.01.2024 13:15:34 [2752, WinNUT]: ups has indicated it's ready to start sending data.
25.01.2024 13:15:34 [2752, WinNUT]: Update Icon
25.01.2024 13:15:34 [2752, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Verbunden
25.01.2024 13:15:34 [2752, WinNUT]: Connection to Nut Host Established
25.01.2024 13:15:34 [2752, String]: New Log to CB_Current Log : Verbindung zum Nut Host 192.168.0.1:3493 hergestellt
25.01.2024 13:15:34 [2752, UPS_Device]: Nut Host Reconnected
25.01.2024 13:15:34 [2752, WinNUT]: ups has indicated it's ready to start sending data.
25.01.2024 13:15:34 [2752, WinNUT]: Update Icon
25.01.2024 13:15:34 [2752, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.2
Verbunden
25.01.2024 13:15:34 [2752, WinNUT]: Connection to Nut Host Established
25.01.2024 13:15:34 [2752, String]: New Log to CB_Current Log : Verbindung zum Nut Host 192.168.0.1:3493 hergestellt
25.01.2024 13:15:39 [2752, UPS_Device]: Enter Retrieve_UPS_Datas
25.01.2024 13:15:39 [2752, UPS_Device]: output.frequency is not supported by server.
25.01.2024 13:15:39 [2752, UPS_Device]: Apply Fallback Value when retrieving output.frequency
25.01.2024 13:15:39 [2752, UPS_Device]: input.frequency is not supported by server.
25.01.2024 13:15:39 [2752, UPS_Device]: Apply Fallback Value when retrieving input.frequency
[... 70 lines cut ...]
ups.serial (UPS serial number) : <REDACTED>
ups.status (UPS status) : OL
ups.test.result (Results of last self test) : NO
ups.time (Internal UPS clock time) : 13:17:23
```

While I wanted to copy the UPS variables also this error appeared – maybe this has to do with WinNUT not having been connected to the UPS at this very moment?
```
System.Runtime.InteropServices.ExternalException (0x800401D0): Der angeforderte Clipboard-Vorgang war nicht erfolgreich.
   bei System.Windows.Forms.Clipboard.ThrowIfFailed(Int32 hr)
   bei System.Windows.Forms.Clipboard.SetDataObject(Object data, Boolean copy, Int32 retryTimes, Int32 retryDelay)
   bei System.Windows.Forms.Clipboard.SetText(String text, TextDataFormat format)
   bei System.Windows.Forms.Clipboard.SetText(String text)
   bei WinNUT_Client.List_Var_Gui.Btn_Clip_Click(Object sender, EventArgs e)
   bei System.Windows.Forms.Control.OnClick(EventArgs e)
   bei System.Windows.Forms.Button.OnClick(EventArgs e)
   bei System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   bei System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   bei System.Windows.Forms.Control.WndProc(Message& m)
   bei System.Windows.Forms.ButtonBase.WndProc(Message& m)
   bei System.Windows.Forms.Button.WndProc(Message& m)
   bei System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   bei System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```

gbakeman, 2024-01-31:
Sorry I missed your issue report - for some reason I did not see an email about it.

This issue was addressed in [Pre-Release v2.3.8754](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8754). Please update to that and let us know how it goes.

gbakeman, 2024-02-08:
Closing as duplicate

closed 2024-02-08 by gbakeman

codiflow, 2024-02-09:
Sorry for my late response, [Pre-Release v2.3.8754](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8754) fixed the issue.

Thanks!