nutdotnet/WinNUT-Client #129: 先连接建立成功，1秒钟不到，.OverflowException: 算术运算导致溢出。连接断开。

aazz-top, 2024-02-01:
我的UPS电源连接在群晖DS220+上。
群晖的设置已经正确，所有状态正常，IP也允许局域网的IP。
通过ssh的 upsc ups 也看见了ups 的状态。
但是windows上无法用winnut连接。连接成功后会立刻连接失败。.OverflowException: 算术运算导致溢出。
先连接建立成功，1秒钟不到，.OverflowException: 算术运算导致溢出。连接断开。
试了3个版本 2.0 2.2 。

System.OverflowException: 算术运算导致溢出。
   在 WinNUT_Client.WinNUT.Event_UpdateBatteryState(String Reason) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:行号 533
   在 WinNUT_Client.WinNUT.UPSDisconnectedEvent() 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:行号 359
   在 WinNUT_Client_Common.UPS_Device.DisconnectedEventHandler.Invoke()
   在 WinNUT_Client_Common.UPS_Device.Disconnect(Boolean cancelReconnect, Boolean forceful) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb:行号 185
   在 WinNUT_Client_Common.UPS_Device.Retrieve_UPS_Datas(Object sender, EventArgs e) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb:行号 366
   在 System.Windows.Forms.Timer.OnTick(EventArgs e)
   在 System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   在 System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

gbakeman, 2024-02-02:
Hello,

Please update to the latest [Pre-Release](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8754) and let me know if you still experience the error.

aazz-top, 2024-02-05:
> Hello,
> 
> Please update to the latest [Pre-Release](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8754) and let me know if you still experience the error.

PC: windosw 10 22H2

[Pre-Release v2.3.8800](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8800)

System.OverflowException: 算术运算导致溢出。
   在 WinNUT_Client.WinNUT.Event_UpdateBatteryState(String Reason) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:行号 587
   在 WinNUT_Client.WinNUT.UPSDisconnectedEvent() 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:行号 480
   在 WinNUT_Client_Common.UPS_Device.DisconnectedEventHandler.Invoke()
   在 WinNUT_Client_Common.UPS_Device.Disconnect(Boolean cancelReconnect, Boolean forceful) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb:行号 186
   在 WinNUT_Client_Common.UPS_Device.Retrieve_UPS_Datas(Object sender, EventArgs e) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb:行号 367
   在 System.Windows.Forms.Timer.OnTick(EventArgs e)
   在 System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   在 System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

aazz-top, 2024-02-05:
> Hello,
> 
> Please update to the latest [Pre-Release](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8754) and let me know if you still experience the error.

PC: windows 10 22H2

[Pre-Release v2.3.8800](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8800)

System.OverflowException: 算术运算导致溢出。
   在 WinNUT_Client.WinNUT.Event_UpdateBatteryState(String Reason) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:行号 587
   在 WinNUT_Client.WinNUT.UPSDisconnectedEvent() 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client\WinNUT.vb:行号 480
   在 WinNUT_Client_Common.UPS_Device.DisconnectedEventHandler.Invoke()
   在 WinNUT_Client_Common.UPS_Device.Disconnect(Boolean cancelReconnect, Boolean forceful) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb:行号 186
   在 WinNUT_Client_Common.UPS_Device.Retrieve_UPS_Datas(Object sender, EventArgs e) 位置 C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb:行号 367
   在 System.Windows.Forms.Timer.OnTick(EventArgs e)
   在 System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   在 System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

gbakeman, 2024-02-07:
I can see where the error is occurring, although I'm not quite sure why. Please download the debug build from [here ](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7820563689?pr=131) once it is available and let me know if it works. If you encounter an error again, please attach the full log file to your comment.

aazz-top, 2024-02-08:
> I can see where the error is occurring, although I'm not quite sure why. Please download the debug build from [here ](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7820563689?pr=131) once [...]

[WinNUT-Client-2024-02-08.log](https://github.com/nutdotnet/WinNUT-Client/files/14203744/WinNUT-Client-2024-02-08.log)
[System.NullReferenceException 未将对象引.txt](https://github.com/nutdotnet/WinNUT-Client/files/14203746/System.NullReferenceException.txt)

thank you very much !
this is log file

gbakeman, 2024-02-08:
It looks like the original problem may have been fixed, although you found another one relating to the log file! I have another build going [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7833577418?pr=131). Please see if you can find any issues in this one. Thank you for your help!

aazz-top, 2024-02-09:
It does not work properly and is automatically disconnected repeatedly.also.
[WinNUT-Client-2024-02-09.log](https://github.com/nutdotnet/WinNUT-Client/files/14216623/WinNUT-Client-2024-02-09.log)

gbakeman, 2024-02-09:
I'm sorry, I'm going to have to look into this more. I think I need to make some changes in how WinNUT logs information so that will make it easier for me to troubleshoot issues, and it also looks like we may be facing an edge case or two being reported by the UPS. Please stand by.

aazz-top, 2024-02-16:
> I'm sorry, I'm going to have to look into this more. I think I need to make some changes in how WinNUT logs information so that will make it easier for me to troubleshoot issues, and it also looks like we may be facing an edge case or two being reported by the UPS. Please stand by.

what can i do for you

gbakeman, 2024-02-17:
I just had a chance to make another small update. I don't expect this will fix the immediate issue, but I hope that it does improve logging somewhat. Please give [this build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7943380239?pr=131) a try and upload the log file. After this, I should be able to fix the current issue you're experiencing. You've been wonderful so far, thank you for your patience.

aazz-top, 2024-02-18:
> I just had a chance to make another small update. I don't expect this will fix the immediate issue, but I hope that it does improve logging somewhat. Please give [this [...]

[WinNUT-Client-2024-02-18.log](https://github.com/nutdotnet/WinNUT-Client/files/14321302/WinNUT-Client-2024-02-18.log)
plz cheak it thanks

gbakeman, 2024-02-18:
I've made some changes that I think will fix the issue you're facing now, [download here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7952048959?pr=131). Whether this one is successful or not, please upload the log again after testing. Thank you again for testing this.

aazz-top, 2024-02-28:
> I've made some changes that I think will fix the issue you're facing now, [download here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/7964806438?pr=131). Whether this one is successful or not, please upload the log again after testing. Thank you again for testing this.

[WinNUT-Client-2024-02-28.log](https://github.com/nutdotnet/WinNUT-Client/files/14430001/WinNUT-Client-2024-02-28.log)

Thank you very much, it is running successfully.Thank youThank youThank youThank youThank you

gbakeman, 2024-02-28:
And thank you for your patience and persistence - you helped me fix at least three different bugs here. I do want to mention that it looks like your UPS or NUT server are reporting strange values for different variables, mostly related to the battery. You may want to try restarting your NUT server to see if that corrects some of the values again.

In the meantime, I will work on sending out another pre-release so you can continue using WinNUT normally without the debug build.

closed 2024-02-28 by gbakeman