nutdotnet/WinNUT-Client #136: 6V When no battery.voltage variable avalibe

masterwishx, 2024-02-28:
updated from WinNUT Client v2.3.8800 to lasted version :  

have Eaton  9E 2000i but seems dont have : battery.voltage variable
and have 6V instead of 12V  like was in versions befor :  

[image]

gbakeman, 2024-02-28:
Hello,

Is it the same way with [pre-release 2.3.8824](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8824)? I made several changes to how missing variables are handled in that one.

masterwishx, 2024-02-28:
> Hello,
> 
> Is it the same way with [pre-release 2.3.8824](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8824)? I made several changes to how missing variables are handled in that one.

Yes updated to this latest version

gbakeman, 2024-02-29:
Ok, thank you for double checking. I think what I'd like to do is merge this issue with #116 since I intend to change how WinNUT interprets missing variables. I'll make a note of your particular situation (missing battery.voltage) in the OP, and feel free to make a comment if you want to provide any additional info. But in the meantime I think I'll look into #135 first.

masterwishx, 2024-02-29:
> Ok, thank you for double checking. I think what I'd like to do is merge this issue with #116 since I intend to change how WinNUT interprets missing variables. I'll make a note of your particular [...]

Sure, just wanted to inform of this case, becose had no this problem befor, as we all know most ups have 12v battery's  :)

gbakeman, 2024-02-29:
Sorry, it sounded like your NUT server/UPS just wasn't providing the variable to you, but now it sounds like there may be an underlying issue. Could you please upload two things for me:
- Variables list from the **File** menu output
- Log file set to _Debug_ mode

masterwishx, 2024-03-01:
> Sorry, it sounded like your NUT server/UPS just wasn't providing the variable to you, but now it sounds like there may be an underlying issue. Could you please upload two things for me:
> 
> * Variables list from the **File** menu output
> * Log file set to _Debug_ mode

I'm using nut plugin in Unraid server, and I'm sure 100% I don't have value of voltage of the battery, only value of ampere 7A.
I will upload soon...

masterwishx, 2024-03-01:
From Nut plugin :   

[image]

masterwishx, 2024-03-01:
Eaton9E2000i (Eaton/unknown 2000/)
battery.capacity (Battery capacity (Ah)) : 7.00
battery.charge (Battery charge (percent of full)) : 100
battery.charge.low (Remaining battery level when UPS switches to LB (percent)) : 20
battery.charge.restart (Minimum battery level for restart after power off (percent)) : 0
battery.protection (Prevent deep discharge of battery) : yes
battery.runtime (Battery runtime (seconds)) : 3470
battery.runtime.low (Remaining battery runtime when UPS switches to LB (seconds)) : 180
battery.type (Battery chemistry) : 
device.mfr (Description unavailable) : Eaton
device.model (Description unavailable) : unknown 2000
device.type (Description unavailable) : ups
driver.debug (Current debug verbosity level of the driver program) : 0
driver.flag.allow_killpower (Safety flip-switch to allow the driver daemon to send UPS shutdown command (accessible via driver.killpower)) : 0
driver.name (Driver name) : usbhid-ups
driver.parameter.pollfreq (Description unavailable) : 30
driver.parameter.pollinterval (Description unavailable) : 2
driver.parameter.port (Description unavailable) : auto
driver.parameter.synchronous (Description unavailable) : auto
driver.state (Description unavailable) : quiet
driver.version (Driver version - NUT release) : 2.8.1
driver.version.data (Description unavailable) : MGE HID 1.46
driver.version.internal (Internal driver version) : 0.52
driver.version.usb (USB library version) : libusb-1.0.26 (API: 0x1000108)
input.bypass.frequency (Description unavailable) : 50.0
input.bypass.voltage (Description unavailable) : 232.0
input.frequency (Input line frequency (Hz)) : 50.0
input.frequency.nominal (Nominal input line frequency (Hz)) : 50
input.transfer.high (High voltage transfer point (V)) : 300
input.transfer.low (Low voltage transfer point (V)) : 100
input.voltage (Input voltage (V)) : 232.0
input.voltage.nominal (Nominal input voltage (V)) : 233
outlet.1.status (Outlet switch status) : on
outlet.desc (Outlet description) : Main Outlet
outlet.id (Outlet system identifier) : 0
outlet.switchable (Outlet switch ability) : no
output.current (Output current (A)) : 1.70
output.frequency (Output frequency (Hz)) : 50.0
output.frequency.nominal (Nominal output frequency (Hz)) : 50
output.voltage (Output voltage (V)) : 229.0
output.voltage.nominal (Nominal output voltage (V)) : 230
ups.beeper.status (UPS beeper status) : muted
ups.delay.shutdown (Interval to wait after shutdown with delay command (seconds)) : 20
ups.delay.start (Interval to wait before (re)starting the load (seconds)) : 30
ups.firmware (UPS firmware) : 
ups.load (Load on UPS (percent of full)) : 20
ups.load.high (Load when UPS switches to overload condition (percent)) : 105
ups.mfr (UPS manufacturer) : Eaton
ups.model (UPS model) : unknown 2000
ups.power (Current value of apparent power (VA)) : 412
ups.power.nominal (UPS power rating (VA)) : 2000
ups.productid (Product ID for USB devices) : ffff
ups.realpower (Current value of real power (W)) : 283
ups.realpower.nominal (UPS real power rating (W)) : 1600
ups.shutdown (Description unavailable) : enabled
ups.start.auto (UPS starts when mains is (re)applied) : yes
ups.start.battery (Allow to start UPS from battery) : yes
ups.start.reboot (UPS reboots when power returns during shutdown delay) : yes
ups.status (UPS status) : OL
ups.temperature (UPS temperature (degrees C)) : 24.9
ups.test.interval (Interval between self tests (seconds)) : 604800
ups.test.result (Results of last self test) : Done and passed
ups.timer.shutdown (Time before the load will be shutdown (seconds)) : -1
ups.timer.start (Time before the load will be started (seconds)) : -1
ups.type (UPS type) : online
ups.vendorid (Vendor ID for USB devices) : 0463

masterwishx, 2024-03-01:
[WinNUT-Client-2024-03-01.log](https://github.com/nutdotnet/WinNUT-Client/files/14458002/WinNUT-Client-2024-03-01.log)

From the Log: 

[image]

gbakeman, 2024-03-02:
Thank you for the detailed amount of the info, I think I understand the issue now! :)

So we're on the same page that your UPS has never provided the `battery.voltage` variable before, that's all good. Up until recently, when WinNUT encounters an error retrieving a variable, it would be reported with a "fallback" or ideal value, which is why you used to see 12V reported as the battery voltage (even though I think UPS batteries are typically 14V?) At any rate, I [recently changed](https://github.com/nutdotnet/WinNUT-Client/pull/131) how WinNUT handles errors retrieving variables, by setting the value to a more obvious of out range value (typically -1). But I think you've just discovered another bug, which is that the gauges themselves are clamping out of range values to their nearest minimum or maximum on the dial range. I think this is a problem which will need to be addressed either in #116 , or in the https://github.com/nutdotnet/AGauge project.

I'll keep this open as a reminder for those other items. Thanks again!

masterwishx, 2024-03-02:
> I think UPS batteries are typically 14V?

The model im Using Eaton 9E2000i have 6x 12V7A

masterwishx, 2024-03-02:
[image]

gbakeman, 2024-03-02:
Ah, 12V does make more sense! I'm surprised yours doesn't provide voltage monitoring considering it seems more advanced than most. Do you know if there might be a firmware update?

masterwishx, 2024-03-02:
> Ah, 12V does make more sense! I'm surprised yours doesn't provide voltage monitoring considering it seems more advanced than most. Do you know if there might be a firmware update?

Not sure , This `9E` Model True online double-conversion but not the most advanced as Eaton have better `9SX` Model .
im using the most newer v2.8.1  of NUT for now. using NUT plugin in Unraid Server.
i was not digging in this question about why its not monitoring 12v , but will try to check it ...

Vladdrako, 2025-01-12:
Hi to all :) I have Powercom BNT-800AP, which does not provide battery voltage. It has a 12V battery. And ofc WinNUT shows a 6V for the battery that looks ugly. But NUT allows us to override values. I set it like this:
```
override.battery.voltage.low = 10.80
override.battery.voltage.high = 14.20
override.battery.voltage.nominal = 12 (not sure I needed this)
```
So, why not calculate the approximate voltage from overrides and battery.charge? I think it's better than nothing.

For example, I override `override.ups.realpower.nominal = 480` to make wattage calculations work. Now I have a power meter socket, and it looks like I need some corrections because WinNUT lowering the value, but I need more testing.

gbakeman, 2025-01-12:
Hi @Vladdrako , good points you made there. In the future, I do want WinNUT to pull more calibration data directly from the UPS instead of expecting the user to provide it, so hopefully that provides some amount of improvement. I could also see us approximating values like a battery voltage, although that might be more complex since as far as I know, battery voltage and charge (percent out of 100) aren't directly proportional so that's something we need to careful with.

In other news, I've been taking a break so haven't gotten much work done lately. There are several pull requests and bugs that I want to get fixed in another pre-release soon, then hopefully I can get back to tackling larger issues like this.

masterwishx, 2025-01-12:
> Hi @Vladdrako , good points you made there. In the future, I do want WinNUT to pull more calibration data directly from the UPS instead of expecting the user to provide it, so hopefully that [...]

After some prs in NUT we have added to Eaton models include mine 9E support for `battery.voltage =84v` and `battery.nominal =72V`

gbakeman, 2026-02-02:
Good points @Yoji808 and thank you for posting. You've touched on some areas I definitely want to improve on, such as how the gauges display invalid values, how WinNUT handles missing variables in the first place, and how spammy logging can get as well. Thanks for your support!