nutdotnet/WinNUT-Client #142: LogFile controls are disabled when the Create LogFile option is on

gbakeman, 2024-03-09:
[image]

**WinNUT Version:** 2.3.8834
**Windows OS Version:** Windows 11

**Describe the bug**
LogFile management controls in the Preferences window are disabled, despite logging apparently being enabled.

masterwishx, 2024-03-10:
Have also in in roaming, also I think when set to info, no logs enabled while it's enabled in setting. But when change to debug then it's enabling the log

masterwishx, 2024-03-10:
Just checked now , it was info and not enabled but enabled in setting .
1. set to debug then apply. (still shown as disabled)
2. closed app (its minimized to tray (by settings) )
3. opened again - its shown as enabled now.
4. changed to info - its enabled now

masterwishx, 2024-03-10:
but if i close app ,then open -  its disabled again

gbakeman, 2024-04-24:
@masterwishx I've just made some changes with @MaxMadcc and we think the problem has been fixed now. If you have the time, please download a debug build from the linked pull request, and I think I can have another release out soon. Thank you for your patience!

closed 2024-04-24 by gbakeman